dsmrreader/dsmr-reader #1632: Na meterwissel of verhuizing is jaarverbruik niet meer goed🚨

rtrinks, 2022-05-15:
### Description

Er wordt aangegeven dat bij verhuizing of meterwissel, dat dit enkel de dag is waarop dat is uitgevoerd.
Maar op jaartotaal blijft dit een issue en heb ik een groot negatief getal wat dan logisch is.

Maar hoe lossen we dit op? Ik kan het nergens vinden en ook in de FAQ niet.

[image]

### DSMR-reader version

DSMR-reader v4.18

dennissiemensma, 2022-05-15:
Bedankt voor je vraag. Heb je de betreffende wisseldag zelf al gecorrigeerd?

Zie https://dsmr-reader.readthedocs.io/nl/v5/how-to/troubleshooting/smart-meter-replacement.html

Als het goed is klopt alles daarna weer, omdat het de maand- en jaartotalen een optelsom zouden moeten zijn van alle individuele dagen.

rtrinks, 2022-05-15:
Aangepast en staat nu in ieder geval weer helemaal goed. Bedankt.

closed 2022-05-15 by rtrinks