dsmrreader/dsmr-reader #1635: Hoogste piekvermogen per dag/maand visualiseren

dennissiemensma, 2022-05-23:
### Feature

Follow-up van #1084.

- Hoogste waarde per dag opslaan
- Hoogste waarde per dag tonen
- Hoogste waarde per maand tonen

dennissiemensma, 2022-07-17:
Tekst alvast aangepast

[image]

VBP8501, 2022-07-17:
Ziet er goed uit 😀

dennissiemensma, 2022-07-17:
Berekening nu ook aangpast als het niet exact 15m is. 

Het resterende werk zit in de grafieken en dagtotalen, dus dat komt later.

dennissiemensma, 2022-07-25:
Nu ook een grafiek werkend:

[image]

dennissiemensma, 2022-08-01:
Ik denk dat ik de komende release de fixes + grafiek uitbreng, anders blokkeert het redelijk wat en het is opzich ook niet erg om kleinere updates te doen. 

De rest blijft gewoon bij de TODO's van dit issue.

dennissiemensma, 2022-08-02:
Onderdeel van [v5.6](https://github.com/dsmrreader/dsmr-reader/releases/tag/v5.6.0)

FredericMa, 2022-08-13:
Ik was net even de kwartiervermogens aan het bekijken en merk dat in de grafiek de eenheid nog kw/15m is in plaats van kwh/15m zoals reeds aangepast in de configuratie pagina. Gezien het discrete waarden zijn, lijkt het me het ook intuïtiever om hier standaard een bar-chart van te maken. Thanks voor het mooie werk!

dennissiemensma, 2022-08-13:
Bedankt voor je aanvulling. 

- Qua eenheid weet ik niet precies wat je bedoelt. Het is juist aangepast van `kwh/15m` naar `kw/15m`. Of lees ik er overheen in de screenshot hierboven? https://github.com/dsmrreader/dsmr-reader/issues/1635#issuecomment-1194613971
- Ik zal de grafiektype later aansluiten op de bar/line voorkeur in de Frontend instellingen

FredericMa, 2022-08-14:
> Qua eenheid weet ik niet precies wat je bedoelt. Het is juist aangepast van `kwh/15m` naar `kw/15m`. Of lees ik er overheen in de screenshot hierboven?

Je hebt gelijk, mijn aanvulling hierop is niet correct. Ik heb de website van de netbeheerder er even bijgenomen en de eenheid zou wel kW moeten zijn ipv kW/15m aangezien het over een vermogen gaat en kW/15m een verbruik aangeeft zoals kWh.

Ook de eenheid van het verbruik is strikt genomen uitgedrukt in kWh ipv kWh/15m maar ik snap wel hoe de benaming tot stand is gekomen.

Op de site van de netbeheerder worden onder de vraag "Wanneer is het piekvermogen gemeten?", de eenheden duidelijk aangegeven: https://www.fluvius.be/nl/veelgestelde-vragen/mijn-fluvius/verbruik

dennissiemensma, 2022-08-14:
Bedankt voor je update. Het is mij om het even hoe het heet, alleen blijf ik het niet hernoemen ;-). Wellicht kun je met @VBP8501 afstemmen wat de juiste term is, dan houden dat die uitkomst aan.

VBP8501, 2022-08-23:
Sorry was even op vakantie :)

kW/15min is fout want dat geeft de versnelling van het vermogen aan :D

kW = kiloJoule/s = gelijkaardig aan km/u
kWh = kW * uur = gelijkaardig aan km afgelegde weg
kW/tijd = gelijkaardig aan in 10s van 0km/u naar 100km/u

Het is ofwel kW ofwel kWh/15min. Maar in de grafiek zou ik het simpel houden en gewoon spreken van kW

dennissiemensma, 2022-08-23:
Ik zal het dan aanpassen naar kW in de volgende versie:

[image]

dennissiemensma, 2022-09-26:
Een deel van de wijzigingen gaat mee met v5.7, de rest weer een bump naar de volgende.

Glodenox, 2022-10-01:
Een andere handige toevoeging lijkt me het beschikbaar maken van deze waarden via MQTT. Ik ben alleen niet zeker in welke vorm dit handig zou zijn. "Huidig kwartiervermogen", "Hoogste kwartiervermogen vandaag"?

dennissiemensma, 2022-10-01:
Ik denk beide. Uiteindelijk zal het hoogste kwartiervermogen onderdeel worden van de dagtotalen en daarmee ook te ontsluiten via het bestaande MQTT model.

Voor de huidige (na opslag) is uiteindelijk wel iets te maken. Eigenlijk net zoals al gebeurt voor de andere datamodellen. Ik heb het aan de lijst toegevoegd voor toekomstige releases. 
Alleen weet ik niet perse of het wat toevoegt, omdat je als gebruiker bij een hoog kwartiervermogen eigenlijk al te laat bent. Het is dan meer een constatering dan een waarschuwing, afhankelijk van of het volgende kwartier nog hoger wordt. Valt nog wel over te brainstormen, maar ik zet het sowieso onderaan de lijst.

Glodenox, 2022-10-01:
De voornaamste use case lijkt me deze data te gebruiken als waarschuwing wanneer je in de huidige periode boven een vooraf ingestelde waarde gaat. Met een eigen script zou je dan een grote verbruiker kunnen uitschakelen om toch nog onder je limiet te blijven.

TheMClaeys, 2022-11-08:
Bedankt voor jullie werk hieraan!
Het is sowieso een meerwaarde via MQTT de data binnen te krijgen.
Vanaf januari beginnen ze ermee hier in België met hun kwartierpiekbeleid helaas, de originele planning was afgelopen zomer, dus ik was in januari al bezig iets in Home Assistant te proberen opzetten om er ons bewuster van te maken.

Ik beschrijf even hoe ik momenteel dergelijke data gebruik:
Momenteel hou ik in Home Assistant de hoogste piek bij per maand bij en calculeer zo mijn gemiddelde (met een minimum van 2,5kW). Zo probeer ik mijn gemiddelde zo laag mogelijk te houden op jaarbasis, maar weet ik ook dat ik er minder naar moet kijken als ik al een piek van bvb 4kW gehad heb in de maand (gezien enkel het hoogste kwartier telt).

Ik heb 2 items op men dashboard staan, een live tracker van het huidige kwartier:
[image]

En een tabel met alle pieken tot nu in het afgelopen jaar:
[image]

Bij een piek over 2,3kW krijgen we al een waarschuwing op Telegram (dan valt er misschien nog iets aan te doen, meestal niet). Aan het eind van het kwartier slaat hij deze waarde op voor deze maand. We krijgen dan geen waarschuwing meer zolang we onder die waarde blijven (want het hoogste kwartier telt voor een maand). Is de waarde toch eens hoger, dan krijgen we terug een waarschuwing en zal hij die maand overschrijven met een hogere waarde. Op die manier weten we ongeveer welke toestel combo's niet goed zijn voor een piek bvb.

Maar mijn huidige implementatie is geprul en buggy (overschrijft soms toch lagere waarde in de maand bvb, die trek ik dan achteraf met de cijfers van Fluvius recht). Via DSMR zou het betrouwbaarder zijn en de implementatie vereenvoudigen.

_(moest deze info het topic vervuilen, geef gerust een seintje dan delete ik het weer)._

dennissiemensma, 2022-11-08:
Bedankt voor je aanvulling. Uiteindelijk zal elke kwartierpiek via MQTT ontsloten worden, alleen heb je er vermoedelijk weinig aan omdat het dan al "te laat" is. Plus dat het niet 100% nauwkeurig is, maar hooguit een benadering van wat de netbeheerder doet/berekent.

Je kunt nog overwegen om zelf elke 5 minuten een delta te trekken van de meterstanden via MQTT en daar een 5-minuten piek of iets voor te berekenen. In principe is een hoge piek daar een voorbode voor een mogelijk aanstaande hoge kwartierpiek, maar heb je nog tijd om het wellicht omlaag te drukken voordat het kwartier afgelopen is.

dennissiemensma, 2022-11-28:
Voor de volgende v5.9 release neem ik weer een (klein deel) mee. Al klaar:
- Uitbreiding REST API voor het opvragen van de "quarter-hour peak electricity consumption" data.

Nog te doen:
- Uitbreiding MQTT voor het doorsturen van diezelfde "quarter-hour peak electricity consumption" data in JSON-formaat en split-topic.

_Alle andere zaken zijn naar verwachting veel meer werk omdat die ook de bestaande datastructuren, MQTT en tests raken._

dennissiemensma, 2022-11-29:
Het MQTT-gedeelte is nu ook klaar. Werkt volgens dezelfde structuur als de andere entiteiten die via MQTT ontsloten worden:

[image]

----

[image]

----

[image]

----

Het is dus de starttijd, eindtijd en het berekende vermogen, zoals ook terug te vinden is in de admin data en grafieken.

dennissiemensma, 2022-11-29:
Ik zie nog wel een potentieel issue met afronding, `0.1695067264573991074882997054` in mijn live voorbeeld, dus ik check even of ik die gewoon kan afronden/ophalen van database.

Het moment van verwerken in MQTT is namelijk na opslaan in de database, maar het is blijkbaar nog de rauwe waarde.

dennissiemensma, 2022-11-29:
Fix gemaakt. Ik zal het allemaal even schaduwdraaien, samen met de rest van de komende v5.9-release.
Als er verder geen issues uit komen dan hoop ik het donderdag- of vrijdagavond te releasen als onderdeel van v5.9.

De resterende TODO's [in het lijstje bovenaan](https://github.com/dsmrreader/dsmr-reader/issues/1635#issue-1245626307) schuiven weer door.

TheMClaeys, 2022-11-30:
> Bedankt voor je aanvulling. Uiteindelijk zal elke kwartierpiek via MQTT ontsloten worden, alleen heb je er vermoedelijk weinig aan omdat het dan al "te laat" is. Plus dat het niet 100% nauwkeurig is, maar hooguit een benadering van wat de netbeheerder doet/berekent.

Je kan 11 maanden bijsturen na een piek, dus het heeft zeker nut, vandaar dat ik ze per maand bij hou en het gemiddelde :) . En van de netbeheerder krijg je ze maar na het eind van de maand in de portal, dus dat heeft tijdens de maand geen nut.

Bedankt voor de moeite en de mooie release die er aan komt!

Tommatheussen, 2022-12-07:
Ik zit in een Facebook groep waar mensen nu updates hebben doorgekregen aan hun DSMR meter, waarbij sommige waarden ivm het piekvermogen kunnen uitgelezen (ipv berekend) worden. 
Een geupdated versie van de protocol beschrijving staat hier:

https://maakjemeterslim.be/rails/active_storage/disk/eyJfcmFpbHMiOnsibWVzc2FnZSI6IkJBaDdDRG9JYTJWNVNTSWhlblJ4YW14cU5XdHRhalZsWkRWbmJIaDNNak4zYmpReGFUazRkd1k2QmtWVU9oQmthWE53YjNOcGRHbHZia2tpVzJGMGRHRmphRzFsYm5RN0lHWnBiR1Z1WVcxbFBTSmxMVTFWUTFOZlVERmZSV1JmTVY4M1h6RXVjR1JtSWpzZ1ptbHNaVzVoYldVcVBWVlVSaTA0SnlkbExVMVZRMU5mVURGZlJXUmZNVjgzWHpFdWNHUm1CanNHVkRvUlkyOXVkR1Z1ZEY5MGVYQmxTU0lVWVhCd2JHbGpZWFJwYjI0dmNHUm1CanNHVkE9PSIsImV4cCI6IjIwMjItMTItMDdUMDg6NDc6NDguNzQxWiIsInB1ciI6ImJsb2Jfa2V5In19--ea338f9bb4fc2bdc0f1d3fdd36e9fe7d4ee5d21c/e-MUCS_P1_Ed_1_7_1.pdf?content_type=application%2Fpdf&disposition=attachment%3B+filename%3D%22e-MUCS_P1_Ed_1_7_1.pdf%22%3B+filename%2A%3DUTF-8%27%27e-MUCS_P1_Ed_1_7_1.pdf

Indien er besloten wordt om deze  waarden in te bouwen in DSMR reader, zou ik toch willen vragen om een fallback te doen op de berekende waarde, aangezien sommige meters geen updates krijgen. Zoals mijn DSMR meter, die kan geen verbinding maken met buitenaf, ik sta dus nog een aantal firmware versies achter...

Glodenox, 2022-12-07:
Ik zie inderdaad deze nieuwe waarden nu al, als ik de raw telegram data bekijk (de pijltjes zijn zelf toegevoegd):
```
/FLU5\253769484_A

0-0:96.1.4(50217)
0-0:96.1.1(3153414733313030303338343634)
0-0:1.0.0(221207100448W)
1-0:1.8.1(014926.618*kWh)
1-0:1.8.2(017371.859*kWh)
1-0:2.8.1(003824.090*kWh)
1-0:2.8.2(001503.592*kWh)
0-0:96.14.0(0001)
-> 1-0:1.4.0(00.644*kW)
-> 1-0:1.6.0(221206221500W)(07.950*kW)
-> 0-0:98.1.0(0)(1-0:1.6.0)(1-0:1.6.0)()
1-0:1.7.0(01.848*kW)
1-0:2.7.0(00.000*kW)
1-0:21.7.0(00.198*kW)
1-0:41.7.0(00.263*kW)
1-0:61.7.0(01.386*kW)
1-0:22.7.0(00.000*kW)
1-0:42.7.0(00.000*kW)
1-0:62.7.0(00.000*kW)
1-0:32.7.0(236.8*V)
1-0:52.7.0(235.5*V)
1-0:72.7.0(232.0*V)
1-0:31.7.0(001.21*A)
1-0:51.7.0(001.59*A)
1-0:71.7.0(006.42*A)
0-0:96.3.10(1)
0-0:17.0.0(999.9*kW)
1-0:31.4.0(999*A)
0-0:96.13.0()
!B4B3
```

Aangezien ik de PDF niet kon downloaden van die locatie, heb ik het elders gevonden en als attachment hier toegevoegd: [e-MUCS_P1_Ed_1_7_1.pdf](https://github.com/dsmrreader/dsmr-reader/files/10174692/e-MUCS_P1_Ed_1_7_1.pdf)

Persoonlijk heb ik in elk geval liever de officiële waarden die de meter aangeeft. Een fallback voor zij op een lagere versie zou inderdaad mooi zijn, aangezien de logica nu toch al bestaat.

VBP8501, 2022-12-07:
Ik zie die nieuwe lijnen ook in mijn ruwe telegrammen

dennissiemensma, 2022-12-07:
@Tommatheussen @Glodenox @VBP8501 dank voor de aanvulling. Ik heb het even verplaatst naar #1764.

Het heeft wel deels te maken met het huidige topic, maar het heeft dusdanig impact dat ik het even los trek. Ook voor het overzicht.

dennissiemensma, 2022-12-07:
Het huidige issue gaat qua resterende TODO's even ON HOLD. Totdat er een plan in #1764 is.

closed 2022-12-08 by dennissiemensma