dsmrreader/dsmr-reader #1639: 💡MQTT - contract gegevens versturen

Diondk, 2022-06-06:
### Feature

Ik ben bezig om in home assistant mijn saldering en verbruik te berekenen. Nu met de stijgende energie prijzen krijg ik elke paar maanden een nieuw contract met hogere tarieven helaas. 

Nu zou het heel fijn zijn als ik de contract gegevens via mqtt kan versturen naar home assistant. zodat ik per contract de saldering kan berekenen, ik heb namelijk nogal wat overschot van mijn panelen. 

Ik zit zelf te denken aan een simpele checkmark bij je energie contract, verzend via mqtt en dat je dan de sensor naam kan instellen. 

waardoor je dus sensoren zou krijgen in HA:

sensor.dsmr_contract_naam_low_tariff_usage
sensor.dsmr_contract_naam_high_tariff_usage
etc

Groet,
Dion

dennissiemensma, 2022-06-06:
Bedankt voor je verzoek. Je kunt proberen de huidige prijzen van de dag pakken uit MQTT en daarmee rekenen. Deze zijn onderdeel van `/admin/dsmr_mqtt/jsondaytotalsmqttsettings/` in DSMR-reader.

Ik denk dat het wat onpraktisch is als DSMR-reader alles gaat spiegelen, want de contracten zijn puur een indicatie en ter vergelijking van dagen onderling. 
Helemaal als je er echt mee gaat rekenen, denk ik dat je zoiets beter bij HA zelf kan leggen (mochten daar nog geen contracten in zitten). Want dat sluit vermoedelijk wat beter aan op het Energy Dashboard en de mogelijke ontwikkelingen daarin.

closed 2022-06-06 by dennissiemensma

Diondk, 2022-06-06:
Hoi, 

Bedankt voor je antwoord, het gaat mij echter niet om de prijzen die bij de contracten staan van dsmr, maar om het verbruik tijdens zo'n contract. Juist omdat dsmr geen ondersteuning heeft voor het salderen en deze waarschijnlijk ook niet gaat komen ivm de moeilijke berekeningen die daar mee gepaard gaan wil ik dit in HA gaan doen. 

Het engiste wat ik dus moet hebben vanuit dsmr naar HA zijn de totale gegevens van een contract deze zoals hier onder dus.

[image]

HA heeft ook al aangegeven dat ze niks gaan bouwen voor het salderen omdat dit ook te complex is.

dennissiemensma, 2022-06-07:
Bedankt voor je aanvulling. Wat je nodig hebt heeft DSMR-reader niet direct beschikbaar. Die gegevens zijn een intepretatie van de gegevens, maar niet kant en klaar beschikbaar. 
En omdat het puur een visuele opsomming is, zit er ook geen trigger of iets achter wat makkelijk mapt naar MQTT.

Ik zie echter dat de DSMR-reader API nog geen ondersteuning heeft voor het ophalen van de energiecontracten (zoals je de invoert). Als ik dat voor je toevoeg dan kun je relatief makkelijk hetzelfde gedrag simuleren wat DSMR-reader ook doet:
- Ophalen alle contracten met bijbehorende looptijd
- Alle dagtotalen van die looptijd ophalen
- Alle resultaten optellen

Je hebt dan volgens mij zowel de prijsinformatie als totaalverbruik/teruglevering en dan kun je ermee doen wat je maar wilt.

Heel veel meer kan ik niet voor je betekenen, omdat ik wat weg probeer te blijven van specifieke oplossingen voor specifieke situaties. Ik beperkt het dan tot het ontsluiten van de generieke datamodellen/collecties via de API.

dennissiemensma, 2022-06-07:
Ik heb #1640 aangemaakt voor het nieuwe endpoint.

dennissiemensma, 2022-06-17:
Achteraf dubbel met #1392. Ik zal verwijzen naar deze en #1640.