dsmrreader/dsmr-reader #1643: 🚨 Deleting a large amount of MQTT messages causes a 400 Reply

dekimpew, 2022-06-10:
### Description

I'm trying to remove a huge amount of mqtt messages that are pending on dsmr because the mqtt broker is/was offline. When selecting all of them (5), the page returns a **Bad Request (400)** instead of deleting the messages.

Deleting messages in batch (100) works fine.

### DSMR-reader version

5.1

### DSMR-reader platform

Native (e.g. manual installation)

### Debug info dump

```shell
DSMR-READER
    App / Python / Database                                                        v5.1 / v3.9.2 / postgresql
    BE sleep / DL sleep / Retention / Override                                     1.0s / 5.0s / 672h / False
    Latest telegram version read / Parser settings                                             "None" / "101"

DATA
    Telegrams total (est.)                                                                            2343294
    Consumption records electricity / gas (est.)                                                46640 / 15179

UNRESOLVED ISSUES
    Too many outgoing MQTT messages queued for transit                                                    now

POSTGRESQL SIZE OF LARGEST TABLES (> 500 MB)
    public.dsmr_datalogger_dsmrreading                                                                 616 MB
```

dennissiemensma, 2022-06-10:
Thanks for reporting. If the admin interface cannot handle thousands of messages, it's something I cannot fix, since it's related to Django framework. However, you can wipe them all in one command using the database. Do you still need support or did you finish purging manually already?

dekimpew, 2022-06-10:
I have not. Having the command will be useful for future references also.

dennissiemensma, 2022-06-10:
Try this:

```
sudo su - postgres
psql dsmrreader

delete from dsmr_mqtt_message;
```

It may take a few seconds, depending on the amount of data to delete (and hardware).

dekimpew, 2022-06-10:
Thanks! That worked instantaneously. I assume this one can be closed then?

dennissiemensma, 2022-06-10:
Great, sure!

closed 2022-06-13 by dennissiemensma