dsmrreader/dsmr-reader #1644: PVoutput net power calculation💡

rmeerkat, 2022-06-14:
### Feature

In dsmr_pvoutput/services.py wordt de v4 waarde 'net power' berekend.

`net_power = last.currently_delivered - last.currently_returned`

Ik vraag me af of dat wel klopt. Moet dat niet de gemiddelde waarde zijn over de interval periode? In mijn geval dus het gemiddelde van 5 minuten. In principe moet je alle net power waarden kunnen optellen en dat moet dan overeen komen met het verbruik van die dag. Dat is volgens mij nu niet zo.

dennissiemensma, 2022-06-14:
Bedankt voor je vraag. Die berekening [zit er inmiddels 5 jaar in](https://github.com/dsmrreader/dsmr-reader/blame/a4e96fb3781052751ae7a76202115b45ff810434/dsmr_pvoutput/services.py#L84) en is [zeer uitgebreid besproken in #9](https://github.com/dsmrreader/dsmr-reader/issues/9#issuecomment-335003057).

Ik zie zo tussendoor dat [het PVOutput endpoint inmiddels veel meer parameters ondersteunt](https://pvoutput.org/help/api_specification.html#add-status-service), maar de kern lijkt hetzelfde als destijds. Ik zie niets over gemiddelden, alleen harde import/export in Watt, wat overigens altijd een momentopname is. Zie evt ook de Net Data tabel daaronder.

Bedoel je wellicht iets anders?

rmeerkat, 2022-06-14:
Die thread heb ik inderdaad helemaal doorgelezen en in de pvoutput documentatie staat inderdaad niets over gemiddelden.

Na de zoveelste SD card crash ben ik maar eens gaan kijken naar een alternatief en kwam bij dsmr reader...
De afgelopen 10 jaar heb ik met mijn eigen P1 software gelogd naar pvoutput. Daar gebruikte ik de gemiddelde waarde en dat zag er goed uit. Ik kon bijvoorbeeld iedere paar uur een piek zien van de plintboiler.

Nu is de 'Power used' grafiek erg 'grillig' bijna random. In [deze](https://forum.pvoutput.org/t/need-clarification-on-v1-v4-parameters/197/6) thread staat het uitgelegd.

Als ik het goed begrijp is het misschien zelfs beter om helemaal geen v4 te sturen. Dan berekend pvoutput het zelf aan de hand van v3.

dennissiemensma, 2022-06-14:
Bedankt voor je aanvulling. Ik heb zelf geen zonnepanelen en gebruik ook geen PVOutput. Daarom is het lastig om hier zomaar wijzigingen in te doen. Helemaal als het al zolang op deze manier gebruikt wordt zonder meldingen daarover.

Als er veel animo voor is dan kan ik het ooit aanpassen, maar je kunt ook overwegen om het er lokaal uit te halen via een patch of iets. 
Overigens is PVOutput slechts een zeer klein deel in DSMR-reader, dus als je specifiek iets voor het uploaden zoekt, kun je wellicht beter een scriptje maken met [dsmr_parser](https://github.com/ndokter/dsmr_parser) en het vervolgens ook uploaden in datzelfde scriptje. Dan zit je ook niet met de rest van DSMR-reader wat je wel of niet gebruikt.

rmeerkat, 2022-06-14:
Ik zal vanavond die v4 er eens uit halen en kijken hoe het er morgen uit ziet. Dat is de meest simpele oplossing denk ik.

dennissiemensma, 2022-06-14:
Als je de waarde van 5 minuten geleden pakt, zit je denk ik met hetzelfde. Namelijk dat het niet perse representatief is. Als je het echte gemiddelde wilt weten, kun je dat het beste opvragen in de database over alle metingen in de periode. 

Echter is dat niet heel triviaal. Ook omdat de betreffende code voor de PVOutput-upload behoorlijk wat condities en checks heeft. Een voorbeeld van gemiddelde pakken op een andere plek staat hier: https://github.com/dsmrreader/dsmr-reader/blob/a4e96fb3781052751ae7a76202115b45ff810434/dsmr_consumption/services.py#L120

dennissiemensma, 2022-06-17:
Ik laat dit voor nu even voor wat het is. Mocht het echt een issue zijn, dan kan het altijd nog aangepast worden. Helemaal als er onder meerdere gebruikers hier de behoefte aan is.

closed 2022-06-17 by dennissiemensma