dsmrreader/dsmr-reader #1645: 🙋

poudenes, 2022-06-14:
### Description

Hoi Allen,

Ik heb mijn backup van DSMR (gemaakt in postgres/timescale) in een postgres dB terug gezet. 
Alles ging goed. DSMR gestart en alle oude waardes en archief waren terug.

Echter zie ik nu een heleboel telegram regels die niet verwerkt zijn:
```
Processed: False
Geeft aan of deze meting al verwerkt is in het samenvoegen van statistieken
```
Timegap van de backup en start nieuwe dB
```
[14 juni 2022 11:31] - False	7281.856	7717.392	     2453.176
[13 juni 2022 15:12] - True	7279.459	7713.720.    2452.711
```

### DSMR-reader version

v5.2

### Debug info dump

```shell
2022-06-14 09:49:37,141 DEBUG    broker       on_log                          159 | MQTT: (Paho on_log) Sending PUBREL (Mid: 294)
2022-06-14 09:49:37,142 DEBUG    broker       run                              98 | MQTT: Waiting for message (#240557254) to be marked published by broker
2022-06-14 09:49:37,143 DEBUG    broker       run                              98 | MQTT: Waiting for message (#240557254) to be marked published by broker
2022-06-14 09:49:37,144 DEBUG    broker       on_log                          159 | MQTT: (Paho on_log) Received PUBCOMP (Mid: 294)
2022-06-14 09:49:37,144 DEBUG    broker       run                             106 | MQTT: Deleting published message (#240557254) from queue
2022-06-14 09:49:37,150 DEBUG    broker       run                              79 | MQTT: Publishing queued message (#240557253) for dsmr/reading/phase_currently_delivered_l3: None
2022-06-14 09:49:37,151 DEBUG    broker       on_log                          159 | MQTT: (Paho on_log) Sending PUBLISH (d0, q2, r1, m295), 'b'dsmr/reading/phase_currently_delivered_l3'' (NULL payload)
2022-06-14 09:49:37,152 DEBUG    broker       run                              98 | MQTT: Waiting for message (#240557253) to be marked published by broker
2022-06-14 09:49:37,154 DEBUG    broker       on_log                          159 | MQTT: (Paho on_log) Received PUBREC (Mid: 295)
2022-06-14 09:49:37,154 DEBUG    broker       on_log                          159 | MQTT: (Paho on_log) Sending PUBREL (Mid: 295)
2022-06-14 09:49:37,155 DEBUG    broker       run                              98 | MQTT: Waiting for message (#240557253) to be marked published by broker
2022-06-14 09:49:37,156 DEBUG    broker       run                              98 | MQTT: Waiting for message (#240557253) to be marked published by broker
2022-06-14 09:49:37,157 DEBUG    broker       on_log                          159 | MQTT: (Paho on_log) Received PUBCOMP (Mid: 295)
2022-06-14 09:49:37,158 DEBUG    broker       run                             106 | MQTT: Deleting published message (#240557253) from queue
2022-06-14 09:49:37,164 DEBUG    broker       run                              79 | MQTT: Publishing queued message (#240557252) for dsmr/reading/phase_currently_delivered_l2: None
2022-06-14 09:49:37,165 DEBUG    broker       on_log                          159 | MQTT: (Paho on_log) Sending PUBLISH (d0, q2, r1, m296), 'b'dsmr/reading/phase_currently_delivered_l2'' (NULL payload)
2022-06-14 09:49:37,166 DEBUG    broker       run                              98 | MQTT: Waiting for message (#240557252) to be marked published by broker
2022-06-14 09:49:37,167 DEBUG    broker       on_log                          159 | MQTT: (Paho on_log) Received PUBREC (Mid: 296)
2022-06-14 09:49:37,168 DEBUG    broker       on_log                          159 | MQTT: (Paho on_log) Sending PUBREL (Mid: 296)
2022-06-14 09:49:37,169 DEBUG    broker       run                              98 | MQTT: Waiting for message (#240557252) to be marked published by broker
2022-06-14 09:49:37,169 DEBUG    broker       run                              98 | MQTT: Waiting for message (#240557252) to be marked published by broker
2022-06-14 09:49:37,170 DEBUG    broker       on_log                          159 | MQTT: (Paho on_log) Received PUBCOMP (Mid: 296)
2022-06-14 09:49:37,171 DEBUG    broker       run                             106 | MQTT: Deleting published message (#240557252) from queue
2022-06-14 09:49:37,177 DEBUG    broker       run                              79 | MQTT: Publishing queued message (#240557251) for dsmr/reading/phase_currently_delivered_l1: 0.155
2022-06-14 09:49:37,178 DEBUG    broker       on_log                          159 | MQTT: (Paho on_log) Sending PUBLISH (d0, q2, r1, m297), 'b'dsmr/reading/phase_currently_delivered_l1'', ... (5 bytes)
2022-06-14 09:49:37,179 DEBUG    broker       run                              98 | MQTT: Waiting for message (#240557251) to be marked published by broker
2022-06-14 09:49:37,180 DEBUG    broker       on_log                          159 | MQTT: (Paho on_log) Received PUBREC (Mid: 297)
2022-06-14 09:49:37,180 DEBUG    broker       on_log                          159 | MQTT: (Paho on_log) Sending PUBREL (Mid: 297)
2022-06-14 09:49:37,181 DEBUG    broker       run                              98 | MQTT: Waiting for message (#240557251) to be marked published by broker
2022-06-14 09:49:37,182 DEBUG    broker       run                              98 | MQTT: Waiting for message (#240557251) to be marked published by broker
2022-06-14 09:49:37,183 DEBUG    broker       on_log                          159 | MQTT: (Paho on_log) Received PUBCOMP (Mid: 297)
2022-06-14 09:49:37,184 DEBUG    broker       run                             106 | MQTT: Deleting published message (#240557251) from queue
2022-06-14 09:49:37,191 DEBUG    broker       run                              79 | MQTT: Publishing queued message (#240557250) for dsmr/reading/electricity_currently_returned: 0.000
2022-06-14 09:49:37,194 DEBUG    broker       on_log                          159 | MQTT: (Paho on_log) Sending PUBLISH (d0, q2, r1, m298), 'b'dsmr/reading/electricity_currently_returned'', ... (5 bytes)
2022-06-14 09:49:37,197 DEBUG    broker       run                              98 | MQTT: Waiting for message (#240557250) to be marked published by broker
2022-06-14 09:49:37,198 DEBUG    broker       on_log                          159 | MQTT: (Paho on_log) Received PUBREC (Mid: 298)
2022-06-14 09:49:37,199 DEBUG    broker       on_log                          159 | MQTT: (Paho on_log) Sending PUBREL (Mid: 298)
2022-06-14 09:49:37,200 DEBUG    broker       run                              98 | MQTT: Waiting for message (#240557250) to be marked published by broker
2022-06-14 09:49:37,200 DEBUG    broker       run                              98 | MQTT: Waiting for message (#240557250) to be marked published by broker
2022-06-14 09:49:37,203 DEBUG    broker       on_log                          159 | MQTT: (Paho on_log) Received PUBCOMP (Mid: 298)
2022-06-14 09:49:37,204 DEBUG    broker       run                             106 | MQTT: Deleting published message (#240557250) from queue
2022-06-14 09:49:37,212 DEBUG    broker       run                              79 | MQTT: Publishing queued message (#240557249) for dsmr/reading/electricity_currently_delivered: 0.155
2022-06-14 09:49:37,212 DEBUG    broker       on_log                          159 | MQTT: (Paho on_log) Sending PUBLISH (d0, q2, r1, m299), 'b'dsmr/reading/electricity_currently_delivered'', ... (5 bytes)
2022-06-14 09:49:37,213 DEBUG    broker       run                              98 | MQTT: Waiting for message (#240557249) to be marked published by broker
2022-06-14 09:49:37,215 DEBUG    broker       on_log                          159 | MQTT: (Paho on_log) Received PUBREC (Mid: 299)
[... 32 lines cut ...]
2022-06-14 09:49:37,257 DEBUG    broker       run                              79 | MQTT: Publishing queued message (#240557245) for dsmr/reading/electricity_delivered_1: 7281.856
2022-06-14 09:49:37,258 DEBUG    broker       on_log                          159 | MQTT: (Paho on_log) Sending PUBLISH (d0, q2, r1, m303), 'b'dsmr/reading/electricity_delivered_1'', ... (8 bytes)
2022-06-14 09:49:37,259 DEBUG    broker       run                              98 | MQTT: Waiting for message (#240557245) to be marked published by broker
2022-06-14 09:49:37,260 DEBUG    broker       on_log                          159 | MQTT: (Paho on_log) Received PUBREC (Mid: 303)
2022-06-14 09:49:37,261 DEBUG    broker       on_log                          159 | MQTT: (Paho on_log) Sending PUBREL (Mid: 303)
[... cut ...]
2022-06-14 09:49:38,321 DEBUG    persistent_clients run                              31 | CLIENTS: Running 1 active client(s)
2022-06-14 09:49:38,332 DEBUG    mixins       run_loop                         75 | dsmr_backend.management.commands.dsmr_backend: Sleeping 1.0s
2022-06-14 09:49:39,340 DEBUG    schedule     execute_scheduled_processes      32 | SP: 0 backend service(s) ready to run
2022-06-14 09:49:39,355 DEBUG    persistent_clients run                              31 | CLIENTS: Running 1 active client(s)
```

dennissiemensma, 2022-06-14:
@poudenes wat is je debug info dump? Niet te verwarren met de debug log. Deze kun je inzien op de "Over & ondersteuning" pagina.

Bijvoorbeeld:
```

DSMR-READER
    App / Python / Database                                                        v5.3 / v3.9.9 / postgresql
    BE sleep / DL sleep / Retention / Override                                     5.0s / 0.5s / 168h / False
    Latest telegram version read / Parser settings                                                 "42" / "4"

DATA
    Telegrams total (est.)                                                                             176472
    Consumption records electricity / gas (est.)                                               122267 / 49233
```

poudenes, 2022-06-14:
Sorry bij deze:

DSMR-READER
    App / Python / Database                                                       v5.2 / v3.10.4 / postgresql
    BE sleep / DL sleep / Retention / Override                                     1.0s / 5.0s / 168h / False
    Latest telegram version read / Parser settings                                                 "50" / "4"

DATA
    Telegrams total (est.)                                                                             136299
    Consumption records electricity / gas (est.)                                                46508 / 38466

UNRESOLVED ISSUES
    Dagstatistieken lopen achter                                                         1 dag, 9 uur geleden

dennissiemensma, 2022-06-14:
Ik zie in je logs niet de melding `null value in column`, vanwaar dat je bij #1619 uit kwam?

Daarnaast zie ik ook niet bij je dump een melding als `Too many unprocessed readings`. Kan het zijn dat de achterstand inmiddels is ingelopen? 

Anders zul je in de DB moeten kijken vanaf welk moment de verwerking gestopt is. 

Bijvoorbeeld met deze query:
```
SELECT id, timestamp, electricity_delivered_1, electricity_delivered_2 FROM dsmr_datalogger_dsmrreading WHERE processed = false ORDER BY timestamp ASC;
```

dennissiemensma, 2022-06-14:
Ik vermoed dat je issues hebt met de systeemtijd of iets. De melding "(over een uur)" zou moeten aangeven wanneer voor het laatst een telegram is verwerkt. Die zou eigenlijk nooit in de toekomst moeten staan.

Draaide je hiervoor dezelfde Docker setup/versies van de DB/DSMR-reader?

dennissiemensma, 2022-06-14:
Was je Docker-setup daar hetzelfde? Want dit neigt naar iets als #1282

poudenes, 2022-06-14:
```
NAS:
postgres@postgres:/$ psql dsmr -c 'select timestamp from dsmr_datalogger_dsmrreading order by timestamp desc limit 5;'
       timestamp        
------------------------
 2022-06-14 10:32:13+02
 2022-06-14 10:32:07+02
 2022-06-14 10:32:01+02
 2022-06-14 10:31:55+02
 2022-06-14 10:31:49+02
(5 rows)

RPi4:
postgres@77b2833f-timescaledb:/$ psql dsmr -c 'select timestamp from dsmr_datalogger_dsmrreading order by timestamp desc limit 5;'
       timestamp        
------------------------
 2022-06-14 07:47:06+00
 2022-06-14 07:47:00+00
 2022-06-14 07:46:54+00
 2022-06-14 07:46:48+00
 2022-06-14 07:46:42+00
(5 rows)
```

Ik zie wel een verschil in timestamp.

dennissiemensma, 2022-06-14:
Het hangt er vanaf wat je met elkaar vergelijkt. Die `+02` en `+00` is erg belangrijk, want die moet je erbij nemen. 

Bij de +2 zul je dat in aantal uren eraf moeten trekken om in de UTC (+00) tijdzone te zitten. Of andersom, bij de +00's twee uur optellen om in onze zomertijd te komen.

Je kunt de oplossing in #1282 checken of je van hetzelfde last hebt.

poudenes, 2022-06-14:
Wat ik heb gedaan is 

/etc/localtime:/etc/localtime:ro

verwijderd uit docker-compose nu is NAS gelijk aan RPi

root@postgres:/# date
Tue 14 Jun 2022 10:42:40 AM CEST
root@postgres:/# su postgres
postgres@postgres:/$ psql dsmr -c 'select timestamp from dsmr_datalogger_dsmrreading order by timestamp desc limit 5;'
       timestamp        
------------------------
 2022-06-14 08:42:31+00
 2022-06-14 08:42:25+00
 2022-06-14 08:42:19+00
 2022-06-14 08:42:13+00
 2022-06-14 08:42:07+00
(5 rows)

poudenes, 2022-06-14:
Ik open Dashboard en zie wel ineens dat alle errors weg zijn. Of dit nu voor of na mijn laatste actie is weet ik niet

dennissiemensma, 2022-06-14:
Dat is gegarandeerd door je wijziging.

Deze tijden zijn overigens hetzelfde:
```
Tue 14 Jun 2022 10:42:40 AM CEST
2022-06-14 08:42:31+00
```

CEST = zomertijd, in Nederland +2 tov UTC (+00)

closed 2022-06-14 by dennissiemensma

poudenes, 2022-06-14:
Ja ik zie onverwerkte telegrams maar op gegeven moment verdwijnen ze.

DANK voor de support !!!

dennissiemensma, 2022-06-14:
Als je de (standaard) "groeperen per minuut" aan hebt staan, worden ze pas de volgende minuut verwerkt.