dsmrreader/dsmr-reader #1646: 💡 tijdstip van MinderGas upload

mdr-github, 2022-06-14:
### Feature

Hallo Dennis, MinderGas.nl ("MG") heeft een feature toegevoegd. Je kunt daar nu instellen of de ingevoerde meterstand de begin- of eindstand van de dag is. Ik krijg bij die site de volgende melding te zien:

> Begin- of eindstand? Het is nu mogelijk om [in te stellen](https://www.mindergas.nl/member/settings/edit) of jouw meterstand moet worden gezien als begin- (00h00m00s) of eindstand (23h59m59s) voor de gekozen datum. Voorheen werd de meterstand altijd gezien als eindstand. Voor bestaande accounts is 'eind van de dag' ingesteld (zoals voorheen). Tijdstippen van bestaande meterstanden zijn onveranderd gebleven.

Omdat DSMR het gasverbruik dagelijks doorgeeft en MG dat weer gebruikt voor het rekenen met graaddagen, is het wel relevant dat de meting aan de juiste dag wordt gekoppeld. Mijn CV is bijvoorbeeld weersafhankelijk geregeld, dus er is een directe tussen gasverbruik en buitentemperatuur. Daarom wilde ik controleren wanneer DSMR de upload doet.

Welke tijd instelling gebruikt DSMR hiervoor, en kunnen we hier rekening mee houden? 

Bij de geplande processen zie ik in mijn installatie nu: "Upload gas meter position to MinderGas.nl | over 9 uur | 15 juni 2022 05:52". Ik weet alleen niet welke stand DSMR dan doorgeeft. De actuele van dat moment? Of de eindstand van de dag lijkt me logischer, ik kan mij voorstellen dat 5:52u een random tijd is om de server load bij MG beheersbaar te houden. 

Dan lijkt het me handig te weten maken welke stand DSMR upload naar MG (dus begin- (00h00m00s) of eindstand (23h59m59s) voor de gekozen datum), en in de docs aan te geven welke instelling gebruikers moeten kiezen bij MinderGas.

Dank en groet, Marc

dennissiemensma, 2022-06-14:
Bedankt voor je melding! DSMR-reader gebruikt de [eindstand van de voorgaande dag](https://github.com/dsmrreader/dsmr-reader/blob/v5/dsmr_mindergas/services.py#L69). 

Het tijdstip van uploaden maakt inderdaad niet zoveel uit, DSMR-reader kiest een [willekeurige tijd](https://github.com/dsmrreader/dsmr-reader/blob/v5/dsmr_mindergas/services.py#L53) om een eventuele load daar te voorkomen, al zal het in de praktijk meevallen (ik heb geen exacte gebruikersaantallen).

Ik zal kijken of ik dit ergens kan vermelden.

mdr-github, 2022-06-14:
Thanks! Dan gaat het al goed, denk dat het dan voldoende is om in de docs erbij te zeggen dat de melding einde dag is en klaar.

dennissiemensma, 2022-06-14:
In het admin scherm wat tekst toegevoegd voor volgende release.

closed 2022-06-14 by dennissiemensma