dsmrreader/dsmr-reader #1648: 🙋 DSMR maakt geen backups meer van data

Roel0412, 2022-06-21:
### Description

Vorige week is mijn SD kaart kapot gegaan. Nu heb ik alles opnieuw weten te installeren op een SSD schijf.

Nu heb ik alleen het probleem dat er geen backups meer gemaakt worden van de database naar de USB stick.

[image]

De laatste backup is van 15 juni.

Alles staat goed ingesteld. De USB stick is ook gemount en te benaderen. Ook heb ik de rechten er goed op staan. 

Heb je nog tips wat ik kan checken?

### DSMR-reader version

5.3

### DSMR-reader platform

Native (e.g. manual installation)

dennissiemensma, 2022-06-21:
Je kunt in `/admin/dsmr_backend/scheduledprocess/` het `Create daily backup` proces aanklikken en **Today** + **Now** aanklikken en opslaan. Als het goed is maakt die dan een backup. Mogelijk zie je wat in de logs.

[image]

[image]

----

Je kunt ook handmatig proberen:

```
sudo su - dsmr
./manage.py dsmr_backup_create --compact
```

Je krijgt dan iets als:
```
(.venv) dsmr@rpi4:~/dsmr-reader $ ./manage.py dsmr_backup_create --compact
Created partial backup: /home/dsmr/dsmr-reader/backups/manually/dsmrreader-postgresql-partial-backup-2022-06-21.sql.gz
```

Dan kun je iig zien of de rechten goed staan.

----

Een mogelijk oorzaak _kan_ zijn dat de client-versie van `psql` niet compatible is met je server `Posgres` versie. Mocht je daar wijzigingen in gedaan hebben.

dennissiemensma, 2022-06-21:
Bij die backup-settings kun je straks de backup herplannen naar "nu" om te simuleren wat die komende nacht gaat doen.

Voor de rechten kun je dit proberen als `pi` user:

```
sudo chown -R dsmr.dsmr /mnt/usb/backups/
```

Roel0412, 2022-06-21:
Ik krijg op alle files in die map Operation not permitted:

[image]

dennissiemensma, 2022-06-21:
Je zult denk ik dan nog even wat moeten debuggen, gezien daar niet 1 oorzaak voor is: https://linuxpip.org/fix-chown-operation-not-permitted/

Roel0412, 2022-06-21:
Is het een optie om de USB stick te formatteren en de backup weer aanzetten?
De data staat nu veilig op de SSD, de oude backups heb ik in principe niet meer nodig volgens mij.
Ik denk dat ie het niet doet, omdat ik het OS enz opnieuw moest installeren op een SSD i.p.v. de kapotte SD kaart.

dennissiemensma, 2022-06-21:
Je kunt de backups ook gewoon op de SSD zelf zetten met de standaardinstellingen. Die slijt als het goed is lang niet zo snel als een SD-kaart.

Eventueel kun je je backups via Dropbox archiveren voor de zekerheid.

Roel0412, 2022-06-21:
Thanks, ik ga d'r ff verder mee rommelen. Ben al wel wat verder gekomen. Heb de backups nu naar de SSD gezet en zal ff kijken voor Dropbox.

closed 2022-06-22 by dennissiemensma