dsmrreader/dsmr-reader #1649: upgrade van 4.x naar 5.x...

newtech2015, 2022-06-22:
### Description

Hallo,

bij de"upgrade stappen" van versie 4.x naar 5.x wordt aangegeven / gewaarschuwd:

    ✋ Upgrade niet wanneer je PostgreSQL 9.x of lager draait. Upgrade PostgreSQL eerst.
    ✋ Upgrade niet wanneer je InfluxDB 1.x draait. Upgrade naar InfluxDB 2.x eerst.
    ✋ Upgrade niet wanneer je MySQL 5.6 of lager draait. Upgrade MySQL eerst.
    ℹ️ Deze upgrade vereist je om Python 3.7 of hoger te draaien. Upgrade-stappen later hieronder.

helaas wordt alleen van python uitgelegd hoe dat kan doen.
via google is het gelukt de PostgreSQL versie te vinden, naar influxDB eb mysql werken de commando's die ik kan vinden allemaal niet.

Enige uitleg gewenst!

Gr ron 

### DSMR-reader version

4.20

### DSMR-reader platform

Other

dennissiemensma, 2022-06-22:
Hoe heb je DSMR-reader geinstalleerd? Je gebruikt waarschijnlijk ofwel Postgres ofwel MySQL, maar niet beiden. Voor Postgres kun je het o.a. in de proceslijst zien:

```
ps faux | grep postgres
```
Ik heb dus versie `11` (`11/main`):
```
pi        7442  0.0  0.0   7352   512 pts/0    S+   14:52   0:00              \_ grep --color=auto postgres
postgres   618  0.0  1.1 204012 22076 ?        S    Jun15   1:15 /usr/lib/postgresql/11/bin/postgres -D /var/lib/postgresql/11/main -c config_file=/etc/postgresql/11/main/postgresql.conf
postgres   620  0.0  3.3 204140 64820 ?        Ss   Jun15   0:45  \_ postgres: 11/main: checkpointer   
postgres   621  0.0  0.4 204012  9312 ?        Ss   Jun15   0:20  \_ postgres: 11/main: background writer   
postgres   622  0.0  0.4 204012  8164 ?        Ss   Jun15   3:46  \_ postgres: 11/main: walwriter   
postgres   623  0.0  0.3 204416  6000 ?        Ss   Jun15   0:24  \_ postgres: 11/main: autovacuum launcher   
postgres   624  0.0  0.1  61272  3480 ?        Ss   Jun15   1:49  \_ postgres: 11/main: stats collector   
postgres   625  0.0  0.2 204312  4752 ?        Ss   Jun15   0:00  \_ postgres: 11/main: logical replication launcher   
postgres  2025  0.0  1.2 205740 24104 ?        Ss   Jun21   0:00  \_ postgres: 11/main: dsmrreader dsmrreader ::1(59036) idle
postgres  2028  0.0  1.0 205636 19364 ?        Ss   Jun21   0:00  \_ postgres: 11/main: dsmrreader dsmrreader ::1(59038) idle
postgres  3220  0.0  0.9 205708 18396 ?        Ss   Jun21   0:00  \_ postgres: 11/main: dsmrreader dsmrreader ::1(59116) idle
postgres  3221  0.0  0.9 205708 19012 ?        Ss   Jun21   0:00  \_ postgres: 11/main: dsmrreader dsmrreader ::1(59118) idle
postgres  7338  0.2  1.0 205672 20320 ?        Ss   14:31   0:02  \_ postgres: 11/main: dsmrreader dsmrreader ::1(59362) idle
postgres  7356  0.1  0.8 205088 16588 ?        Ss   14:36   0:01  \_ postgres: 11/main: dsmrreader dsmrreader ::1(59364) idle
```

Voor InfluxDB kan ik je helaas niet helpen. Het is optioneel voor DSMR-reader, als je het eerder wel gebruikt, dan heb je vermoedelijk versie 1.x. Voor upgraden zie: https://docs.influxdata.com/influxdb/v2.3/upgrade/v1-to-v2/

Het is trouwens altijd een optie om DSMR-reader niet te upgraden, want in de basis verandert er weinig.

newtech2015, 2022-06-23:
Hoi,

ik heb python geupdate, zoals in in handleiding staat.
als ik nu als root user "python3 --version" doe, komt er inderdaad versie 3.9.9. zoals je verwacht

doe ik dit als dsmr user (eerst dus sudo su - dsmr), dan krijg ik alleen mijn oude versie te zien (3.6.9)
gaat dsmr nu na installeren 5.x gewoon 3.9.9 gebruiken, of is dit niet ok?

gr ron

dennissiemensma, 2022-06-23:
Als dat nog je v4 installatie is dan klopt dat. Als het goed is staat verderop in de stappen dat je de zogenaamde virtualenv opnieuw moet aanmaken. Die pakt dan de andere python versie als het goed is.

newtech2015, 2022-06-23:
hoi,

nu opweg met de upgrade, na ./manage.py check krijg ik een hele lijste met fouten.
iets mis met python, schat ik zo(?) enig idee?

/manage.py check
Traceback (most recent call last):
  File "/home/dsmr/.local/lib/python3.9/site-packages/django/db/backends/postgresql/base.py", line 25, in <module>
    import psycopg2 as Database
  File "/home/dsmr/.local/lib/python3.9/site-packages/psycopg2/__init__.py", line 51, in <module>
    from psycopg2._psycopg import (                     # noqa
ImportError: /home/dsmr/.local/lib/python3.9/site-packages/psycopg2/_psycopg.cpython-39-arm-linux-gnueabihf.so: undefined symbol: PQencryptPasswordConn

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/dsmr/dsmr-reader/./manage.py", line 10, in <module>
    execute_from_command_line(sys.argv)
  File "/home/dsmr/.local/lib/python3.9/site-packages/django/core/management/__init__.py", line 419, in execute_from_command_line
    utility.execute()
  File "/home/dsmr/.local/lib/python3.9/site-packages/django/core/management/__init__.py", line 395, in execute
    django.setup()
  File "/home/dsmr/.local/lib/python3.9/site-packages/django/__init__.py", line 24, in setup
    apps.populate(settings.INSTALLED_APPS)
  File "/home/dsmr/.local/lib/python3.9/site-packages/django/apps/registry.py", line 114, in populate
    app_config.import_models()
  File "/home/dsmr/.local/lib/python3.9/site-packages/django/apps/config.py", line 301, in import_models
    self.models_module = import_module(models_module_name)
  File "/usr/local/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 850, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/home/dsmr/.local/lib/python3.9/site-packages/django/contrib/auth/models.py", line 3, in <module>
    from django.contrib.auth.base_user import AbstractBaseUser, BaseUserManager
  File "/home/dsmr/.local/lib/python3.9/site-packages/django/contrib/auth/base_user.py", line 48, in <module>
    class AbstractBaseUser(models.Model):
  File "/home/dsmr/.local/lib/python3.9/site-packages/django/db/models/base.py", line 122, in __new__
    new_class.add_to_class('_meta', Options(meta, app_label))
  File "/home/dsmr/.local/lib/python3.9/site-packages/django/db/models/base.py", line 326, in add_to_class
    value.contribute_to_class(cls, name)
  File "/home/dsmr/.local/lib/python3.9/site-packages/django/db/models/options.py", line 207, in contribute_to_class
    self.db_table = truncate_name(self.db_table, connection.ops.max_name_length())
  File "/home/dsmr/.local/lib/python3.9/site-packages/django/utils/connection.py", line 15, in __getattr__
    return getattr(self._connections[self._alias], item)
  File "/home/dsmr/.local/lib/python3.9/site-packages/django/utils/connection.py", line 62, in __getitem__
    conn = self.create_connection(alias)
  File "/home/dsmr/.local/lib/python3.9/site-packages/django/db/utils.py", line 204, in create_connection
    backend = load_backend(db['ENGINE'])
  File "/home/dsmr/.local/lib/python3.9/site-packages/django/db/utils.py", line 111, in load_backend
    return import_module('%s.base' % backend_name)
  File "/usr/local/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "/home/dsmr/.local/lib/python3.9/site-packages/django/db/backends/postgresql/base.py", line 29, in <module>
    raise ImproperlyConfigured("Error loading psycopg2 module: %s" % e)
django.core.exceptions.ImproperlyConfigured: Error loading psycopg2 module: /home/dsmr/.local/lib/python3.9/site-packages/psycopg2/_psycopg.cpython-39-arm-linux-gnueabihf.so: undefined symbol: PQencryptPasswordConn

newtech2015, 2022-06-23:
zie nu ik verder terug kijk in putty, dat nu commando: pip3 install -r ~/dsmr-reader/dsmrreader/provisioning/requirements/base.txt een aantal regels er normaal uitzien, dan aantal in geel:  (weet niet of dit hiermee te maken heeft):
  WARNING: The scripts pyserial-miniterm and pyserial-ports are installed in '/home/dsmr/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
  WARNING: The script sqlformat is installed in '/home/dsmr/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
  WARNING: The script pycodestyle is installed in '/home/dsmr/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
  WARNING: The script gunicorn is installed in '/home/dsmr/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
  WARNING: The script normalizer is installed in '/home/dsmr/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
  WARNING: The script stone is installed in '/home/dsmr/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
  WARNING: The script django-admin is installed in '/home/dsmr/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
  WARNING: The script autopep8 is installed in '/home/dsmr/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.

dennissiemensma, 2022-06-23:
Wat krijg je te zien met:

```
sudo su - dsmr
pip freeze
```

newtech2015, 2022-06-23:
pip freeze
asgiref==3.5.2
autopep8==1.6.0
certifi==2022.6.15
charset-normalizer==2.0.12
coreapi==2.3.3
coreschema==0.0.4
crcmod==1.7
Django==3.2.13
django-admin-rangefilter==0.8.4
django-admin-sortable==2.3
django-colorfield==0.6.3
django-debug-toolbar==3.4.0
django-filter==21.1
django-mathfilters==1.0.0
django-solo==2.0.0
djangorestframework==3.13.1
dropbox==11.31.0
gunicorn==20.1.0
idna==3.3
influxdb-client==1.29.1
itypes==1.2.0
Jinja2==3.1.2
MarkupSafe==2.1.1
paho-mqtt==1.6.1
Pillow==9.1.1
ply==3.11
psycopg2-binary==2.9.3
pycodestyle==2.8.0
pyserial==3.5
pyserial-asyncio==0.6
python-dateutil==2.8.2
python-decouple==3.6
pytz==2021.3
PyYAML==6.0
requests==2.27.1
Rx==3.2.0
six==1.16.0
sqlparse==0.4.2
stone==3.3.1
toml==0.10.2
uritemplate==4.1.1
urllib3==1.26.9

dennissiemensma, 2022-06-23:
Dat lijkt verder wel goed. Geeft `/.manage.py check` nog steeds een fout?

dennissiemensma, 2022-06-23:
En zie je wel `(.venv)` ervoor staan?
```
(.venv) dsmr@rpi4:~/dsmr-reader $ ./manage.py check
System check identified no issues (0 silenced).
```

newtech2015, 2022-06-23:
nee, geen .venv

nu ik t nog s heel goed lees vanaf t begin, zie ik ook dat ik iets heel belangrijks verkeerd gelezen had: ik heb WEL nog een postSQL9.x... had dit snel gelezen, en dacht dat ik minimaal 9.x moest hebben.

kan het dit zijn? hoe moet ik dit upgraden?

dennissiemensma, 2022-06-23:
Wat toont `lsb_release -a`?

newtech2015, 2022-06-23:
lsb_release -a
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 9.13 (stretch)
Release:        9.13
Codename:       stretch

dennissiemensma, 2022-06-23:
Ik denk dat je het beste een keertje je RaspberryPi opnieuw kan installeren met een nieuw OS (versie 10 of 11). Je draait nu nog een relatief oude versie (9): https://en.wikipedia.org/wiki/Raspberry_Pi_OS#Release_history

Met een nieuwe OS-versie zit er vaak ook nieuwere software bij in, zoals Postgres. En dan kun je direct DSMR-reader v5.x installeren en een DB-backup terugzetten.

Alleen is een herinstallatie iets waar je de tijd voor wilt nemen, dus je kunt ook nog overwegen om voor nu even op DSMR-reader 4 te blijven.

dennissiemensma, 2022-06-23:
Je kunt terug met iets als:

```
sudo su - dsmr
deactivate

rm -rf ~/.virtualenvs/
mv ~/.old-v4-virtualenvs ~/.virtualenvs/ 

git checkout v4
# Controleer of je echt op v4 zit
git branch

# In /home/dsmr/.bashrc vervang je:
# deze regel: source ~/dsmr-reader/.venv/bin/activate
# weer met:
source ~/.virtualenvs/dsmrreader/bin/activate
```

Daarna
```
logout
sudo su - dsmr

python3 --version
./manage.py check
```

De namen van de envvars zouden moeten werken in DSMR-reader 4.20 waar je op zat.

dennissiemensma, 2022-06-23:
Als alles goed lijkt, kun je v4 weer starten met:
```
sudo supervisorctl start all
```

Mits je alle stappen verderop de guide nog niet gedaan had.

newtech2015, 2022-06-23:
ik dacht dus slim te zijn, image van sd kaartje terug gezet. (database staat op externe hdd)

echter, het wil nog steeds niet werken:

Server Error
Sorry, something unexpected happened.
Exception:

OperationalError: could not connect to server: Connection refused
	Is the server running on host "localhost" (127.0.0.1) and accepting
	TCP/IP connections on port 5432?
could not connect to server: Cannot assign requested address
	Is the server running on host "localhost" (::1) and accepting
	TCP/IP connections on port 5432?

Traceback:

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/core/handlers/base.py", line 181, in _get_response
    response = wrapped_callback(request, *callback_args, **callback_kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/views/generic/base.py", line 70, in view
    return self.dispatch(request, *args, **kwargs)

  File "/home/dsmr/dsmr-reader/dsmr_frontend/mixins.py", line 10, in dispatch
    if not FrontendSettings.get_solo().always_require_login:

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/solo/models.py", line 55, in get_solo
    obj, created = cls.objects.get_or_create(pk=cls.singleton_instance_id)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/db/models/manager.py", line 85, in manager_method
    return getattr(self.get_queryset(), name)(*args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/db/models/query.py", line 573, in get_or_create
    return self.get(**kwargs), False

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/db/models/query.py", line 425, in get
    num = len(clone)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/db/models/query.py", line 269, in __len__
    self._fetch_all()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/db/models/query.py", line 1308, in _fetch_all
    self._result_cache = list(self._iterable_class(self))

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/db/models/query.py", line 53, in __iter__
    results = compiler.execute_sql(chunked_fetch=self.chunked_fetch, chunk_size=self.chunk_size)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/db/models/sql/compiler.py", line 1154, in execute_sql
    cursor = self.connection.cursor()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/utils/asyncio.py", line 26, in inner
    return func(*args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/db/backends/base/base.py", line 259, in cursor
    return self._cursor()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/db/backends/base/base.py", line 235, in _cursor
    self.ensure_connection()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/utils/asyncio.py", line 26, in inner
    return func(*args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/db/backends/base/base.py", line 219, in ensure_connection
    self.connect()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/db/utils.py", line 90, in __exit__
    raise dj_exc_value.with_traceback(traceback) from exc_value

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/db/backends/base/base.py", line 219, in ensure_connection
    self.connect()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/utils/asyncio.py", line 26, in inner
    return func(*args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/db/backends/base/base.py", line 200, in connect
    self.connection = self.get_new_connection(conn_params)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/utils/asyncio.py", line 26, in inner
    return func(*args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/db/backends/postgresql/base.py", line 187, in get_new_connection
    connection = Database.connect(**conn_params)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/psycopg2/__init__.py", line 127, in connect
    conn = _connect(dsn, connection_factory=connection_factory, **kwasync)

moet ik nu de backup terug zetten?
hoe is de werkwijze, dat de data ook op externe hdd komt dan?

newtech2015, 2022-06-23:
kennelijk is de database verneukt.
op zich geen probleem, want ik heb een backup.

die terug zetten blijkt echter pain in the ass....
ik heb van bv verplaasten van database naar externe hdd, een beschrijving gemaakt, alle rechten ed kloppen.
dit staat nu in mijn manual om backup terug te kunnen zetten:

# Als ROOT user
supervisorctl stop all
sudo systemctl stop postgresql  (stop de database)
sudo sudo -u postgres dropdb dsmrreader (verwijderen database)
sudo sudo -u postgres createdb -O dsmrreader dsmrreader (aanamken database zonder inhout)
zcat /home/dsmr/dsmr-reader/backups/dsmrreader-postgresql-backup-Thursday.sql.gz | sudo sudo -u postgres psql dsmrreader (vullen database met inhoud uit backup file)
 
* database verwijden - kan niet (kan ik me nog voorstellen)
* nieuwe aanmaken zonder inhoud - geen root rechten (???)
verder kom ik niet

elke poging, gaat stuk op "dsmr password" (heb ik niet), of geen rechten

newtech2015, 2022-06-23:
root@raspberrypi:~# sudo -u postgres createdb -O dsmrreader dsmrreader
could not change directory to "/root": Permission denied
createdb: could not connect to database template1: could not connect to server: No such file or directory
        Is the server running locally and accepting
        connections on Unix domain socket "/var/run/postgresql/.s.PGSQL.5432"?

weet iemand iets?

dennissiemensma, 2022-06-23:
Je hoeft het issue niet telkens te sluiten hoor. Je kunt op de knop ernaast drukken :rocket:

dennissiemensma, 2022-06-23:
Je kunt als het goed is gewoon wissel naar de Postgres systeemuser met `sudo su - postgres` en dan kun je alles uitvoeren.

Je kunt kijken of Postgres uberhaupt werkt. Bijv.
```
sudo systemctl status postgresql.service 
```

dennissiemensma, 2022-06-23:
En herstarten met 
```
sudo systemctl restart postgresql.service
```

dennissiemensma, 2022-06-23:
Ik zie in je andere issue dat je de database stopt. Dat hoeft niet. Die moet altijd draaien, anders kun je er geen operaties op uitvoeren.

newtech2015, 2022-06-23:
postgress werkt, zover ik zie...  maar blijf niks mogen richting database

root@raspberrypi:~# sudo systemctl status postgresql.service
● postgresql.service - PostgreSQL RDBMS
   Loaded: loaded (/lib/systemd/system/postgresql.service; enabled; vendor prese
   Active: inactive (dead) since Thu 2022-06-23 17:59:52 CEST; 1h 48min ago
  Process: 568 ExecStart=/bin/true (code=exited, status=0/SUCCESS)
 Main PID: 568 (code=exited, status=0/SUCCESS)

Jun 23 17:56:57 raspberrypi systemd[1]: Starting PostgreSQL RDBMS...
Jun 23 17:56:57 raspberrypi systemd[1]: Started PostgreSQL RDBMS.
Jun 23 17:59:52 raspberrypi systemd[1]: Stopped PostgreSQL RDBMS.

root@raspberrypi:~# sudo -u postgres createdb -O dsmrreader dsmrreader
could not change directory to "/root": Permission denied
createdb: could not connect to database template1: could not connect to server: No such file or directory
        Is the server running locally and accepting
        connections on Unix domain socket "/var/run/postgresql/.s.PGSQL.5432"?
root@raspberrypi:~# ^C
root@raspberrypi:~# sudo sudo -u postgres dropdb dsmrreader
could not change directory to "/root": Permission denied
dropdb: could not connect to database template1: could not connect to server: No such file or directory
        Is the server running locally and accepting
        connections on Unix domain socket "/var/run/postgresql/.s.PGSQL.5432"?

newtech2015, 2022-06-23:
om als postgress user de database leeg te maken of te verwijden, vraagt ie een sudo password for postgress

dennissiemensma, 2022-06-23:
Ik weet niet of de DB draait, gezien:
```
Active: inactive (dead) since Thu 2022-06-23 17:59:52 CEST; 1h 48min ago"
```

Je hoeft er **geen** sudo voor te zetten als je al `postgres` user bent. Alleen als je `pi` bent

newtech2015, 2022-06-23:
Jij weet, uiteraard, veel meer dan mij over dsmr en alle services die deze gebruikt...

wat kan ik nu doen, om het weer goed te krijgen.
heb dus pi-sd kaart image van VOOR de ellende erin zitten.

deze zou toch goed moeten zijn, heeft gedraaid!
echter is Database FUBAR, .. dit zou te herstellen moeten zijn, maar helaas

nu kan ik mijn hele pi opnieuw gaan installeren, maar hoe zinvol is dat, ben me overtuigd dat dsmr op mn sd kaart goed moet zijn.

gaat me sowieso nog hoop tijd kosten dit, maar... wat zou jij doen, wat is slim?

dennissiemensma, 2022-06-23:
Heb je al de database proberen te (her)starten als `pi` user? 

```
sudo systemctl restart postgresql.service
```

Je was hier op de goede weg, alleen moet de DB dus weer draaien: https://github.com/dsmrreader/dsmr-reader/issues/1649#issuecomment-1164532528

newtech2015, 2022-06-23:
nog eens getest:
root@raspberrypi:~# sudo systemctl restart postgresql.service
root@raspberrypi:~# sudo systemctl status postgresql.service
● postgresql.service - PostgreSQL RDBMS
   Loaded: loaded (/lib/systemd/system/postgresql.service; enabled; vendor prese
   Active: active (exited) since Thu 2022-06-23 22:37:50 CEST; 25s ago
  Process: 26392 ExecStart=/bin/true (code=exited, status=0/SUCCESS)
 Main PID: 26392 (code=exited, status=0/SUCCESS)

Jun 23 22:37:50 raspberrypi systemd[1]: Starting PostgreSQL RDBMS...
Jun 23 22:37:50 raspberrypi systemd[1]: Started PostgreSQL RDBMS.

root@raspberrypi:~# sudo sudo -u postgres dropdb dsmrreader
could not change directory to "/root": Permission denied
dropdb: could not connect to database template1: could not connect to server: No such file or directory
        Is the server running locally and accepting
        connections on Unix domain socket "/var/run/postgresql/.s.PGSQL.5432"?

root@raspberrypi:~# sudo sudo -u postgres dropdb dsmrreader
could not change directory to "/root": Permission denied
dropdb: could not connect to database template1: could not connect to server: No such file or directory
        Is the server running locally and accepting
        connections on Unix domain socket "/var/run/postgresql/.s.PGSQL.5432"?
root@raspberrypi:~# ^C
root@raspberrypi:~# sudo sudo -u postgres createdb -O dsmrreader dsmrreader
could not change directory to "/root": Permission denied
createdb: could not connect to database template1: could not connect to server: No such file or directory
        Is the server running locally and accepting
        connections on Unix domain socket "/var/run/postgresql/.s.PGSQL.5432"?

dennissiemensma, 2022-06-23:
Oke je DB draait weer, denk ik. Zie je hier ook processen?

```
ps faux | grep postgres
```

Zo ja, staat er nog iets in de logfile van postgres?
```
sudo tail -n50 /var/log/postgresql/postgresql-11-main.log
```

De bestandsnaam is in jouw geval waarschijnlijk anders, maar vermoedelijk wel in `/var/log/postgresql/`.

newtech2015, 2022-06-23:
ps faux | grep postgres
root     31553  0.0  0.0   4368   520 pts/0    S+   23:17   0:00          \_ grep postgres

onderin log (/var/log/postgresql/postgresql-9.6-main.log, staat:
pg_ctl: directory "/mnt/sata/data/postgresql/9.6/main" is not a database cluster directory
pg_ctl: directory "/mnt/sata/data/postgresql/9.6/main" is not a database cluster directory
pg_ctl: directory "/mnt/sata/data/postgresql/9.6/main" is not a database cluster directory
pg_ctl: directory "/mnt/sata/data/postgresql/9.6/main" is not a database cluster directory
pg_ctl: directory "/mnt/sata/data/postgresql/9.6/main" is not a database cluster directory
pg_ctl: directory "/mnt/sata/data/postgresql/9.6/main" is not a database cluster directory

wat zegt dit?

laatse vraag van jou:
root@raspberrypi:~# sudo cat /etc/postgresql/9.6/main/postgresql.conf | grep data_directory
data_directory = '/mnt/sata/data/postgresql/9.6/main'   # use data in another directory

hierna nog eens: 
root@raspberrypi:~# sudo -u postgres createdb -O dsmrreader dsmrreader
could not change directory to "/root": Permission denied
createdb: could not connect to database template1: could not connect to server: No such file or directory
        Is the server running locally and accepting
        connections on Unix domain socket "/var/run/postgresql/.s.PGSQL.5432"?

helaas, nog niks

dennissiemensma, 2022-06-23:
De DB werkt niet (meer) omdat de mount voor de dataopslag niet werkt/goed is/niet bestaat:

```
pg_ctl: directory "/mnt/sata/data/postgresql/9.6/main" is not a database cluster directory
```
Je kunt hem het beste in `/etc/postgresql/11/main/postgresql.conf` terugzetten op de standaard locatie. Ik vermoed:

```
/var/lib/postgresql/9.6/main
```

En mogelijk moet je daarna nog [een nieuw cluster maken](https://postgrespro.com/docs/postgresql/9.4/creating-cluster).

Doe in ieder geval:
```
sudo mkdir -p /var/lib/postgresql/9.6/main
sudo chown postgres.postgres /var/lib/postgresql/9.6/main

sudo su - postgres
initdb -D /var/lib/postgresql/9.6/main
logout
```

En dan kun je hem eens proberen te herstarten, zodat die nieuwe locatie en datamap gebruikt:
```
sudo systemctl restart postgresql.service
```

En dan kun je proberen:
```
sudo sudo -u postgres createdb -O dsmrreader dsmrreader
```

Ongerelateerd: De "could not change directory to "/root": Permission denied" melding kun je negeren.

newtech2015, 2022-06-24:
hoi Dennis,

alweer 1/2 dag aan het proberen. 
de eerste regels, die gewoon linux zijn, dat werkt, zodra het ook maar iets met postgresql wordt... mag helemaal niks meer,

postgres@raspberrypi:~$ initdb -D /var/lib/postgresql/9.6/main
-su: initdb: command not found

commando kent mijn os niet dus

ook op sd kaartje, heb ik niet de rechten om een database te maken, leeg te maken, of wat dan ook.

kan het misschien nog iets met rechten / ownership zijn?
de map /var/lib/postgres  heeft owner + group postgres en rechten: 0755
de map /var/lib/postgres/9.6  heeft owner + group root en rechten 0755
de map /var/lib/postgres/9.6/main heeft owner + group postgres en rechten: 0755

hoort dit zo te staan?

wat is postgresql een vreselijk programma, als t neerkomt op datbase herstel!

dennissiemensma, 2022-06-24:
Het ligt niet aan Postgres. Bij een standaardinstallatie is een backup terugzetten zo gedaan. Het is de combinatie dat er "iets" stuk is waardoor Postgres niet meer werkt en dat het op afstand erg lastig debuggen is waar het aan ligt. 

Plus dat ik zelf op versie 11 zit, dus ik kan niet makkelijk voor je proberen wat nog in versie 9 werkt. Je kunt deze nog proberen als alternatief:

> Alternatively, you can run initdb via the [pg_ctl](https://postgrespro.com/docs/postgresql/9.4/app-pg-ctl) program like so:

```
sudo su - postgres
pg_ctl -D /var/lib/postgresql/9.6/main initdb
```

Ik weet trouwens niet wat je bedoelt met "ook op SD-kaartje". Als het goed is pak je gewoon het standaardpad, ongeacht waar die onderwater naar weg schrijft.

----

Mocht dit niet werken dan denk ik dat je uiteindelijk het snelste geholpen bent door gewoon je RaspberryPi opnieuw te installeren, met een nieuwe versie van het OS: https://github.com/dsmrreader/dsmr-reader/issues/1649#issuecomment-1164410802

*In de toekomst (vanaf DSMR-reader 6 of 7) zal ik trouwens alleen nog maar Docker ondersteunen en niet meer deze handmatige installaties. Dat maakt dit waarschijnlijk een stuk makkelijker op te lossen.*

newtech2015, 2022-06-24:
root@raspberrypi:~# sudo su - postgres
postgres@raspberrypi:~$ pg_ctl -D /var/lib/postgresql/9.6/main initdb
-su: pg_ctl: command not found

commando kent mijn os ook niet, zucht

dennissiemensma, 2022-06-24:
Wellicht is het dan deze, ik zie nu dat deze ook al voor versie 9 gedocumenteerd stond: https://dsmr-reader.readthedocs.io/en/v4/tutorial/installation/step-by-step.html#database-backend-postgresql

```
sudo su - postgres
pg_createcluster 9.6 main --start
```

Waarbij ik niet weet of je ook echt 9.6 hebt of een andere versie.

En anders is een herinstallatie van je Pi sws beter.

newtech2015, 2022-06-24:
postgres@raspberrypi:~$ pg_createcluster 9.6 main --start
Error: cluster configuration already exists

maar dan mag ik weer niks:
root@raspberrypi:~# sudo sudo -u postgres createdb -O dsmrreader dsmrreader
could not change directory to "/root": Permission denied
createdb: could not connect to database template1: could not connect to server: No such file or directory
        Is the server running locally and accepting
        connections on Unix domain socket "/var/run/postgresql/.s.PGSQL.5432"?

dennissiemensma, 2022-06-24:
Die melding over permissies mag je dus negeren. Je kunt nog proberen Postgres opnieuw te installeren:

```
sudo apt-get purge postgresql
sudo apt-get install -y postgresql
```

newtech2015, 2022-06-24:
sudo apt-get purge postgresql
sudo apt-get install -y postgresql

gedaan, echter geen resultaat

dennissiemensma, 2022-06-24:
Dan resteert alleen nog een herinstallatie van de Pi. Anders blijf je bezig.

newtech2015, 2022-06-24:
ik ben nu, op een andere rapberry pi, een nieuwe installatie aan het doen...

op een gegeven moment staat er iets van "wil je een database importeren, dan is nu de tijd omdat te doen".
daarna volgt (bij restore stukje"
sudo su - dsmr
./deploy.sh

logout
sudo supervisorctl start all
moet ik dit nu overlsaan, en de oorsponkelijke handleiding verder volgen?

dennissiemensma, 2022-06-24:
Jazeker, dat is het makkelijkste moment omdat er nog niets draait of aangemaakt is op dat punt.

dennissiemensma, 2022-06-25:
Je kunt een andere poort instellen https://github.com/dsmrreader/dsmr-reader/issues/1274

dennissiemensma, 2022-06-25:
Is het inmiddels gelukt?

newtech2015, 2022-06-25:
nee, volgende week ofzo, ga ik alle andere services die ook op de pi (met dsmr) draaien over zetten op een andere pi.
daarna de dsmr pi als een dsmr-only pi inrichten.

dsmr is een mooie tool, maar t blijft maar zware problemen geven met regelmaat, vreet ook sd-kaartjes
als er "straks" alleen maar dsmr op staat, is t in elk gaval wat gemakkelijker te hertsellen

dennissiemensma, 2022-06-25:
Als je veel waarde hecht aan de data, dan kun je het beste uberhaupt geen SD-kaartjes meer gebruiken. Het ligt namelijk niet zozeer aan de software, maar meer aan de kern van de Pi, want die is nooit bedoeld geweest om intensief I/O te schrijven naar een SD-kaart. En dat gaat juist niet samen met dit project, achteraf gezien.
Er zijn wel wat opties om "minder" te schrijven (bijv slechts elke 10 seconden 1 telegram), maar linksom of rechtsom gaan de kaartjes uiteindelijk dood.

Mocht je tzt meer informatie hebben over de andere zware problemen waar je regelmatig tegenaan loopt, dan hoor ik het graag. Want in principe doet DSMR-reader niet heel veel en is het ook gebouwd om uiteindelijk om te gaan met bijvoorbeeld downtime, tot een zeker hoogte. 

En als je er echt minder tijd aan kwijt wilt zijn, kun je ook overwegen om simpelweg als vervanger een willekeurige clouddienst af te nemen die met jouw toestemming je slimme meter uitleest (volgens mij zijn er wel diverse die zoiets aanbieden). Dan hoef je ook geen SD-kaartjes te vervangen en ook niet technische kennis te hebben van hoe dingen onderwater werken en kunt fixen. 

Alles zelf kunnen doen is leuk, maar dit is de schaduwzijde ervan, helemaal als je er niet zo 1-2-3 uit komt.

closed 2022-06-30 by dennissiemensma

newtech2015, 2022-07-03:
Hoi,
heb nu een pi, die start via hdd, en probeer weer dsmr5 te installeren.
echter, vanaf regel 

Set password for database user: 
sudo -u postgres psql -c "alter user dsmrreader with password 'dsmrreader';"

gaat t mis.  moet ik de regel in een keer plakken in putty, of alleen sudo -u postgres psql -c ?

beiden, gaat niet, heb geen root rechten om dit te doen, als root... vreemd.

gebruik nu, Raspbian GNU/Linux 11 (bullseye)

krijg deze output:
sudo -u postgres psql -c "alter user dsmrreader with password 'dsmrreader';"
could not change directory to "/root": Permission denied
ALTER ROLE

??

dennissiemensma, 2022-07-03:
Wat gaat er niet goed? Zoals eerder aangegeven kun je de melding `could not change directory to "/root": Permission denied` negeren.

newtech2015, 2022-07-04:
heb weer een nieuwe installatie gedaan, dit keer raspberry pi gebruikt ZONDER sd kaartje, opstarten via usb HDD.
dit keer ging installatie wel goed.
heb nu v5 dsmr draaien, backup kunnen importeren, werkt als een zonnetje, merkbaar sneller dan eerst.
bedankt voor de tip hierover nog!

newtech2015, 2022-10-11:
Hoi,
DSMR geinstalleerd volgens de handleiding.kennelijk gebruik ik dan dus postgress, want die versie kan ik wel vinden.Dan (?) gebruik ik dus kennelijk niet mysql, zoals ik begrijp, en als influx alleen een optioneel pakket is, gebruik ik dat dus ook niet.
dan wel een vraag over de python versie upgrade:ik heb de handleiding gevolgd, en als ik als niet-dsmr user "sudo su - dsmr", doe krijg ik netjes versie 3.9.9 te zien.

doe ik het echter, zoals in voorbeeld als DSMR user, dan krijg ik 3.6.9 te zien.

ben ik nu wel / of niet klaar, om te upgraden naar 5.x?

gr ron

dennissiemensma, 2022-10-11:
Ik weet niet of de DB draait, gezien "Active: inactive (dead) since Thu 2022-06-23 17:59:52 CEST; 1h 48min ago".Je hoeft er geen sudo voor te zetten als je als postgres user bent.

dennissiemensma, 2022-10-11:
> ben ik nu wel / of niet klaar, om te upgraden naar 5.x?

Ik denk dat er wat dingen door elkaar heen lopen. Mijn eerdere mailcomment van vandaag was blijkbaar een oude mail.

In juli gaf je aan v5 te draaien, dat is prima toch?
- https://github.com/dsmrreader/dsmr-reader/issues/1649#issuecomment-1173670177

dennissiemensma, 2022-10-11:
> kan ik de dsmr pagina een andere plek geven bv:  xx.xx.xx.xx/dsmr (bedoel ip adres met xx)?

In `/etc/nginx/sites-enabled/dsmr-webinterface` kun je de regel:
```
listen      80;
```

Aanpassen naar iets anders:
```
listen      8080;
```

Daarna:
```
sudo nginx -t
```
Check of er zoiets uit komt:
> nginx: the configuration file /etc/nginx/nginx.conf syntax is ok
> nginx: configuration file /etc/nginx/nginx.conf test is successful

En:
```
sudo systemctl restart nginx.service
```

Daarna is DSMR-reader beschikbaar op: http://xx.xx.xx.xx:8080

newtech2015, 2022-10-11:
krijg hele dag al reply's op mails van maanden terug??snap hier niets van!
gr ron

 
 

   
   - #1649 (comment)

dennissiemensma, 2022-10-11:
Ik denk dat je ze het beste kan negeren. Ik vermoed dat ofwel het mails waren die nooit goed verstuurd zijn of die Github nooit heeft verwerkt.