dsmrreader/dsmr-reader #1655: 💡Optie om de piekspanningen per dag over de verschillende fases te tonen

fjbosscha, 2022-07-04:
### Feature

Tenzij ik iets nog niet goed begrepen of gevonden heb, zou ik graag per dag de maximale spanning over de dag per fase bij willen houden. Binnen domoticz had ik die optie zelf toegevoegd. 
Reden is dat de piekspanning hier in de straat hoog is (> 248 volt). Dit resulteerde af en toe in het uitvallen van de omvormer. 
Deze optie biedt mij de mogelijkheid om sneller data richting Liander te sturen. 
Daarnaast dan graag per seconde een meting (maar wellicht is dat al mogelijk).

fjbosscha, 2022-07-04:
Graag deze dan toevoegen aan de statistieken

dennissiemensma, 2022-07-05:
Bedankt voor je verzoek. Deze informatie kun je in principe al zelf in de gaten houden via ofwel (1) de **API** ofwel (2) **MQTT**. Voordeel is ook dat je er zelfs een alert of iets op kan zetten op de plek waar je dat uitleest/ontvangt. 
Want waar jij dit per dag nodig hebt, zal iemand anders het weer per minuut of andere interval willen zien. Daar is voor mij geen beginnen aan en tevens onnodig door de beschikbare data-koppelingen.

Er is ook nog een andere optie (3), namelijk het achteraf opvragen van deze waarden in je **database**. Ik heb zelf geen meter die voltages geeft, maar dit zou moeten werken:

```
sudo su - postgres
psql dsmrreader

select date_trunc('day', timestamp) "day", MAX(phase_voltage_l1) as max_l1, MAX(phase_voltage_l2) AS max_l2, MAX(phase_voltage_l3) AS max_l3 from dsmr_datalogger_dsmrreading group by day order by day desc limit 50;

# Of als losse regels

select date_trunc('day', timestamp) "day", MAX(phase_voltage_l1) as max_l1, MAX(phase_voltage_l2) AS max_l2, MAX(phase_voltage_l3) AS max_l3 
from dsmr_datalogger_dsmrreading 
group by day 
order by day desc 
limit 50;
```

Dit geeft als het goed is de maximale voltages per dag, van de afgelopen 50 dagen. Mits de data nog niet uitgedund is door automatisch opruiming.

----

In alle gevallen zul je een eventuele datalogger-sleep (en wellicht backend-sleep) in de DSMR-reader-admin moeten verlagen naar bijv een halve seconde, mocht je dat nog niet gedaan hebben. Mits je de informatie zo realtime mogelijk wilt hebben.

_Je zoekt overigens naar de velden `phase_voltage_l1`, `phase_voltage_l2` en/of `phase_voltage_l3` in de API/MQTT/DB._

closed 2022-07-05 by dennissiemensma

fjbosscha, 2022-07-05:
Bedankt voor de reactie. het SQL-commando zat ik wel in een script die dit maandelijks voor mij eruit haalt.
Vond het zelf wel een mooie uitbreiding van de statistieken bij meter vlaggetjes/statistieken. Zie nu wel spanningspieken per fase daar staan, maar wat die inhouden is mij onduidelijk.

fjbosscha, 2022-07-05:
Nee, ik bedoelde:

Aantal stroomstoringen in alle fasen | 9
-- | --
Aantal langdurige stroomstoringen in alle fasen | 4
Aantal spanningsdippen in fase L1 | 22
Aantal spanningsdippen in fase L2 alleen voor meters met meerdere fasen | 22
Aantal spanningsdippen in fase L3 alleen voor meters met meerdere fasen | 22
Aantal spanningspieken in fase L1 |  
Aantal spanningspieken in fase L2 alleen voor meters met meerdere fasen |  
Aantal spanningspieken in fase L3

dennissiemensma, 2022-07-05:
Dat zijn gegevens die de netbeheerder meer zeggen denk ik. Je kunt ze terugvinden in de DSMR-protocol-specs:
- https://github.com/dsmrreader/dsmr-reader/blob/v5/docs/_static/downloads/netbeheer-nederland-dsmr-specs-v5.0.2.pdf

Specifiek op pagina 21 van 28 van die PDF, de `sags` en `swells`.

dennissiemensma, 2022-07-05:
Je kunt trouwens op de URL `/admin/dsmr_datalogger/meterstatisticschange/` zien wanneer ze gewijzigd zijn, mits DSMR-reader ze toen al trackte. Die datumtijd kan je wellicht ook nog helpen.

fjbosscha, 2022-07-05:
Dank voor de tips.
Neem dan maar aan dat een sag een spanning < 207 volt is (230 - 10%) en een swel een spanning > 253 volt is (230 + 10%)
10 % is de afwijking die maximaal mag (plus en min).