dsmrreader/dsmr-reader #1664: Readings komen niet meer binnen

fverpoest, 2022-07-08:
### Description

Ik heb sedert enkele weken volgend probleem. Na enkele dagen komen er geen reads meer binnen binnen DSMR Reader van de P1-poort. Een herstart lost het probleem op, maar na enkele dagen komt het steeds terug.

Wanneer ik de poort manueel wil uitlezen krijg ik het volgende te zien:
cu: open (/dev/ttyUSB0): Cannot allocate memory
cu: /dev/ttyUSB0: Line in use

Ik heb intussen een nieuwe installatie uitgevoerd (op een nieuwe SD-kaart, moest toch eens vervangen worden) en nu ook gekozen voor Docker (omwille van andere redenen, jullie kennen de voordelen van Docker). Echter is het probleem er na 2-3 dagen al weer terug.

Wat zou er fout kunnen lopen?

P1 poort is aangesloten op een P1-splitter van HomeWizard. Vandaaruit gaat de kabel via USB de Pi in. Een tweede poort op de actieve splitter is bestemd voor de P1 meter van HomeWizard. Hierop blijven alle metingen wel netjes binnen komen.

### DSMR-reader version

5 (Belgium)

### DSMR-reader platform

Docker (e.g. Xirixiz)

dennissiemensma, 2022-07-08:
Bedankt voor je vraag. Het is lastig te zeggen wat er speelt, maar je kunt wellicht beginnen bij de logfiles. Ik zou vooral in de logfile van de `datalogger` fouten verwachten: https://dsmr-reader.readthedocs.io/nl/v5/how-to/troubleshooting/logfiles.html

Verder geven splitters soms wat problemen omdat er altijd maar 1 kan zijn die het uitleest. Het verschilt per splitter, maar weet dat dat ook niet altijd lekker gaat. Je `Line in use` foutmelding van `cu` geeft al aan hoe fragiel het is, al zal dat meer veroorzaakt worden omdat DSMR-reader dan concurreert met `cu` om de verbinding. 
Als je DSMR-reader uitzet, verwacht ik dat `cu` wel goed werkt. Als `cu` dan nog steeds niet werkt, ligt het probleem sowieso bij je splitter.

fverpoest, 2022-07-09:
Bedankt Dennis voor je weeral vlotte ondersteuning.

Ik heb DSMR gestopt, maar er komen nog geen readings binnen. Het ligt hem dus aan de splitter. Ik ga even moeten nadenken hoe ik hier dan best verder ga, DSMR heeft dingetjes aan boord die HomeWizard niet heeft en omgekeerd dus zou ze graag beiden behouden.

In ieder geval, deze mag dicht :)

closed 2022-07-09 by fverpoest