dsmrreader/dsmr-reader #1671: 🚨 Upgrade 5.4 -> 5.5 failed

mthmulders, 2022-07-18:
### Description

Feitelijk hetzelfde gedrag als in #1663, alleen nu met de upgrade van 5.4 naar 5.5. Dit keer weet ik dat ik de terminal even open moet laten, en de eerste, volledige log moet overnemen in dit rapport.

<details>
  <summary>Volledige log</summary>

```

 --- You are currently running version:
5.4

 --- Checking Python version.
Running Python 3.8.0 (default, Dec  9 2021, 17:53:27)
[GCC 8.4.0]
[WARNING] Your Python version can still be used with your current DSMR-reader version. However, Python 3.10.0 or higher is preferred. Consider using it for new installations in the future.

 --- Checking for local file changes.

 --- Pulling remote repository for new commits...
remote: Enumerating objects: 235, done.
remote: Counting objects: 100% (235/235), done.
remote: Compressing objects: 100% (94/94), done.
remote: Total 235 (delta 153), reused 208 (delta 129), pack-reused 0
Receiving objects: 100% (235/235), 3.93 MiB | 11.37 MiB/s, done.
Resolving deltas: 100% (153/153), completed with 63 local objects.
From https://github.com/dsmrreader/dsmr-reader
   d02cee37..34673e9f  v5         -> origin/v5
 * [new branch]        dev/v5.6   -> origin/dev/v5.6
 * [new tag]           v5.5.0     -> v5.5.0

 --- Merging/updating checkout.
Updating d02cee37..34673e9f
Fast-forward
 README.md                                                                           |   8 +++----
 docs/_static/screenshots/v4/frontend/about.png                                      | Bin 88669 -> 0 bytes
 docs/_static/screenshots/v4/frontend/archive.png                                    | Bin 124833 -> 0 bytes
 docs/_static/screenshots/v4/frontend/compare.png                                    | Bin 85425 -> 0 bytes
 docs/_static/screenshots/v4/frontend/configuration.png                              | Bin 85354 -> 0 bytes
 docs/_static/screenshots/v4/frontend/dashboard.png                                  | Bin 115344 -> 0 bytes
 docs/_static/screenshots/v4/frontend/energy-contracts.png                           | Bin 284655 -> 0 bytes
 docs/_static/screenshots/v4/frontend/export.png                                     | Bin 123552 -> 0 bytes
 docs/_static/screenshots/v4/frontend/live.png                                       | Bin 228425 -> 0 bytes
 docs/_static/screenshots/v4/frontend/statistics.png                                 | Bin 135697 -> 0 bytes
 docs/_static/screenshots/v4/frontend/trends.png                                     | Bin 222320 -> 0 bytes
 docs/_static/screenshots/{v4 => v5}/admin/apisettings.png                           | Bin
 docs/_static/screenshots/{v4 => v5}/admin/backupsettings.png                        | Bin
 docs/_static/screenshots/{v4 => v5}/admin/consumptionsettings.png                   | Bin
 docs/_static/screenshots/{v4 => v5}/admin/dropboxsettings.png                       | Bin
 docs/_static/screenshots/{v4 => v5}/admin/emailbackupsettings.png                   | Bin
 docs/_static/screenshots/{v4 => v5}/admin/emailsettings.png                         | Bin
 docs/_static/screenshots/{v4 => v5}/admin/energysupplierprice.png                   | Bin
 docs/_static/screenshots/{v4 => v5}/admin/jsondaytotalsmqttsettings.png             | Bin
 docs/_static/screenshots/{v4 => v5}/admin/jsontelegrammqttsettings.png              | Bin
 docs/_static/screenshots/{v4 => v5}/admin/mindergassettings.png                     | Bin
 docs/_static/screenshots/{v4 => v5}/admin/mqttbrokersettings.png                    | Bin
 docs/_static/screenshots/{v4 => v5}/admin/notificationsetting.png                   | Bin
 docs/_static/screenshots/{v4 => v5}/admin/pvoutputaddstatussettings.png             | Bin
 docs/_static/screenshots/{v4 => v5}/admin/pvoutputapisettings.png                   | Bin
 docs/_static/screenshots/{v4 => v5}/admin/rawtelegrammqttsettings.png               | Bin
 docs/_static/screenshots/{v4 => v5}/admin/splittopicdaytotalsmqttsettings.png       | Bin
 docs/_static/screenshots/{v4 => v5}/admin/splittopicmeterstatisticsmqttsettings.png | Bin
 docs/_static/screenshots/{v4 => v5}/admin/splittopictelegrammqttsettings.png        | Bin
 docs/_static/screenshots/v5/frontend/about.png                                      | Bin 0 -> 378751 bytes
 docs/_static/screenshots/v5/frontend/archive.png                                    | Bin 0 -> 759681 bytes
 docs/_static/screenshots/v5/frontend/compare.png                                    | Bin 0 -> 325983 bytes
 docs/_static/screenshots/v5/frontend/configuration.png                              | Bin 0 -> 342064 bytes
 docs/_static/screenshots/v5/frontend/dashboard.png                                  | Bin 0 -> 440125 bytes
 docs/_static/screenshots/v5/frontend/energy-contracts.png                           | Bin 0 -> 1483993 bytes
 docs/_static/screenshots/v5/frontend/export.png                                     | Bin 0 -> 373936 bytes
[... 188 lines cut ...]
5.5

### DSMR-reader platform

Native (e.g. manual installation)

### Debug info dump

```shell
DSMR-READER
    App / Python / Database                                                        v5.4 / v3.8.0 / postgresql
    BE sleep / DL sleep / Retention / Override                                    1.0s / 0.5s / 8064h / False
    Latest telegram version read / Parser settings                                               "None" / "3"
    (!) Database migrations pending                                                                     1 (!)

DATA
    Telegrams total (est.)                                                                            2790100
    Consumption records electricity / gas (est.)                                               507810 / 18880

UNRESOLVED ISSUES
    Database growing large: 1486 MB, consider data cleanup (if not already enabled)                                     now

POSTGRESQL SIZE OF LARGEST TABLES (> 500 MB)
    public.dsmr_datalogger_dsmrreading                                                                1060 MB
```

dennissiemensma, 2022-07-18:
Om je virtualenv opnieuw aan te maken kun je dit doen:

```
sudo supervisorctl stop all

sudo su - dsmr
deactivate

rm -rf ~/dsmr-reader/.venv/

python3 -m venv ~/dsmr-reader/.venv/
source ~/dsmr-reader/.venv/bin/activate
pip3 install pip --upgrade
pip3 install -r ~/dsmr-reader/dsmrreader/provisioning/requirements/base.txt

# Als je bij deploy hieronder een melding krijgt dat de PID files niet bestaan, dat klopt. DSMR-reader was immers gestopt.
./deploy.sh

logout
sudo supervisorctl start all
```

Als daar geen fouten uit komen, zou het goed moeten draaien.

mthmulders, 2022-07-19:
> Bedankt voor je melding. Ik denk dat je die mappen wel kan weggooien.

Perfect, dat bleek al genoeg te zijn. Zoals ik al zei, ik ken de Python wereld niet goed genoeg om te weten of je dat straffeloos kunt doen...

Dank voor de bevestiging, hij draait weer als een zonnetje - en volgende upgrades gaan hopelijk weer net zo soepel als voorheen!

dennissiemensma, 2022-07-19:
Top, fijn om te horen!