dsmrreader/dsmr-reader #1672: 🚨No recent readings since version 5.4

Rockvolleyball, 2022-07-18:
### Description

Since version 5.4 a couple of times the readings stopped working. Never had this issue before, cable and Pi are not replaced and Pi is up to date.

Not sure what is happening and how to debug. Error message is:  
Currently installed: DSMR-reader v5.5
No recent readings received   (16 hours ago)
Day statistics are lagging behind   (1 day, 9 hours ago)

Is this a known issue? If not, what can I do to debug?

### DSMR-reader version

5.5

### DSMR-reader platform

Native (e.g. manual installation)

dennissiemensma, 2022-07-18:
Start by enabling DEBUG info and check [your logfiles](https://dsmr-reader.readthedocs.io/nl/v5/how-to/troubleshooting/logfiles.html). I would expect an error or more information in the datalogger-logfile.

dennissiemensma, 2022-07-20:
There may be something wrong with your Pi or its (OS) installation: 
```
serial.serialutil.SerialException: [Errno 12] could not open port /dev/ttyUSB0: [Errno 12] Cannot allocate memory: '/dev/ttyUSB0'
```

A bit similar is #1664, except there it also errors with "Line in use", which in your case differs.

----

What is the output of?

```
lsb_release -a

free -m

df -h

df -i
```

<details>
  <summary>E.g.:</summary>

```
pi@rpi4:~ $ lsb_release -a
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 10 (buster)
Release:	10
Codename:	buster

pi@rpi4:~ $ free -m
              total        used        free      shared  buff/cache   available
Mem:           1872         366         103         108        1402        1294

pi@rpi4:~ $ df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/root        30G  6.8G   22G  25% /
devtmpfs        776M     0  776M   0% /dev
tmpfs           937M   12K  937M   1% /dev/shm
tmpfs           937M   18M  919M   2% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           937M     0  937M   0% /sys/fs/cgroup
/dev/mmcblk0p1  253M   49M  204M  20% /boot
tmpfs           188M     0  188M   0% /run/user/1000

pi@rpi4:~ $ df -i
Filesystem      Inodes  IUsed   IFree IUse% Mounted on
/dev/root      1854288 139042 1715246    8% /
devtmpfs         88621    410   88211    1% /dev
tmpfs           171053      2  171051    1% /dev/shm
tmpfs           171053    551  170502    1% /run
tmpfs           171053      3  171050    1% /run/lock
tmpfs           171053     15  171038    1% /sys/fs/cgroup
/dev/mmcblk0p1       0      0       0     - /boot
tmpfs           171053     10  171043    1% /run/user/1000
```
</details>

Rockvolleyball, 2022-07-21:
Buster is a little bit outdated ;)

Nothing special there:
lsb_release -a
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 11 (bullseye)
Release:        11
Codename:       bullseye

free -m
               total        used        free      shared  buff/cache   available
Mem:            3838         413        2447          89         977        3204
Swap:             99           0          99

df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/root        30G  5.3G   23G  19% /
devtmpfs        1.8G     0  1.8G   0% /dev
tmpfs           1.9G   24K  1.9G   1% /dev/shm
tmpfs           768M  804K  767M   1% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
/dev/mmcblk0p1  253M   50M  203M  20% /boot
tmpfs           384M     0  384M   0% /run/user/1000

df -i
Filesystem      Inodes IUsed   IFree IUse% Mounted on
/dev/root      1903584 99787 1803797    6% /
devtmpfs         84204   419   83785    1% /dev
tmpfs           166636     3  166633    1% /dev/shm
tmpfs           819200   645  818555    1% /run
tmpfs           166636     3  166633    1% /run/lock
/dev/mmcblk0p1       0     0       0     - /boot
tmpfs            98276    17   98259    1% /run/user/1000

At this moment the software is running fine a couple of days. No special things in the log at this moment.

dennissiemensma, 2022-07-21:
Okay great, let me know if the problem returns. It's probably related to some random problem reading the serial port or so.

Rockvolleyball, 2022-08-01:
Problems are returning, sorry. Already replaced my Pi, same results. Problems did start with version 5.4 :-(

dennissiemensma, 2022-08-01:
Je kunt proberen te downgraden om te kijken of dat iets oplost: https://dsmr-reader.readthedocs.io/nl/v5/how-to/upgrading/downgrade.html

Let op dat je dezelfde volgorde handhaaft als in de link. Eerst het downgrade-script naar 5.3 (kan direct dat script, hoeft niet voor elke release ertussen), daarna de code wisselen naar de 5.3 tag + deploy.

Rockvolleyball, 2022-08-02:
Heb hem teruggedraaid naar 5.3. Zal de resultaten laten weten!

dennissiemensma, 2022-08-04:
Ik sluit deze even voor nu, gezien het de vorige keer ook bijna 2 weken duurde.

Ik kan je sterk aanraden om `atop` te installeren, deze zit in de standaard repo van Debian als het goed is. Het maakt periodiek een snapshot van je processen en geheugengebruik (ik dacht standaard elke minuut), zodat je vrij makkelijk achteraf kan achterhalen hoe de belasting van het systeem rond een tijdstip was. 
Dit werkt extreem goed bij crashes, waarbij het OS vanzelfsprekend down of herstart wordt en dus niet te zien is wat er gebeurt/gebeurde (helemaal als je niet actief op het systeem zat), maar in dit geval scheelt het dat je systeem niet crasht. 

Desondanks kan het wellicht wat toevoegen bij het zoeken naar een oorzaak, omdat je dan ook de aanloop kan terugzien naar het moment dat de datalogger weer `[Errno 12] Cannot allocate memory` geeft. Voor zover ik kan googlen komt dat vermoedelijk echt door geheugen issues, maar is het niet heel triviaal hoe dat dan bij jou tot een probleem zou leiden, gezien je meer dan zat geheugen hebt.

closed 2022-08-04 by dennissiemensma

Rockvolleyball, 2022-08-08:
Het blijft helaas bal @dennissiemensma. Ik ga binnenkort verder speuren met atop e.d.

Wel een stomme vraag wellicht, maar hoe krijg ik na de downgrade weer nieuwe versies? Deploy.sh blijft v5.3 installeren nu.

dennissiemensma, 2022-08-08:
@Rockvolleyball 

```
sudo su - dsmr
git checkout v5
./deploy.sh
```

Dan zit je weer op de v5 branch en een re-deploy pakt dan weer de laatste v5.x versie