dsmrreader/dsmr-reader #1673: 💡Optie in settings om terug te gaan naar oude grafiek stijl

RoboMagus, 2022-07-18:
### Feature

Voor V5.5 was het voor mij overzichtelijker te zien in grafieken wat wordt terug geleverd en verbruikt door teruglevering als negatieve waarden te interpreteren (onder kant van de grafiek). Sinds de laatste versie is dat niet meer het geval.
Ik hoop dat hiervoor in de configuratie settings een optie gemaakt kan worden waarmee tussen de grafiekstijlen gekozen kan worden.

Eventueel inclusief de optie om de cumulatieve verbruik / teruglevering totalen te plotten. (Iets wat ik zelf wel een fijne toevoeging vindt, maar opties zijn altijd fijner dan verplichtingen 😉)

dennissiemensma, 2022-07-18:
Bedankt voor je verzoek. Weet dat je (vanzelfsprekend) altijd vrij bent om op een versie te blijven of terug te gaan. In principe zijn alle upgrades weer terug te draaien. 

Voor nu kun je het beste teruggaan naar v5.4 of waar je ook op zat: https://dsmr-reader.readthedocs.io/nl/v5/how-to/upgrading/downgrade.html

Het is voor mij niet te doen om alles configureerbaar te maken. Ik heb er al een keer naar gekeken, maar daar is eCharts te complex voor. Af en toe zullen er dingen wijzigen, soms onzichtbaar, maar in dit geval wat duidelijker. 
Soms probeer ik wat, soms draai ik iets terug, soms is iets een experiment. Soms verandert de GUI overigens ook vele upgrades niet. 
Upgrades zijn daarnaast nooit noodzakelijk, tenzij expliciet aangegeven in de changelogs (zelden). De kern van DSMR-reader verandert zelden.

Verder is het goed om te weten dat ik geen vaste feedback loop heb. Ik heb geen idee hoeveel gebruikers er zijn, hoeveel er welke versie draaien en hoe representatief de handvol mensen zijn die zich op Github melden. 

Ik zal kijken of ik ooit ergens op goedkope hosting een latest versie van DSMR-reader met dummy-data kan draaien, waarbij het iets makkelijker is om vooraf wijzigingen te zien. Dat zeggende hebben is een downgrade van een minor release (naar bijvoorbeeld de vorige) ook relatief weinig werk hoor, dus voel je vooral niet vastgezet ofzo.

closed 2022-07-18 by dennissiemensma

dennissiemensma, 2022-07-18:
Bijgewerkt: https://dsmr-reader.readthedocs.io/nl/v5/how-to/upgrading/downgrade.html