dsmrreader/dsmr-reader #1675: 🚨

Mitch73-2, 2022-07-23:
### Description

Op dit moment draai ik onderstaande versie van HA en zie ik in de logs dat er iets fout gaat.
De afgelopen anderhalf jaar heeft dit zonder enig probleem gedraaid.

Version core-2022.7.6 
Version supervisor-2022.07.0 

### DSMR-reader version

5.5

### DSMR-reader platform

Docker (e.g. Xirixiz)

### Debug info dump

```shell
Logger: dsmr_parser.clients.protocol
Source: /usr/local/lib/python3.10/site-packages/dsmr_parser/clients/protocol.py:141
First occurred: 08:16:33 (11 occurrences)
Last logged: 09:35:43

2022-07-22 17:12:14 ERROR (MainThread) [dsmr_parser.clients.protocol] disconnected due to exception
Traceback (most recent call last):
File "/usr/local/lib/python3.10/site-packages/serial_asyncio/__init__.py", line 115, in _read_ready
data = self._serial.read(self._max_read_size)
File "/usr/local/lib/python3.10/site-packages/serial/serialposix.py", line 595, in read
raise SerialException(
serial.serialutil.SerialException: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
2022-07-22 17:14:05 WARNING (MainThread) [dsmr_parser.clients.protocol] Invalid telegram. The CRC checksum '55218' does not match the expected '16801'

DSMR_DEBUGINFO DUMP :
DSMR-READER
    App / Python / Database                                                       v5.5 / v3.10.5 / postgresql
    BE sleep / DL sleep / Retention / Override                                     1.0s / 5.0s / 672h / False
    Latest telegram version read / Parser settings                                                 "42" / "4"

DATA
    Telegrams total (est.)                                                                             262738
    Consumption records electricity / gas (est.)                                                61876 / 10348
```

dennissiemensma, 2022-07-23:
Ik begrijp niet exact wat je vraag is en hoe je setup is. Dit project doet zelf niets met HA.

Als de metingen niet binnenkomen kun je kijken of je toevallig de meter uitleest met twee soorten software. Ik zie namelijk dat je DSMR-parser gebruikt (via HA module vermoedelijk) en DSMR-reader.

Je kunt je ze tegelijkertijd laten uitlezen, maar **niet** op dezelfde seriele poort, gezien deze melding:
```
device reports readiness to read but returned no data (device disconnected or multiple access on port?)
```

Wellicht moet je bij de DSMR-module van HA zelf zijn.

Mitch73-2, 2022-07-23:
Hoi Dennis,

Na jouw bovenstaande opmerking of ik de meter niet uitlees met twee soorten software ben ik eens gaan spitten in HA en kwam ik tot de ontdekking dat "dsmr slimme meter" van HA ook gebruikt wordt. Tja, dan is het niet zo verwonderlijk dat ze samen ruzie maken en errors tot gevolg hebben. 

De "dsmr slimme meter" eruit gegooid en nu werkt het weer, zit me alleen af te vragen waarom en wanneer ik die "dsmr slimme meter" erin heb gezet.

dennissiemensma, 2022-07-24:
Oke top!

closed 2022-07-24 by dennissiemensma