dsmrreader/dsmr-reader #1676: 🚨 Live graph total kWh confusing

xorinzor, 2022-07-24:
### Description

Since the total usage and total redelivery back to the grid have been added, the graphs have been quite confusing.
That's because the "total" lines are based on the last 24 hours instead of just day by day (ie: start from 00:00).

Since everything else is measured on a day by day basis, I'd consider this to be a bug, as knowing what you've used in the last 24 hours isn't very useful either.

Example of my live graph showing that the "total" starts from saturday 14:45 (which is 24 hours ago). But now that includes power data from yesterday, instead of only the data of today.
[image]

### DSMR-reader version

5.5

dennissiemensma, 2022-07-24:
Thanks for reporting.The live graphs have a dynamic range, varying from one hour to a week, depending on what you've configured. The kWh was added due to #979. It's not meant to be a diff for the current day, since the dashboard states that as well, but rather to support the Wattage displayed along the range set. E.g. peaking 5kW says nothing about the actual usage, so this will help in checking consumption in a specific time window. In a future version the line may be a bit more transparent or disabled by default (you can click on the legend to disable it).

dennissiemensma, 2022-07-25:
In the next release the data series will still be available in the legend, but de-selected by default, for now. It simply flips the current state.

[image]

This may change again, depending on whether most users want to see the kWh as well, or not.

closed 2022-07-25 by dennissiemensma