dsmrreader/dsmr-reader #1677: Unable to configure dropbox backup 🙋

roypaschedag, 2022-07-26:
### Description

Het instellen van de dropbox backup resulteert in een error.

Ik vermoed een dsmr issue en niet specifiek door gebruik van docker (xirixiz image). Daarom hier geplaatst.

Stappen om te reproduceren:
1. Op de dropbox admin pagina klik ik op de autoriseer link.
2. dropbox.com pagina opent met de vraag voor toestemming.
3. Ik klik op "Toestaan"
4. Toegangscode is gegenereerd -> kopieer
5. Ik plak de toegangscode in het lege veld en druk op "opslaan"
6. Error pagina opent (lokaal ip:7777/admin/dsmr_backup/dropboxsettings/) met de tekst:

```
Server Error

  Sorry, something unexpected happened.

Exception:

  AttributeError: '_SSLAdapter' object has no attribute '_ca_certs'

Traceback:

  File "/usr/local/lib/python3.10/site-packages/django/core/handlers/base.py", line 181, in _get_response
    response = wrapped_callback(request, *callback_args, **callback_kwargs)

  File "/usr/local/lib/python3.10/site-packages/django/utils/decorators.py", line 130, in _wrapped_view
    response = view_func(request, *args, **kwargs)

  File "/usr/local/lib/python3.10/site-packages/django/views/decorators/cache.py", line 44, in _wrapped_view_func
    response = view_func(request, *args, **kwargs)

  File "/usr/local/lib/python3.10/site-packages/django/contrib/admin/sites.py", line 232, in inner
    return view(request, *args, **kwargs)

  File "/usr/local/lib/python3.10/site-packages/solo/admin.py", line 72, in change_view
    return super(SingletonModelAdmin, self).change_view(

  File "/usr/local/lib/python3.10/site-packages/django/contrib/admin/options.py", line 1660, in change_view
    return self.changeform_view(request, object_id, form_url, extra_context)

  File "/usr/local/lib/python3.10/site-packages/django/utils/decorators.py", line 43, in _wrapper
    return bound_method(*args, **kwargs)

  File "/usr/local/lib/python3.10/site-packages/django/utils/decorators.py", line 130, in _wrapped_view
    response = view_func(request, *args, **kwargs)

  File "/usr/local/lib/python3.10/site-packages/django/contrib/admin/options.py", line 1540, in changeform_view
    return self._changeform_view(request, object_id, form_url, extra_context)

  File "/usr/local/lib/python3.10/site-packages/django/contrib/admin/options.py", line 1586, in _changeform_view
    self.save_model(request, new_object, form, not add)

  File "/app/dsmr_backup/admin.py", line 68, in save_model
    auth_flow = pickle.loads(obj.serialized_auth_flow)  # noqa: S3-1

  File "/usr/local/lib/python3.10/site-packages/requests/adapters.py", line 170, in __setstate__
    self.init_poolmanager(

  File "/usr/local/lib/python3.10/site-packages/dropbox/session.py", line 57, in init_poolmanager
    ca_certs=self._ca_certs,
```

 

### DSMR-reader version

5.5

### DSMR-reader platform

Docker (e.g. Xirixiz)

### Debug info dump

```shell
DSMR-READER
    App / Python / Database                                                       v5.5 / v3.10.5 / postgresql
    BE sleep / DL sleep / Retention / Override                                     1.0s / 0.5s / 672h / False
    Latest telegram version read / Parser settings                                                 "42" / "4"

DATA
    Telegrams total (est.)                                                                             291794
    Consumption records electricity / gas (est.)                                                99306 / 30965
```

dennissiemensma, 2022-07-26:
Bedankt voor je melding! 

Ik kan het zelf nu ook reproduceren. Het *lijkt* een issue te zijn in de Dropbox SDK die ik gebruik. Downgraden van die SDK lijkt het op te lossen. Ik zal een patch uitbrengen die de SDK downgrade.

dennissiemensma, 2022-07-26:
@roypaschedag wil je eens in je Docker container proberen dit command te runnen:

```
python3 -m pip install dropbox==11.32.0
```

En daarna de container reloaded/herstarten en vervolgens opnieuw proberen Dropbox te koppelen?

Ben benieuwd of dat het ook fixt bij jou.

dennissiemensma, 2022-07-26:
Ik kan het ook herleiden in de Dropbox SDK release:
- https://github.com/dropbox/dropbox-sdk-python/compare/v11.32.0...v11.33.0

Inhoudelijk zegt het me niets, maar het lijkt er wel op dat er *iets* is veranderd met `_ca_certs`.

dennissiemensma, 2022-07-27:
Oke dank voor het dubbelchecken!

closed 2022-07-27 by dennissiemensma

dennissiemensma, 2022-07-27:
Patch via [v5.5.1](https://github.com/dsmrreader/dsmr-reader/releases/tag/v5.5.1)

dennissiemensma, 2022-07-27:
Ik heb het issue nu ook helder in de Dropbox SDK. Ik zal het hen voorleggen of ze het willen en kunnen fixen voor een toekomstige release.

dennissiemensma, 2022-07-27:
https://github.com/dropbox/dropbox-sdk-python/pull/440