dsmrreader/dsmr-reader #1679: 🙋 PVOutput: Data found, but not in sync. Last data timestamp is before expected

maikvitesse, 2022-08-01:
### Description

Ik krijg continu de onderstaande melding:

`2022-08-01 14:51:15,168 WARNING  services     get_export_data                 110 | PVOutput: Data found, but not in sync. Last data timestamp is before expected (2022-08-01 14:49:00+02:00 < 2022-08-01 14:50:00+02:00)`

Hoe kan ik dit oplossen?

### DSMR-reader version

5.5.1

### DSMR-reader platform

Docker (e.g. Xirixiz)

### Debug info dump

```shell
DSMR-READER
    App / Python / Database                                                     v5.5.1 / v3.10.5 / postgresql
    BE sleep / DL sleep / Retention / Override                                    1.0s / 10.0s / 168h / False
    Latest telegram version read / Parser settings                                                 "50" / "4"

DATA
    Telegrams total (est.)                                                                              97102
    Consumption records electricity / gas (est.)                                                55391 / 47306
```

dennissiemensma, 2022-08-01:
Bedankt voor je melding. Ik gebruik deze feature zelf niet, dus het is even graven hoe het ook alweer zat. In de commit-history heb ik destijds verwezen naar #467:

- https://github.com/dsmrreader/dsmr-reader/blame/v5/dsmr_pvoutput/services.py#L107

Dat is dan ook zo'n 4 jaar geleden.

Ik zie dit terug: https://github.com/dsmrreader/dsmr-reader/issues/467#issuecomment-410433958

> Afhankelijk van hoe snel je datalogger groepeert (als je dat aan hebt staan), kan het zijn dat je de melding `[i] PVOutput: Data found, not in sync. Last data timestamp < expected` krijgt. Dat betekent dat hij wil uploaden, maar nog geen data heeft. Dus hij wacht net zolang totdat er data is. Dit kom je denk ik alleen tegen wanneer de upload-vertraging `0` is.

Weet je trouwens zeker dat je dit niet op je vorige installatie had, maar je daar de logs niet gecheckt had wellicht?

maikvitesse, 2022-08-01:
Oh kijk aan, bedankt voor je snelle reactie!
Ik heb de uploadvertraging nu op 5 minuten gezet, lijkt goed te gaan.

Zou kunnen inderdaad dat ik de log op mijn vorige installatie over het hoofd heb gezien.

closed 2022-08-01 by maikvitesse

dennissiemensma, 2022-08-01:
Fijn om te horen, dank voor je terugkoppeling!