dsmrreader/dsmr-reader #1685: 💡 Meer mogelijkheden m.b.t. backups

PhyxionNL, 2022-08-03:
### Feature

Op het moment is het alleen mogelijk om een dagelijkse backup te maken. Dit is voor de meeste systemen geen probleem, maar op een RPi met SD kaart gaat de kaart daardoor wel erg snel achteruit (ik ben inmiddels weer aan mijn tweede toe dit jaar). Ik wil best wat extra verlies accepteren qua data om de SD kaart te sparen, dus naast de mogelijkheid om een dagelijkse backup te maken zou een wekelijkse al een hoop schelen (ik heb verder bijna niks op de RPi draaien wat schrijft, of in zeer beperkte mate).

dennissiemensma, 2022-08-03:
Bedankt voor je verzoek. Ik denk dat het minder schrijven niet perse de levensduur verlengt. Wellicht kun je eerder kijken of je je SD-kaart kan sparen door bijvoorbeeld voor de opslag een USB-stick of externe SSD/HDD aan de Pi kan aansluiten. Vooral laatstgenoemde heeft doorgaans vele schrijfuren nodig voordat je daar tegen limieten aan zit.
Een andere SD-kaart kan ook wel helpen, maar het blijft soms willekeur.

Verder kun je nog je datalogger sleep en backend sleep **verhogen** als je dat nog niet gedaan hebt.

---

Als je desondanks perse minder backups wilt maken, kun je de backup in een cronjob 6 dagen uitstellen met dit command:
```
sudo -u postgres psql -c "update dsmr_backend_scheduledprocess set planned = planned + interval '6 day' where module = 'dsmr_backup.services.backup.run';" dsmrreader
```

Als je die cronjob inplant op bijvoorbeeld https://crontab.guru/#0_0_*_*_2, dan wordt die elke dinsdag uitgevoerd en 6 dagen opgeschoven, maakt die maandag weer een backup en herhaalt die het schuiven weer op dinsdag. Het is wat spartaans maar zou wel moeten werken.

dennissiemensma, 2022-08-03:
Voor de volgende release zit er een ongedocumenteerde optie in om de interval te overschrijven.

closed 2022-08-03 by dennissiemensma

PhyxionNL, 2022-08-04:
Super, bedankt @dennissiemensma.

dennissiemensma, 2022-11-21:
De ongedocumenteerd optie wordt waarschijnlijk op den duur vervangen door een soortgelijke optie via de interface.

Dit in het kader van o.a. #1746 en het later lostrekken van de ingebouwde ondersteuning voor Dropbox door "externe" cloud-backup buiten DSMR-reader om.