dsmrreader/dsmr-reader #1687: 🙋

JeeKee80, 2022-08-05:
### Description

Ik heb hulp nodig bij het volgende: ik heb mn rapsberry pi volledig opnieuw geïnstalleerd zodat alles weer keurig schoon is. Helaas heb ik de oude database NIET met een dump opgeslagen (stom), maar ik heb wel de partial backups van 4 jaar.

Nu zou ik heel graag weer al die oude dat terug willen krijgen in m'n nieuwe database, maar ik kom er compleet niet uit hoe dat dan zou moeten. 

Zou iemand me kunnen helpen?

### DSMR-reader version

laatste

dennissiemensma, 2022-08-05:
Bedankt voor je vraag. Je hebt hiervoor je laatste partial backup nodig. Deze bevat alleen alle dagtotalen, wat doorgaans het "belangrijkste" is.

- Als je nieuwe installatie nog niet draait kun je de partial terugzetten op het moment dat de handleiding aangeeft dat dat kan: #1481
- Als je installatie al wel draait en het je zelf niet lukt, dan kun je dit doen: https://github.com/dsmrreader/dsmr-reader/issues/1272#issuecomment-785400344

dennissiemensma, 2022-08-08:
@JeeKee80 is het gelukt?

JeeKee80, 2022-08-13:
> Not yet tried it. Gonna try to tomorrow, but thanks for the fast response! Greatly appreciated!

Werkte uiteindelijk perfect. Even wat gerommel met een extra kolom, maar verder helemaal top! Thanks voor de hulp!

closed 2022-08-14 by dennissiemensma

JeeKee80, 2022-10-11:
Not yet tried it. Gonna try to tomorrow, but thanks for the fast response! Greatly appreciated!