dsmrreader/dsmr-reader #1688: Gasverbruik niet zichtbaar

wiwa1978, 2022-08-06:
### Description

Ben een Belgische gebruiker. Ik heb DSMR successvol geinstalleerd op RPi4. Ik zie enkel mijn elektriciteitsverbruik via DSMR. 

Op de webpagina van Fluvius, de netbeheerder, kan ik enkel de P1 poort openstellen van de elektriciteitsmeter. Heb ergens gelezen dat de gasmeter draadloos communiceert met de elektriciteitsmeter en dat dus het gasverbruik wel degelijk ook via de P1 poort uit te lezen is. In elk geval zie ik mijn verbruik op de webapp van Fluvius maar dus niet op DSMR.

Moet ik ergens iets instellen op DSMR om gasverbruik te zien? Bij de configuratie van DSMR staat "Disable gas capability" niet aangevinkt.

### DSMR-reader version

v5.6

### Debug info dump

```shell
DSMR-READER
    App / Python / Database                                                        v5.6 / v3.9.2 / postgresql
    BE sleep / DL sleep / Retention / Override                                     1.0s / 5.0s / 672h / False
    Latest telegram version read / Parser settings                                               "None" / "4"

DATA
    Telegrams total (est.)                                                                               1457
    Consumption records electricity / gas (est.)                                                      107 / -

UNRESOLVED ISSUES
    No day statistics found                                                                               now
```

dennissiemensma, 2022-08-06:
Je kunt het beste een telegram van je meter delen. Deze vind je terug in DSMR-reader op de URL: `/admin/dsmr_datalogger/meterstatistics/`

Zorg ervoor dat je deze regels weghaalt/niet deelt hier, gezien dat je meter-serienummers zijn die niemand wat aangaat:

```
0-0:96.1.1(....)

0-1:96.1.0(....)
```

wiwa1978, 2022-08-07:
Bedankt! Hier is een telegram:

/FLU5\253769484_A

0-0:96.1.4(50216)
0-0:96.1.1(****)
0-0:1.0.0(220807150912S)
1-0:1.8.1(006454.525*kWh)
1-0:1.8.2(006480.316*kWh)
1-0:2.8.1(000000.000*kWh)
1-0:2.8.2(000000.171*kWh)
0-0:96.14.0(0002)
1-0:1.7.0(02.615*kW)
1-0:2.7.0(00.000*kW)
1-0:21.7.0(00.153*kW)
1-0:41.7.0(02.357*kW)
1-0:61.7.0(00.103*kW)
1-0:22.7.0(00.000*kW)
1-0:42.7.0(00.000*kW)
1-0:62.7.0(00.000*kW)
1-0:32.7.0(244.8*V)
1-0:52.7.0(244.3*V)
1-0:72.7.0(244.2*V)
1-0:31.7.0(000.97*A)
1-0:51.7.0(009.65*A)
1-0:71.7.0(000.93*A)
0-0:96.3.10(1)
0-0:17.0.0(999.9*kW)
1-0:31.4.0(999*A)
0-0:96.13.0()
0-1:24.1.0(003)
0-1:96.1.1(***)
0-1:24.4.0(1)
0-1:24.2.3(220807150500S)(03378.631*m3)
!95BE

dennissiemensma, 2022-08-07:
Deze regel geeft wel aan dat er een gasmeter is aangesloten:
```
0-1:24.2.3(220807150500S)(03378.631*m3)
```

Zie je ook niets qua gasverbruik op deze URL? `/admin/dsmr_consumption/gasconsumption/`

dennissiemensma, 2022-08-07:
Oh ik zie het denk ik al. Je gebruikt nog de standaard DSMR-versie, op `/admin/dsmr_datalogger/dataloggersettings/` kun je bij DSMR version aangeven dat je Fluvius gebruikt. Daarna zou het beter moeten werken.

wiwa1978, 2022-08-07:
Bedankt Dennis. Ik zie nu inderdaad een grafiek "Recent Gas Consumption" met heel lage waarden (verbruik ook bijna geen gas nu) dus ik neem aan dat dit het momentele verbruik is. Is er een manier dat ik deze grafiek omhoog kan plaatsen (als 2de grafiek)?

dennissiemensma, 2022-08-07:
Je kunt het beste even door alle configuratie heen klikken, al zul je het meeste laten voor wat het is. De volgorde kun je ergens in `/admin/dsmr_frontend/` aanpassen.

closed 2022-08-07 by dennissiemensma