dsmrreader/dsmr-reader #1689: Home Assistant energie integratie is kapot door de error "Unexpected state class"

vdbrink, 2022-08-08:
### Feature

Sinds versie 2022.8 werkt in Home Assistant de dsmr integratie niet meer.
De error die hij geeft is "Unexpected state class"

Ik heb ook een issue gelogd in HA, https://github.com/home-assistant/core/issues/76347
Ik weet niet aan welke kant dit opgelost dient te worden.

dennissiemensma, 2022-08-08:
Bedankt voor je melding. DSMR-reader ondersteunt alleen een API en MQTT. Als er iets in HA stuk is, en helemaal tussen twee HA versies, dan vermoed ik dat je daar eerst zal moeten zoeken. Je kunt nog proberen om DSMR-reader te downgraden naar een eerdere versie om te zien of dat iets oplost, maar ik vermoed uit je verhaal dat je DSMR-reader niet geüpgraded hebt, alleen HA. Mocht je een concrete bug hebben in DSMR-reader/MQTT dan kijk ik er tzt uiteraard graag naar.

dennissiemensma, 2022-08-09:
Oorzaak lijkt al gevonden in HA-issue.