dsmrreader/dsmr-reader #1693: 💡Re-calculate prices button

hjkoster400d, 2022-08-20:
### Feature

Dank voor je mooie project.

Afgelopen week was ik bezig met terugwerkende kracht een energy contract aan te maken en het viel mij op dat het herberekenen van de prijzen kinderlijk eenvoudig is (https://dsmr-reader.readthedocs.io/en/v5/how-to/troubleshooting/recalculate-prices.html). 

Misschien is het een mooie feature om dit met een druk op de knop vanuit de contracten pagina te kunnen doen?

Mss is er een goeie reden waarom dit nog niet zo is hoor, maar het leek mij wel een mooie feature.

Alvast bedankt.

M.v.g.
Rik Koster

dennissiemensma, 2022-08-21:
Bedankt voor je vriendelijke woorden.

De reden dat het geen knop is, is dat acties via websites (dus ook DSMR-reader) vaak een timeout hebben van 30 of 60 seconden. Als iets langer duurt dan dat, breekt de verbinding op of wordt soms het proces (in dit geval het herberekenen) geannuleerd. Op den duurt gebeurt dat ook hierbij.
Technisch zijn daar uiteraard wel oplossingen voor, maar ik vind/vond dat niet opwegen tegen hoevaak iemand het waarschijnlijk nodig heeft en hoe relatief weinig moeite het via de command line is.

Dat is tevens een van de redenen waarom bijvoorbeeld een backup terugzetten ook via de command line gaat.

closed 2022-08-21 by dennissiemensma

PhyxionNL, 2022-11-01:
Om hier maar even op in te haken, ik zou ook graag een herbereken knop willen want met docker is het wat omslachtiger om via CLI te doen. Ik heb DSMR reader nu 2.5 jaar draaien en het wordt uitgevoerd in nog geen 10 seconden. En anders gewoon een melding dat het op de achtergrond wordt uitgevoerd en hier geen voortgang van bekend is 🙂

dennissiemensma, 2022-11-01:
Uiteraard is alles te maken, maar het is vooral de vraag of het de moeite waard is. Het op de achtergrond draaien is relatief veel werk voor iets waarvan ik (had) verwacht dat gebruikers het zelden nodig hebben. Als dit in de praktijk inmiddels anders is is het het heroverwegen waard.

Overigens doet het script er bij mij al 25 seconden over en dan heb ik ook nog snelle hardware, dus ik kan er niet vanuit gaan dat het bij iedereen even snel gaat.

PhyxionNL, 2022-11-02:
Eerder was het niet zo noodzakelijk, maar met de huidige wisselende energieprijzen zit ik elke maand aan het herberekenen.
Ik heb DSMR-reader hier draaien in docker op een RPi 3B, niet snel dus, en 2.5 jaar duurt nog geen 10 seconden 🙂
Is het misschien een idee om bij het toevoegen/wijzigen van een energiecontract een extra knop te maken met "Opslaan en herberekenen", wat dan alleen de periode van het energiecontract herberekend?