dsmrreader/dsmr-reader #1695: Buienradar temperatuur

PascalBu, 2022-08-24:
### Description

Sinds gisteren middag 15 uur blijft de temperatuur hangen:
[image]

Overigens zie ik dit ook in home assistant, dus waarschijnlijk iets veranderd bij Buienradar?

### DSMR-reader version

5.6

PascalBu, 2022-08-24:
Never mind, het lijkt alweer opgelost

closed 2022-08-24 by PascalBu

dennissiemensma, 2022-08-24:
Bedankt voor je melding. Het kan inderdaad zijn dat er even issues waren. Ik zie hetzelfde

[image]