dsmrreader/dsmr-reader #1697: 🙋Geen dag gegevens meer, maar wel realtime gegevens

rtrinks, 2022-08-29:
### Description

Zie mijn verbruik realtime en ook teruglevering werkt gewoon, evenals mijn verbruik in de maand augustus.
Wat ik niet zie is wat ik vandaag heb qua verbruik en teruglevering, daar staat enkel: 'Nog geen gegevens beschikbaar.'

P1 module herstart en DSMR up-to-date en reboot gegeven. Alle drie opties werken en staan status OK

Wel zie ik volgende staan:

   Database groeit gestaag: 2616 MB, overweeg dataopschoning (indien nog niet ingeschakeld)   (een seconde geleden)
   Te veel onverwerkte telegrammen: 1004377   (1 week, 4 dagen geleden)
   Dagstatistieken lopen achter   (1 week, 5 dagen geleden)

Automatische opschoning staat op: Schoon de meeste brondata op na een week (aanbevolen)

[image]

### DSMR-reader version

DSMR-reader v4.20

### DSMR-reader platform

Docker (e.g. Xirixiz)

dennissiemensma, 2022-08-29:
Wat zie je in de logs van het `dsmr_backend` proces? https://dsmr-reader.readthedocs.io/nl/v5/how-to/troubleshooting/logfiles.html

Eventueel kun je DEBUG mode aanzetten als je geen fouten ziet. Dan zie je exact wat dat proces doet en ook waarom iets niet.

dennissiemensma, 2022-08-29:
Los van bovenstaande kun je ook nog even eenmalig een `vacuumdb` doen. Dat is vaak nodig als je pas later dataopschoning hebt aangezet. PostgreSQL geeft die ruimte niet perse vrij, vandaar ook die melding over de grootte.

Let wel dat dit niet je oorspronkelijke issue oplost, maar het kan niet kwaad. Ik weet niet of dit nog werkt, maar je kunt dit proberen: 
- https://github.com/dsmrreader/dsmr-reader/issues/1165#issuecomment-720326035

dennissiemensma, 2022-08-29:
Veelvoorkomende oorzaak van achterlopende data met de Docker-variant betreft trouwens tijdzones. Je kunt sws checken of het daar toevallig niet in zit:
- #1325
- #1645
- #1282

rtrinks, 2022-09-02:
@dennissiemensma bedankt voor alle informatie. Probleem is uit zich zelf weer opgelost. Staat er plotseling weer tussen.

Enige melding die ik nog heb staan is:    Database groeit gestaag: 2618 MB, overweeg dataopschoning (indien nog niet ingeschakeld)   (nu)

Ga met de info van het weekend even aan de slag. Dankjewel.

dennissiemensma, 2022-09-04:
Een vacuumdb zou die melding moeten oplossen. Zie mijn comment/verwijzing hierboven.

closed 2022-09-20 by dennissiemensma