dsmrreader/dsmr-reader #1698: 💡 Week/Maandrapporten via mail

AbelLykens, 2022-09-01:
### Feature

Het zou leuk zijn om een week- of maandrapport naar de mensen in mijn huishouden te kunnen sturen met verbruik en kosten. Zegmaar de archief-pagina van een maand, in de mail.

Een django-admin-command om tekstuele output te leveren die ik dan in `|mail` kan pipen zou ook werken.

Benieuwd of zoiets kan :)

dennissiemensma, 2022-09-04:
Bedankt voor je verzoek. Ik denk dat je uiteindelijk ook een soortgelijk iets voor elkaar kan krijgen via de API of database. De tabel met dagstatistieken zou alle gegevens moeten hebben die je zoekt.

AbelLykens, 2022-09-19:
```from decimal import Decimal

from django.core.management.base import BaseCommand, CommandError
from django.utils.translation import gettext as _
from django.db import models

from dsmr_stats.models.statistics import DayStatistics

import datetime

class Command(BaseCommand):
    help = _('Returns stats of the week')

    def handle(self, **options):
        today = datetime.date.today()
        wds = DayStatistics.objects.filter(day__range=( today - datetime.timedelta(days=7), today ))

        total_cost = wds.aggregate(models.Sum('total_cost'))['total_cost__sum']
        electricity_costs = wds.aggregate(models.Sum('electricity1_cost'))['electricity1_cost__sum'] + wds.aggregate(models.Sum('electricity2_cost'))['electricity2_cost__sum']
        gas_cost = wds.aggregate(models.Sum('gas_cost'))['gas_cost__sum']

        print(f"Energiekosten afgelopen 7 dagen € {total_cost}, waarvan € {electricity_costs} voor stroom en € {gas_cost} voor gas.")
```

Mooi is anders, oh well.. Het werkt. Durf het (met deze lelijke code) alleen niet te PRren haha ;)

dennissiemensma, 2022-09-20:
Bedankt voor je update, die code lijkt me prima hoor. Bedankt voor het delen voor als iemand ooit hetzelfde wilt en dit issue vindt.

_Houd er trouwens wel rekening mee dat je dit niet te dicht na middernacht uitvoert, omdat je soms wel eens wat verschuivingen hebt qua tijdzones. Na 2:00 's nachts zou altijd goed moeten zijn en dan weet je ook zeker dat vandaag echt vandaag is NL-tijd, en niet onbedoeld "gisteren" in UTC._

closed 2022-09-20 by dennissiemensma