dsmrreader/dsmr-reader #1705: 💡Energy Contract Details

darf53, 2022-09-09:
### Feature

Putting in the details of my energy contract I'm missing following features:

- We do not have different tarifs like "hoogtarief" and "laagtarief". Having this split on every dashboard is confusing. Also when we need to enter the contract details it is not clear to me if I need to add it or not and do I put same price plans in both fields.

- On the admin page, where you can add the contract details, there's no indication what data you need to provide apart from the fact it is in "€". But what metric is used. (m3 of kWh or what volume...)

dennissiemensma, 2022-09-09:
Hi, you can have the tariffs merged into one. Somewhere in Frontend settings in admin. Just put the same price for all tariffs then. Prices are per M3 or kWh.

dennissiemensma, 2022-09-20:
I will see if I can add the units to the help text some time.

dennissiemensma, 2022-09-23:
In the next release, the units (e.g. €/kWh) will be displayed.

[image]

closed 2022-09-23 by dennissiemensma