dsmrreader/dsmr-reader #1706: Server Error | DSMR-reader 🚨

markdiesel, 2022-09-12:
### Description

DSMR / Django throws the error message below. 
After sudo supervisorctl restart all the server is working again for a short time interval 

Error-message on website; 

> 
Server Error
Sorry, something unexpected happened.
Exception:

OperationalError: server closed the connection unexpectedly
	This probably means the server terminated abnormally
	before or while processing the request.

Traceback:

  File "/home/dsmr/dsmr-reader/.venv/lib/python3.9/site-packages/django/core/handlers/base.py", line 181, in _get_response
    response = wrapped_callback(request, *callback_args, **callback_kwargs)

  File "/home/dsmr/dsmr-reader/.venv/lib/python3.9/site-packages/django/views/generic/base.py", line 70, in view
    return self.dispatch(request, *args, **kwargs)

  File "/home/dsmr/dsmr-reader/dsmr_frontend/mixins.py", line 10, in dispatch
    if not FrontendSettings.get_solo().always_require_login:

  File "/home/dsmr/dsmr-reader/.venv/lib/python3.9/site-packages/solo/models.py", line 55, in get_solo
    obj, created = cls.objects.get_or_create(pk=cls.singleton_instance_id)

  File "/home/dsmr/dsmr-reader/.venv/lib/python3.9/site-packages/django/db/models/manager.py", line 85, in manager_method
    return getattr(self.get_queryset(), name)(*args, **kwargs)

  File "/home/dsmr/dsmr-reader/.venv/lib/python3.9/site-packages/django/db/models/query.py", line 581, in get_or_create
    return self.get(**kwargs), False

  File "/home/dsmr/dsmr-reader/.venv/lib/python3.9/site-packages/django/db/models/query.py", line 431, in get
    num = len(clone)

  File "/home/dsmr/dsmr-reader/.venv/lib/python3.9/site-packages/django/db/models/query.py", line 262, in __len__
    self._fetch_all()

  File "/home/dsmr/dsmr-reader/.venv/lib/python3.9/site-packages/django/db/models/query.py", line 1324, in _fetch_all
    self._result_cache = list(self._iterable_class(self))

  File "/home/dsmr/dsmr-reader/.venv/lib/python3.9/site-packages/django/db/models/query.py", line 51, in __iter__
    results = compiler.execute_sql(chunked_fetch=self.chunked_fetch, chunk_size=self.chunk_size)

  File "/home/dsmr/dsmr-reader/.venv/lib/python3.9/site-packages/django/db/models/sql/compiler.py", line 1175, in execute_sql
    cursor.execute(sql, params)

  File "/home/dsmr/dsmr-reader/.venv/lib/python3.9/site-packages/django/db/backends/utils.py", line 66, in execute
    return self._execute_with_wrappers(sql, params, many=False, executor=self._execute)

  File "/home/dsmr/dsmr-reader/.venv/lib/python3.9/site-packages/django/db/backends/utils.py", line 75, in _execute_with_wrappers
    return executor(sql, params, many, context)

  File "/home/dsmr/dsmr-reader/.venv/lib/python3.9/site-packages/django/db/backends/utils.py", line 84, in _execute
    return self.cursor.execute(sql, params)

  File "/home/dsmr/dsmr-reader/.venv/lib/python3.9/site-packages/django/db/utils.py", line 90, in __exit__
    raise dj_exc_value.with_traceback(traceback) from exc_value

  File "/home/dsmr/dsmr-reader/.venv/lib/python3.9/site-packages/django/db/backends/utils.py", line 84, in _execute
    return self.cursor.execute(sql, params)

### DSMR-reader version

5.6

### DSMR-reader platform

Native (e.g. manual installation)

dennissiemensma, 2022-09-12:
Hi please provide a debug info dump from the About page.

markdiesel, 2022-09-12:
Hi Dennis,

Below the dum info

DSMR-READER
    App / Python / Database
            v5.6 / v3.9.2 / postgresql
    BE sleep / DL sleep / Retention / Override
            1.0s / 0.5s / 672h / False
    Latest telegram version read / Parser settings
                            "50" / "4"

DATA
    Telegrams total (est.)
                               1169939
    Consumption records electricity / gas (est.)
                         83650 / 35780

dennissiemensma, 2022-09-12:
You can try the PostgreSQL database logs. I do not see errors in DSMR-reader you can resolve yourself. Maybe the database has some issues.

markdiesel, 2022-09-13:
Hi Dennis,

In de bijlage de laatste logfile van postgresql, misschien kun je hier iets
mee. Zag dat de backup ook niet meer gemaakt wordt sinds zondag. Heb
geprobeerd om een handmatige backup te maken. deze liep vast met
onderstaande output;

sudo
/home/dsmr/dsmr-reader/dsmrreader/provisioning/postgresql/psql-backup.sh
 --- Dumping backup of 'dsmrreader' to: /data/backup-postgres-Tuesday.sql.gz
pg_dump: last built-in OID is 16383
pg_dump: reading extensions
pg_dump: identifying extension members
pg_dump: reading schemas
pg_dump: reading user-defined tables
pg_dump: reading user-defined functions
pg_dump: reading user-defined types
pg_dump: reading procedural languages
pg_dump: reading user-defined aggregate functions
pg_dump: reading user-defined operators
pg_dump: reading user-defined access methods
pg_dump: reading user-defined operator classes
pg_dump: reading user-defined operator families
pg_dump: reading user-defined text search parsers
pg_dump: reading user-defined text search templates
pg_dump: reading user-defined text search dictionaries
pg_dump: reading user-defined text search configurations
pg_dump: reading user-defined foreign-data wrappers
pg_dump: reading user-defined foreign servers
pg_dump: reading default privileges
pg_dump: reading user-defined collations
pg_dump: reading user-defined conversions
pg_dump: reading type casts
pg_dump: reading transforms
pg_dump: reading table inheritance information
pg_dump: reading event triggers
pg_dump: finding extension tables
pg_dump: finding inheritance relationships
pg_dump: reading column info for interesting tables
pg_dump: finding the columns and types of table "public.auth_group"
pg_dump: finding default expressions of table "public.auth_group"
pg_dump: finding the columns and types of table
"public.auth_group_permissions"
pg_dump: finding default expressions of table
"public.auth_group_permissions"
pg_dump: finding the columns and types of table "public.auth_permission"
pg_dump: finding default expressions of table "public.auth_permission"
pg_dump: finding the columns and types of table "public.auth_user"
pg_dump: finding default expressions of table "public.auth_user"
pg_dump: finding the columns and types of table "public.auth_user_groups"
pg_dump: finding default expressions of table "public.auth_user_groups"
pg_dump: finding the columns and types of table
"public.auth_user_user_permissions"
pg_dump: finding default expressions of table
"public.auth_user_user_permissions"
pg_dump: finding the columns and types of table "public.django_admin_log"
pg_dump: finding default expressions of table "public.django_admin_log"
pg_dump: finding check constraints for table "public.django_admin_log"
pg_dump: finding the columns and types of table "public.django_content_type"
pg_dump: finding default expressions of table "public.django_content_type"
pg_dump: finding the columns and types of table "public.django_migrations"
pg_dump: finding default expressions of table "public.django_migrations"
pg_dump: finding the columns and types of table "public.django_session"
pg_dump: finding the columns and types of table
"public.dsmr_api_apisettings"
pg_dump: finding default expressions of table "public.dsmr_api_apisettings"
pg_dump: finding the columns and types of table
"public.dsmr_backend_backendsettings"
pg_dump: finding default expressions of table
"public.dsmr_backend_backendsettings"
[... 273 lines cut ...]
pg_dump: reading rewrite rules
pg_dump: reading policies
pg_dump: reading row-level security policies
pg_dump: reading publications
pg_dump: reading publication membership
pg_dump: reading subscriptions
pg_dump: reading large objects
pg_dump: reading dependency data
pg_dump: error: query failed: ERROR:  invalid page in block 3 of relation
base/16385/2756
pg_dump: error: query was: SELECT classid, objid, refclassid, refobjid,
deptype FROM pg_depend WHERE deptype != 'p' AND deptype != 'e'
UNION ALL
SELECT 'pg_opfamily'::regclass AS classid, amopfamily AS objid, refclassid,
refobjid, deptype FROM pg_depend d, pg_amop o WHERE deptype NOT IN ('p',
'e', 'i') AND classid = 'pg_amop'::regclass AND objid = o.oid AND NOT
(refclassid = 'pg_opfamily'::regclass AND amopfamily = refobjid)
UNION ALL
SELECT 'pg_opfamily'::regclass AS classid, amprocfamily AS objid,
refclassid, refobjid, deptype FROM pg_depend d, pg_amproc p WHERE deptype
NOT IN ('p', 'e', 'i') AND classid = 'pg_amproc'::regclass AND objid =
p.oid AND NOT (refclassid = 'pg_opfamily'::regclass AND amprocfamily =
refobjid)
ORDER BY 1,2
-rw-r--r-- 1 root root 20 Sep 13 18:38 /data/backup-postgres-Tuesday.sql.gz

dennissiemensma, 2022-09-13:
query failed: ERROR: invalid page in block 3 of relation base/16385/2756. Dat is meestal disk corruptie. Heb je nog een backup? Vermoedelijk is de sd kaart stuk.

markdiesel, 2022-09-16:
> Dank je, daar was ik al bang voor . Zag ook meldingen in de andere logs.
> ga de SD vervangen en dan alles overzetten. Heb backup gelukkig.
> Als daarna nog meldingen komen meldt ik mij weer, bedankt zover
>
> Mark

Hi Dennis,

Ik heb de installatie gemigreerd naar een andere schijf op de Raspberry, ik
kan nu weer een backup maken.
Als ik ik nu kijk in het dashboard zie ik wel de live data, maar geen
grafiek over de dag. De tijdlijn gaat niet verder dan het moment dat de
database gemigreerd is.
HomeAssistant laat wel live data zien op basis van MQTT -data

Zie afbeeldingen

Is er een manier om de database te herstellen ?

[image: afbeelding.png]

[image: afbeelding.png]

Geen recent verbruik, wel kwartier gegevens. Opbrengst en verbruik gestopt
om 19:58 uur ,

[image: afbeelding.png]

dennissiemensma, 2022-09-16:
Staan er ook geen meldingen op de pagina met algemene info? Wellicht loopt je dataverwerking nog achter. En anders kun je debug level logging tijdelijk aanzetten en kijken of de logs van dsmr_backend wat aangeven.

dennissiemensma, 2022-09-27:
@markdiesel ben je hier nog verder mee gekomen?

markdiesel, 2022-09-29:
Hoi Dennis ,

Zie laatste line; probleem opgelost, alles draait weer als een zonnetje .
bedankt voor de support

dennissiemensma, 2022-09-29:
Oke top!

closed 2022-09-29 by dennissiemensma

markdiesel, 2022-10-11:
> Assigned #1706 <https://github.com/dsmrreader/dsmr-reader/issues/1706> to

Na overzetten rootfs van sd naar sdd was tijdzone aangepast, Deze goedgezet
en alles werd weer verwerkt.
resume;
sd corrupt --> error in database
rootfs overgezet naar sdd op raspberry (mbv rpi clone)
verwijzingen naar postgresql goedgezet
tijdzone gewijzigd van Amsterdam naar London
tijdzone aangepast
probleem opgelost

bedankt