dsmrreader/dsmr-reader #1707: 🙋 Django missing while upgrading to V5?

Djjvb, 2022-09-19:
### Description

I am updating to  V5  from V4.20 (I came from 3.12). All was going  well until step 5: upgrade to dsmr-reader step 5. Especially at point 'installing dependencies'. 

I  entered the code:

source ~/dsmr-reader/.venv/bin/activate

pip3 install pip --upgrade
pip3 install -r ~/dsmr-reader/dsmrreader/provisioning/requirements/base.txt

And got an error:
ERROR: Could not find a version that satisfies the requirement django-debug-toolbar==3.5.0 (from versions: 0.7.0, 0.8.0, 0.8.1, 0.8.2, 0.8.3, 0.8.4, 0.8.5, 0.9.1, 0.9.2, 0.9.3, 0.9.4, 0.10.0, 0.10.1, 0.10.2, 0.11.0, 1.0, 1.0.1, 1.1, 1.2, 1.2.1, 1.2.2, 1.3.0, 1.3.2, 1.4, 1.5, 1.6, 1.7, 1.8, 1.9, 1.9.1, 1.10, 1.10.1, 1.11, 1.11.1, 2.0a1, 2.0, 2.1, 2.2, 2.2.1, 3.0a1, 3.0a2, 3.0a3, 3.0a4, 3.0, 3.1, 3.1.1, 3.2a1, 3.2, 3.2.1, 3.2.2, 3.2.3, 3.2.4)
ERROR: No matching distribution found for django-debug-toolbar==3.5.0

When  I tried to move on, with the "./manage.py check" command I got the following error:

Traceback (most recent call last):
  File "./manage.py", line 8, in <module>
    from django.core.management import execute_from_command_line
ModuleNotFoundError: No module named 'django'

I also  tried to logout  and run it from the pi@raspberrypi line. But that didn't work either. 

So how do I  proceed? 

### DSMR-reader version

4.20

### DSMR-reader platform

Native (e.g. manual installation)

dennissiemensma, 2022-09-20:
There seems to be a package dependency issue. Which python version are you running?

Djjvb, 2022-09-20:
I don't know what happened, but it  was on 3.6.9 (again?). I've updated  it to 3.9.(13?). But again at the ./manage.py check command I get another error:

Traceback (most recent call last):
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.9/site-packages/django/db/backends/postgresql/base.py", line 25, in <module>
    import psycopg2 as Database
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.9/site-packages/psycopg2/__init__.py", line 51, in <module>
    from psycopg2._psycopg import (                     # noqa
ImportError: /home/dsmr/dsmr-reader/.venv/lib/python3.9/site-packages/psycopg2/_psycopg.cpython-39-arm-linux-gnueabihf.so: undefined symbol: PQencryptPasswordConn

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/dsmr/dsmr-reader/./manage.py", line 10, in <module>
    execute_from_command_line(sys.argv)
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.9/site-packages/django/core/management/__init__.py", line 419, in execute_from_command_line
    utility.execute()
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.9/site-packages/django/core/management/__init__.py", line 395, in execute
    django.setup()
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.9/site-packages/django/__init__.py", line 24, in setup
    apps.populate(settings.INSTALLED_APPS)
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.9/site-packages/django/apps/registry.py", line 114, in populate
    app_config.import_models()
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.9/site-packages/django/apps/config.py", line 301, in import_models
    self.models_module = import_module(models_module_name)
  File "/usr/local/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 850, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.9/site-packages/django/contrib/auth/models.py", line 3, in <module>
    from django.contrib.auth.base_user import AbstractBaseUser, BaseUserManager
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.9/site-packages/django/contrib/auth/base_user.py", line 48, in <module>
    class AbstractBaseUser(models.Model):
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.9/site-packages/django/db/models/base.py", line 122, in __new__
    new_class.add_to_class('_meta', Options(meta, app_label))
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.9/site-packages/django/db/models/base.py", line 326, in add_to_class
    value.contribute_to_class(cls, name)
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.9/site-packages/django/db/models/options.py", line 207, in contribute_to_class
    self.db_table = truncate_name(self.db_table, connection.ops.max_name_length())
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.9/site-packages/django/utils/connection.py", line 15, in __getattr__
    return getattr(self._connections[self._alias], item)
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.9/site-packages/django/utils/connection.py", line 62, in __getitem__
    conn = self.create_connection(alias)
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.9/site-packages/django/db/utils.py", line 204, in create_connection
    backend = load_backend(db['ENGINE'])
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.9/site-packages/django/db/utils.py", line 111, in load_backend
    return import_module('%s.base' % backend_name)
  File "/usr/local/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "/home/dsmr/dsmr-reader/.venv/lib/python3.9/site-packages/django/db/backends/postgresql/base.py", line 29, in <module>
    raise ImproperlyConfigured("Error loading psycopg2 module: %s" % e)
django.core.exceptions.ImproperlyConfigured: Error loading psycopg2 module: /home/dsmr/dsmr-reader/.venv/lib/python3.9/site-packages/psycopg2/_psycopg.cpython-39-arm-linux-gnueabihf.so: undefined symbol: PQencryptPasswordConn

Any thoughts?

dennissiemensma, 2022-09-20:
You may want to check some old similar issues:
- https://github.com/dsmrreader/dsmr-reader/issues/1649#issuecomment-1164351803
- https://github.com/dsmrreader/dsmr-reader/issues/1577#issue-1129117700

Can you try those suggestions?

Djjvb, 2022-09-28:
@dennissiemensma Thanks for asking. They issues were similar. I had the postgresql version  of 9.6. Initially I saw  no version, so I thought it didn't apply to me. I am a real Noob. So I totally rely on your great tutorials and copy paste. But have no clue as to what i am doing. 

So I've finally managed to upgrade  my raspberry os to 10 (thinking it would upgrade the postgresql  also), but during the installation I got a message that the main version 9.6 was old so I needed to upgrade. And something of a  standard 11/main cluster and  upgrading the old one.   

So now I have to figure out how to do that. And hopefully don't  destroy anything. And maybe I have to restore  an old database? I am really driving blind here. I've really got no linux/programming experience. So I've to google everything and hope for the best.

dennissiemensma, 2022-09-28:
In all cases logs are your friends. Most are written to `/var/log/*`, e.g. `/var/log/postgresql/` and `/var/log/supervisor/`.

Upgrading Postgres should be fine, but for any database upgrade, no matter the vendor, make sure to make a complete SQL dump first. Both for backup and in some cases for the upgrade. 

For example:
```
sudo su - postgres
pg_dump -d dsmrreader | gzip --fast > dsmrreader-export.sql.gz
```

But this will only work when Postgres runs.

DSMR-reader has a "flat" DB-structure with only basic SQL features, so most dumps should be able to carry over to any upgraded DB versions.

If Postgres runs, just check whether the DB still exists by running:

```
sudo su - postgres
psql --list
```

E.g.:
```
postgres@rpi4:~$ psql --list
                                   List of databases
    Name    |   Owner    | Encoding |   Collate   |    Ctype    |   Access privileges   
------------+------------+----------+-------------+-------------+-----------------------
 dsmrreader | dsmrreader | UTF8     | en_GB.UTF-8 | en_GB.UTF-8 | 
 postgres   | postgres   | UTF8     | en_GB.UTF-8 | en_GB.UTF-8 | 
 template0  | postgres   | UTF8     | en_GB.UTF-8 | en_GB.UTF-8 | =c/postgres          +
            |            |          |             |             | postgres=CTc/postgres
 template1  | postgres   | UTF8     | en_GB.UTF-8 | en_GB.UTF-8 | =c/postgres          +
            |            |          |             |             | postgres=CTc/postgres
(4 rows)

```

dennissiemensma, 2022-09-28:
Also, if you ever need to upgrade again (not now), consider using the containerized/Docker version of DSMR-reader. 

It allows you to run any version of DSMR-reader/DB no matter the OS version. You can then upgrade either or both, but does not require the OS to update.

dennissiemensma, 2022-10-10:
@Djjvb any progress?