dsmrreader/dsmr-reader #1708: Mogelijkheid tot definiëren prijsverschil voor verbruik 'tot' en gebruik 'vanaf'

JBS5, 2022-09-21:
### Feature

Gezien het aangekondigde prijsplafond voor gas en elektriciteit, zou het mooi zijn als je in DSMR onderscheid kunnen maken tussen het tarief voor verbruik _tot_ en het tarief _vanaf_ verbruik X. 
Dit geldt zowel voor elektriciteit als gas.

dennissiemensma, 2022-09-21:
Bedankt voor je suggestie. Dit project beperkt zich tot eenvoudige dagprijzen om je een globale indicatie te geven hoeveel een dag ongeveer kost. Alle andere prijsberekeningen, kortingen, subsidies, regelingen, etc blijven daardoor bewust buiten de scope van dit project. Ik denk dat je het beste vanaf 1 januari je contract in DSMR-reader kan dupliceren en stoppen/starten op die dag tot de oorspronkelijke looptijd, met eventuele plafond prijzen, mocht dat voor jou van toepassing zijn. Je kunt dan relatief makkelijk zien hoeveel je vanaf de eerste dag van die regeling hebt verbruikt. Mits het überhaupt op die manier gaat werken, gezien de details nog onbekend zijn.

JBS5, 2022-09-21:
> Bedankt voor je suggestie. Dit project beperkt zich tot eenvoudige dagprijzen om je een globale indicatie te geven hoeveel een dag ongeveer kost. Alle andere prijsberekeningen, kortingen, subsidies, [...]

Oké, duidelijk, dank voor je reactie. Praktisch gezien zou je dan, bij het bereiken van het gestelde plafond waarna de normale c.q. marktprijs gaat gelden, een nieuw contract moeten starten. Kwestie van even in de gaten houden dus.

closed 2022-09-21 by dennissiemensma

ramondeklein, 2022-10-24:
> Dit project beperkt zich tot eenvoudige dagprijzen om je een globale indicatie te geven hoeveel een dag ongeveer kost.

Ik wou al een issue openen, waarbij er een vorm van ondersteuning komt voor de nieuwe dynamische providers zoals Frank Energy, EasyEnergy, Tibber, ... Maar dat wordt denk ik helemaal complex, dus neem aan dat dit ook buiten scope valt.

dennissiemensma, 2022-10-24:
> Maar dat wordt denk ik helemaal complex, dus neem aan dat dit ook buiten scope valt.

Ja klopt, plus dat ik eigenlijk vind dat je van dergelijke leveranciers tegenwoordig mag verwachten dat ze jou zelf via een app of API of iets de mogelijkheid geven om live inzicht te geven in kosten. 

Het heeft verder ook weinig zin om die prijberekeningen te gaan spiegelen in eigen software. De leveranciers mogen in deze tijd ook wel wat innoveren, zodat je zelf weet wanneer het tarief goedkoop is (handmatig of geautomatiseerd).