dsmrreader/dsmr-reader #1709: 🚨 DSMRreader keeps queueing MQTT messages even with MQTT disabled

dekimpew, 2022-09-23:
### Description

I used to push MQTT messages to a broker for further use, but the MQTT broker has been shut down. I disabled MQTT in the dsmr admin page, but it still queues up messages up until the maximum amount of 5000 messages.

When I delete the messages in the pgsql database, I see it immediately starts filling the queue again:
```
dsmrreader=> delete from dsmr_mqtt_message;
DELETE 5000
dsmrreader=> delete from dsmr_mqtt_message;
DELETE 5000
dsmrreader=> delete from dsmr_mqtt_message;
DELETE 66
dsmrreader=> delete from dsmr_mqtt_message;
DELETE 33
```

In the admin page, MQTT is disabled and the hostname field is empty. It does not allow me to empty the Client ID field (might be intentional?)

- Restarting the software doesn't disable MQTT.
- Re-enabling and disabling MQTT again doesn't fix it.
- The issue originated in v5.5 and carried over into 5.6.

(Please ignore the day statistics error, my gas meter broke a few days ago)

### DSMR-reader version

5.6

### DSMR-reader platform

Native (e.g. manual installation)

### Debug info dump

```shell
DSMR-READER
    App / Python / Database                                                        v5.6 / v3.9.2 / postgresql
    BE sleep / DL sleep / Retention / Override                                     1.0s / 5.0s / 672h / False
    Latest telegram version read / Parser settings                                             "None" / "101"

DATA
    Telegrams total (est.)                                                                            2424871
    Consumption records electricity / gas (est.)                                                52542 / 20145

UNRESOLVED ISSUES
    Day statistics are lagging behind                                                     1 day, 10 hours ago

POSTGRESQL SIZE OF LARGEST TABLES (> 500 MB)
    public.dsmr_datalogger_dsmrreading                                                                 670 MB
```

dennissiemensma, 2022-09-23:
Thanks for reporting, did you also disable all data sources in the admin? There is likely one still active, generating messages. The broker settings do not affect the data sources, as one may or may not intentionally disable it temporarily, hence the queue mechanism as well.

dennissiemensma, 2022-09-27:
@dekimpew did disabling the data source fix it?

dekimpew, 2022-09-28:
@dennissiemensma i'll be able to check the result on friday. There were some sources still enabled like you said.
That said, I would expect the "mqtt enable" button to be some kind of global on/off switch but its good to know its more granular.

dennissiemensma, 2022-09-28:
It can work both ways, but in DSMR-reader it's configured separately, to prevent broker downtime from interrupting the data flow entirely.

closed 2022-09-29 by dennissiemensma