dsmrreader/dsmr-reader #1710: DSMR v2.2, null value in column

git1k2, 2022-09-24:
### Description

Hoi,

Ik heb een DSMR v2.2 en gebruik een slimmelezer (ipv4 verbinding naar esp-link). Nu zie ik langzaam grafieken ontstaan, maar ook veel fouten in de logging.

Iemand een idee waar dit aan ligt?

Foutmelding:
```
django.db.utils.IntegrityError: null value in column "electricity_delivered_1" of relation "dsmr_datalogger_dsmrreading" violates not-null constraint
DETAIL:  Failing row contains (83, 2022-09-24 18:04:06.522431+00, null, 0.000, 26257.913, 0.000, 0.580, 0.000, 2022-09-24 18:00:00+00, 10197.528, f, null, null, null, null, null, null, null, null, null, null, null, null).
```

Telegram:
```
/XMX5XMXABCE000046867

0-0:96.1.1(***redacted***)
1-0:1.8.1(28877.601*kWh)
1-0:1.8.2(26257.913*kWh)
1-0:2.8.1(00000.000*kWh)
1-0:2.8.2(00000.000*kWh)
0-0:96.14.0(0001)
1-0:1.7.0(0000.60*kW)
1-0:2.7.0(0000.00*kW)
0-0:96.13.1()
0-0:96.13.0()
0-1:96.1.0(***redacted***)
0-1:24.1.0(03)
0-1:24.3.0(220924190000)(2C)(60)(1)(0-1:24.2.0)(m3)
(10197.528)
!
```

### DSMR-reader version

5.6

### DSMR-reader platform

Docker (e.g. Xirixiz)

dennissiemensma, 2022-09-24:
Zie eerst https://github.com/dsmrreader/dsmr-reader/issues/1282#issue-805921399

git1k2, 2022-09-25:
Dank voor je hulp! Heb het gelezen en geprobeerd, maar dit loste het niet op. Het lijkt eerder op een probleem met het telegram zelf. 

In het console van esp-link zie ik het volgende telegram:
```
...
1-0:1.8.1(28883.658*kWh)
1-0:1.8.2(26257.913*kWh)
1-0:2.8.1(00000.000*kWh)
1-0:2.8.2(00000.000*kWh)
...
```
Maar bekijk ik dit met netcat / telnet, gaat er iets mis bij 1-0:1.8.1:
```
...
1-0:1.8.183.658*kWh)
1-0:1.8.2(26257.913*kWh)
1-0:2.8.1(00000.000*kWh)
1-0:2.8.2(00000000*kWh)
...
```

Ik ga even verder met zoeken.

git1k2, 2022-09-25:
Heb een RPi met ser2net geconfigureerd in plaats van de ESP met esp-link en dan werkt het prima! Geen dsmr-reader probleem dus. Ik ga later nog even uitzoeken waarom de ESP in het console wel goede data ziet, maar over het netwerk tekens mist.

ser2net config voor DSMR v2.2:
```
connection: &con0096
    accepter: tcp,23
    enable: on
    options:
      max-connections: 10
      telnet-brk-on-sync: true
    connector: serialdev,
              /dev/serial/by-id/usb-FTDI_USB_HS_SERIAL_CONVERTER_FTVW1XQK-if00-port0,
              9600e71,local
```

closed 2022-09-25 by git1k2

dennissiemensma, 2022-09-25:
Oke dank voor de terugkoppeling!