dsmrreader/dsmr-reader #1713: 🙋dsmr_backend/datalogger/webinterface does not run longer than 4s. keeps restarting

Jasper-GIT, 2022-09-28:
### Description

Get "Bad Gateway 502" backend/datalogger/webinterface processes does not run longer than 4s. 
Was trying to update python 3.6.9 to 3.9.x but the make process stuck. At the end, no python update and still on the 3.6.9

### DSMR-reader version

4.1x

### DSMR-reader platform

Other

### Debug info dump

```shell
pi@raspberrypi:~ $ sudo supervisorctl status
dsmr_backend                     RUNNING   pid 4629, uptime 0:00:04
dsmr_datalogger                  STARTING
dsmr_webinterface                STARTING
pi@raspberrypi:~ $ sudo supervisorctl status
dsmr_backend                     RUNNING   pid 4647, uptime 0:00:03
dsmr_datalogger                  STARTING
dsmr_webinterface                STARTING
pi@raspberrypi:~ $ sudo supervisorctl status
dsmr_backend                     STARTING
dsmr_datalogger                  RUNNING   pid 4675, uptime 0:00:01
dsmr_webinterface                RUNNING   pid 4669, uptime 0:00:03
pi@raspberrypi:~ $ sudo supervisorctl status
dsmr_backend                     STARTING
dsmr_datalogger                  RUNNING   pid 4692, uptime 0:00:03
dsmr_webinterface                RUNNING   pid 4686, uptime 0:00:04
pi@raspberrypi:~ $ sudo supervisorctl status
dsmr_backend                     STARTING
dsmr_datalogger                  RUNNING   pid 4709, uptime 0:00:02
dsmr_webinterface                RUNNING   pid 4703, uptime 0:00:03
pi@raspberrypi:~ $ sudo supervisorctl status
dsmr_backend                     RUNNING   pid 5436, uptime 0:00:02
dsmr_datalogger                  RUNNING   pid 5433, uptime 0:00:03
dsmr_webinterface                RUNNING   pid 5426, uptime 0:00:03
```

Jasper-GIT, 2022-09-28:
Nu lijkt hij helemaal kapot.

`
pi@raspberrypi:~ $ sudo supervisorctl
dsmr_backend                     BACKOFF   Exited too quickly (process log may have details)
dsmr_datalogger                  BACKOFF   Exited too quickly (process log may have details)
dsmr_webinterface                FATAL     can't find command '/home/dsmr/.virtualenvs/dsmrreader/bin/gunicorn'
`

dennissiemensma, 2022-09-28:
Bedankt voor je melding. Welke DSMR-reader draai je? En probeerde je alleen de Python-versie te upgraden?

DSMR-reader draait in een zogenaamde virtualenv, wat gewoon een kopie is van whatever Python je gebruikte bij de installatie. Na de Python upgrade moet je DSMR-reader stoppen, de oude virtualenv verplaatsen (of verwijderen als je zeker van je zaak bent), een nieuwe virtualenv maken en dan een deel van de installatie opnieuw doen (de "pip install" voor alle packages in de nieuwe virtualenv).

```
FATAL can't find command '/home/dsmr/.virtualenvs/dsmrreader/bin/gunicorn'
```
geeft aan dattie 1 van de packages (gunicorn) niet kan vinden.

---

Wat krijg je te zien als je dit doet:
```
sudo su - dsmr

python --version
pip freeze
```

Jasper-GIT, 2022-09-28:
Bedankt voor je reactie. Ik was bezig met de upgrade naar 5.x maar bij de eerste stap, het upgraden van python ging het al mis.
python --version geeft 2.7.13 aan. Wat ik vreemd vind, want voordat ik een reboot deed, gaf hij nog 3.6.9 Hoe pas ik dit aan?
pip freeze geeft:
gunicorn==20.1.0
psycopg2-binary==2.8.6
six==1.15.0

dennissiemensma, 2022-09-28:
Oke, wellicht draait Python 2 nog, niet erg, wat geeft:

```
sudo su - dsmr

python3 --version
```

Jasper-GIT, 2022-09-28:
python3 --version geeft:

Python 3.6.9

dennissiemensma, 2022-09-28:
En wat geeft dit allemaal?

```
sudo su - dsmr

which python3
which python3.9
ls -l ~/dsmr-reader/.venv/bin/
```

Jasper-GIT, 2022-09-28:
which python3: /usr/local/bin/python3
which python3.9: Niks want bij de make bleef de pi hangen

ls -l ~/dsmr-reader/.venv/bin/:

-rw-r--r-- 1 dsmr dsmr 2206 sep 28 20:23 activate
-rw-r--r-- 1 dsmr dsmr 1262 sep 28 20:23 activate.csh
-rw-r--r-- 1 dsmr dsmr 2426 sep 28 20:23 activate.fish
-rwxr-xr-x 1 dsmr dsmr  258 sep 28 20:08 easy_install
-rwxr-xr-x 1 dsmr dsmr  258 sep 28 20:08 easy_install-3.6
-rwxr-xr-x 1 dsmr dsmr  240 sep 28 20:09 pip
-rwxr-xr-x 1 dsmr dsmr  240 sep 28 20:09 pip3
-rwxr-xr-x 1 dsmr dsmr  240 sep 28 20:09 pip3.6
lrwxrwxrwx 1 dsmr dsmr    7 sep 28 20:08 python -> python3
lrwxrwxrwx 1 dsmr dsmr   22 sep 28 20:08 python3 -> /usr/local/bin/python3

dennissiemensma, 2022-09-28:
Ik denk dat je die make nogmaals moet proberen, want Python 3.6 gaat sowieso niet werken voor de onderliggende software waar DSMR-reader weer op gebouwd is. 

Je kunt nog een keertje Python 3.9 proberen (eventueel 3.9.14 ipv 3.9.13 van de tutorial) en anders is 3.8 ook wel prima. Al zou het niet veel uit moeten maken qua installatie.

### 3.9.14
```
sudo apt-get install python3-dev libffi-dev libssl-dev -y
wget https://www.python.org/ftp/python/3.9.14/Python-3.9.14.tar.xz
tar xJf Python-3.9.14.tar.xz
cd Python-3.9.14
./configure --enable-optimizations --with-lto
make
sudo make install
sudo pip3 install --upgrade pip
```

### Of 3.8.9
```
sudo apt-get install python3-dev libffi-dev libssl-dev -y
wget https://www.python.org/ftp/python/3.8.9/Python-3.8.9.tar.xz
tar xJf Python-3.8.9.tar.xz
cd Python-3.8.9
./configure --enable-optimizations --with-lto
make
sudo make install
sudo pip3 install --upgrade pip
```

dennissiemensma, 2022-09-28:
Mocht een van beide lukken, verwijder dan de vorige nieuwe virtualenv:
```
/home/dsmr/dsmr-reader/.venv/
```

En check of je echt de goede Python versie hebt onder `python3 --version`. Al de nieuwe Python versie wel geinstalleerd is, maar niet de default, doe dan zoiets bij het aanmaken van de virtualenv:

```
# NIET
python3 -m venv ~/dsmr-reader/.venv/

# maar
python3.8 -m venv ~/dsmr-reader/.venv/
# of
python3.9 -m venv ~/dsmr-reader/.venv/
```

Daarna hoef je binnen die virtualenv niet meer die specifieke versie aan te geven. Maar als het goed is hoeft dat sowieso niet.

Jasper-GIT, 2022-09-28:
Vreemd want tot vanmorgen draaide DSMR gewoon op python 3.6.9 update naar 5.x heb ik nog niet gedaan

Jasper-GIT, 2022-09-28:
Ik ga dan even 3.9.14 proberen. niet in sudo su - dsmr toch?

dennissiemensma, 2022-09-28:
> Vreemd want tot vanmorgen draaide DSMR gewoon op python 3.6.9 update naar 5.x heb ik nog niet gedaan

Alleen voor DSMR-reader v5.x is die hogere Python-versie nodig. Of liever gezegd, voor het onderliggende Django wat ook geupgrade is in die DSMR-reader upgrade. En die ondersteunt niet meer die oude Python versie.

----

> Ik ga dan even 3.9.14 proberen. niet in sudo su - dsmr toch?

Nee klopt, gewoon als pi (sudo) user of whatever je standaard user op je Pi is.

Jasper-GIT, 2022-09-28:
> Alleen voor DSMR-reader v5.x is die hogere Python-versie nodig. Of liever gezegd, voor het onderliggende Django wat ook geupgrade is in die DSMR-reader upgrade. En die ondersteunt niet meer die oude [...]

Ik draai nog altijd DSMR v4.1x, hier moet die python ook voor werken? Wellicht dat ik later deze week dan upgrade naar 5.x

dennissiemensma, 2022-09-28:
Als je nu nog op v4 zit dan hoef je niet die Python versie te upgraden dacht ik. Het is sowieso aan te raden om eerst naar v4.20 te gaan.

dennissiemensma, 2022-09-28:
Los van dit verhaal kun je trouwens overwegen om een volgende keer over te stappen op de Docker-versie van DSMR-reader. Scheelt een hoop installatie-gedoe, helemaal bij major upgrades van DSMR-reader.

Al is werken met Docker ook niet mogelijk zonder enige kennis.

dennissiemensma, 2022-09-28:
Wellicht heb ik je verhaal dan verkeerd begrepen. Als je vorige installatie niet meer lekker werkte, kun je het beste in de Supervisor logs kijken van DSMR-reader voor een oorzaak.

dennissiemensma, 2022-09-28:
In het ergste geval kun je je v4 virtualenv opnieuw aanmaken, als de vorige stuk is gegaan door de mislukte Python upgrade.

https://dsmr-reader.readthedocs.io/nl/v4/how-to/installation/quick.html

```
sudo -u dsmr mkdir /home/dsmr/.virtualenvs
sudo -u dsmr virtualenv /home/dsmr/.virtualenvs/dsmrreader --system-site-packages --python python3
sudo -u dsmr /home/dsmr/.virtualenvs/dsmrreader/bin/pip3 install -r /home/dsmr/dsmr-reader/dsmrreader/provisioning/requirements/base.txt
```

De oude `.virtualenvs` map kun je dan even hernoemen of iets. En het derde command zou alles moeten installeren qua packages voor whatever v4-versie je lokaal hebt staan.

Jasper-GIT, 2022-09-28:
> Wellicht heb ik je verhaal dan verkeerd begrepen. Als je vorige installatie niet meer lekker werkte, kun je het beste in de Supervisor logs kijken van DSMR-reader voor een oorzaak.

Ja, dat denk ik. want ik heb nog niets geupdate, en na de poging om python 3.9 te installeren ging het mis. 

Hier bijvoorbeeld de log van de webinterface: 

`==> /var/log/supervisor/dsmr_webinterface.log <==
    from gunicorn.app.wsgiapp import run
ModuleNotFoundError: No module named 'gunicorn'
Traceback (most recent call last):
  File "/home/dsmr/.virtualenvs/dsmrreader/bin/gunicorn", line 5, in <module>
    from gunicorn.app.wsgiapp import run
ModuleNotFoundError: No module named 'gunicorn'
Traceback (most recent call last):
  File "/home/dsmr/.virtualenvs/dsmrreader/bin/gunicorn", line 5, in <module>
    from gunicorn.app.wsgiapp import run
ModuleNotFoundError: No module named 'gunicorn'

==> /var/log/supervisor/dsmr_webinterface.log.1 <==
    return getattr(connections[DEFAULT_DB_ALIAS], item)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/db                                                                                                                                                             /utils.py", line 211, in __getitem__
    self.ensure_defaults(alias)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/db                                                                                                                                                             /utils.py", line 172, in ensure_defaults
    conn = self.databases[alias]
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/ut                                                                                                                                                             ils/functional.py", line 48, in __get__
    res = instance.__dict__[self.name] = self.func(instance)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/db                                                                                                                                                             /utils.py", line 160, in databases
    if DEFAULT_DB_ALIAS not in self._databases:
TypeError: argument of type 'NoneType' is not iterable

==> /var/log/supervisor/dsmr_webinterface.log.2 <==
    return getattr(connections[DEFAULT_DB_ALIAS], item)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/db                                                                                                                                                             /utils.py", line 211, in __getitem__
    self.ensure_defaults(alias)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/db                                                                                                                                                             /utils.py", line 172, in ensure_defaults
    conn = self.databases[alias]
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/ut                                                                                                                                                             ils/functional.py", line 48, in __get__
    res = instance.__dict__[self.name] = self.func(instance)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/db                                                                                                                                                             /utils.py", line 160, in databases
    if DEFAULT_DB_ALIAS not in self._databases:
TypeError: argument of type 'NoneType' is not iterable
==> /var/log/supervisor/dsmr_webinterface.log.3 <==
    return getattr(connections[DEFAULT_DB_ALIAS], item)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/db                                                                                                                                                             /utils.py", line 211, in __getitem__
    self.ensure_defaults(alias)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/db                                                                                                                                                             /utils.py", line 172, in ensure_defaults
    conn = self.databases[alias]
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/ut                                                                                                                                                             ils/functional.py", line 48, in __get__
    res = instance.__dict__[self.name] = self.func(instance)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/db                                                                                                                                                             /utils.py", line 160, in databases
    if DEFAULT_DB_ALIAS not in self._databases:
TypeError: argument of type 'NoneType' is not iterable

==> /var/log/supervisor/dsmr_webinterface.log.4 <==
    return getattr(connections[DEFAULT_DB_ALIAS], item)
[... cut ...]
[2022-09-28 15:38:36 +0200] [8306] [INFO] Reason: Worker failed to boot.
[2022-09-28 15:38:38 +0200] [8317] [INFO] Starting gunicorn 20.0.4
[2022-09-28 15:38:38 +0200] [8317] [INFO] Listening at: unix:/var/tmp/gunicorn--                                                                                                                                                             dsmr_webinterface.socket (8317)
[2022-09-28 15:38:38 +0200] [8317] [INFO] Using worker: sync
[2022-09-28 15:38:38 +0200] [8322] [INFO] Booting worker with pid: 8322
`

Jasper-GIT, 2022-09-28:
Nog altijd die gunicorn die hij mist

dennissiemensma, 2022-09-28:
Ja zie mijn vorige comment. Ik denk dat dat beter is: https://github.com/dsmrreader/dsmr-reader/issues/1713#issuecomment-1261420053

dennissiemensma, 2022-09-28:
Dit gaat er wel vanuit dat je verder nog niets had aangepast voor de upgrade naar v5

Jasper-GIT, 2022-09-28:
> Ja zie mijn vorige comment. Ik denk dat dat beter is: [#1713 (comment)](https://github.com/dsmrreader/dsmr-reader/issues/1713#issuecomment-1261420053)

top, ga ik dat proberen.

Log van de datalogger:

`==> /var/log/supervisor/dsmr_datalogger-stderr---supervisor-ywefQC.log <==
    from django.core.management import execute_from_command_line
ModuleNotFoundError: No module named 'django'
Traceback (most recent call last):
  File "/home/dsmr/dsmr-reader/manage.py", line 8, in <module>
    from django.core.management import execute_from_command_line
ModuleNotFoundError: No module named 'django'
Traceback (most recent call last):
  File "/home/dsmr/dsmr-reader/manage.py", line 8, in <module>
    from django.core.management import execute_from_command_line
ModuleNotFoundError: No module named 'django'

`

Jasper-GIT, 2022-09-28:
sudo -u dsmr mkdir /home/dsmr/.virtualenvs

Dit kan ik niet uitvoeren want hij bestaat al, hier dus renamen ?

dennissiemensma, 2022-09-28:
Gewoon die hele bestaande map hernoemen:

```
sudo mv /home/dsmr/.virtualenvs /home/dsmr/OLD-.virtualenvs
```

Jasper-GIT, 2022-09-28:
Bij t 3e commando uit je post gaat t mis:

sudo -u dsmr /home/dsmr/.virtualenvs/dsmrreader/bin/pip3 install -r /home/dsmr/dsmr-reader/dsmrreader/provisioning/requirements/base.txt
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting https://qpypi3.qpython.org/repository/1745924/PyCRC-1.21.tar.gz (from -r /home/dsmr/dsmr-reader/dsmrreader/provisioning/requirements/base.txt (line 20))
ERROR: Could not install packages due to an OSError: HTTPSConnectionPool(host='qpypi3.qpython.org', port=443): Max retries exceeded with url: /repository/1745924/PyCRC-1.21.tar.gz (Caused by ResponseError('too many 500 error responses',))

dennissiemensma, 2022-09-28:
Welke DSMR-reader versie heb je exact?

Jasper-GIT, 2022-09-28:
waar vind ik dat nu nog?

dennissiemensma, 2022-09-28:
```
cat /home/dsmr/dsmr-reader/dsmrreader/__init__.py
```

dennissiemensma, 2022-09-28:
Dat verklaart wel een hoop. Ik dacht dat je v4.x draaide. In v3.x is namelijk pas deze fout opgelost: https://github.com/dsmrreader/dsmr-reader/issues/1713#issuecomment-1261427120

Je kunt checken wat voor branch je draait:
```
sudo su - dsmr

git branch
```

Jasper-GIT, 2022-09-28:
dsmr@raspberrypi:~/dsmr-reader$ git branch
* master
  v4

dennissiemensma, 2022-09-28:
```
git checkout v4
```

En probeer het daarna opnieuw (check de versie daarna ook even)

Jasper-GIT, 2022-09-28:
`
dsmr@raspberrypi:~/dsmr-reader$ git checkout v4
Switched to branch 'v4'
Your branch is up-to-date with 'origin/v4'.
dsmr@raspberrypi:~/dsmr-reader$ git branch
  master
* v4
`

dennissiemensma, 2022-09-28:
```
cat /home/dsmr/dsmr-reader/dsmrreader/init.py
```

Bedoelde ik

Jasper-GIT, 2022-09-28:
> cat /home/dsmr/dsmr-reader/dsmrreader/init.py

pi@raspberrypi:/ $ cat /home/dsmr/dsmr-reader/dsmrreader/init.py
cat: /home/dsmr/dsmr-reader/dsmrreader/init.py: Bestand of map bestaat niet

Jasper-GIT, 2022-09-28:
jaa, dat is m, sorry. Had ik ook kunnen weten. Ik kopieerde t :)

VERSION = (4, 11, 0, 'final', 0)

__version__ = get_version(VERSION)

Jasper-GIT, 2022-09-28:
nu werkt t 3e commando uit je eerdere post wel. Die loopt nu

Jasper-GIT, 2022-09-28:
Nu is alles netjes running, maar webserver laat niks zien als ik in de browser er naar toe ga.

dennissiemensma, 2022-09-28:
Ik denk dat je die Python upgrade maar moet laten voor wat het is en v4 weer werkend krijgen. Eventueel een andere dag naar v4.20 updaten (zou zonder issues moeten kunnen).

----

En dan op een ander moment gaan verdiepen in https://github.com/dsmrreader/dsmr-reader/issues/1713#issuecomment-1261416637, want ik denk dat het je een hoop gedoe scheelt als je eenmaal op Docker zit. Ook voor toekomstige updates.

Jasper-GIT, 2022-09-29:
Bedankt voor je hulp, inderdaad is het nu werkend na de reboot van de pi.
Op een of andere wijze blijft de make van python 3.9.x hangen na een lang tijd. Kernel panic?
Na reboot werkt alles wel naar behoren.
Ik zal me eens verdiepen in Docker, bedankt zover.

dennissiemensma, 2022-09-29:
Ja, ik zou zelfs in dat geval de database laten voor wat het is, totdat je die ooit moet upgraden. Dan blijft je data helemaal wat het is

Want Docker is puur een manier van hosten. Dus je kunt zowel de DSMR-reader applicatie als de database hosten via Docker, maar het kan ook 1 van beiden. Dus als je een nieuwe DSMR-reader tegen je huidige DB aan laat praten (wel je huidige DSMR-reader dan stoppen), dan zorgt de Docker image voor DSMR-reader als het goed is dat de migratie gedaan wordt.

Zorg er wel voor dat je altijd een DB backup maakt, gewoon voor het geval dat. En lees de changelog bij grote upgrades, zoals van v4 naar v5, want soms moet je nog wat extra's doen. In het geval van v5 is bijvoorbeeld de koppeling met Dropbox veranderd, mocht je die gebruiken.

closed 2022-09-29 by dennissiemensma

dennissiemensma, 2022-09-29:
Het Docker project voor DSMR-reader staat trouwens hier apart: https://github.com/xirixiz/dsmr-reader-docker
Het is dus vooral het bouwen van de images, DSMR-reader inhoudelijk komt gewoon uit het huidige project. Xirixiz heeft daar omheen alles geschreven wat nodig is om het via Docker te hosten.

Bij nieuwe releases maakt hij een nieuwe versie van de DSMR-reader docker image. Die komt dan weer beschikbaar op zijn Docker Hub en is te gebruiken voor je lokale installatie als update.

Er komt een moment dat DSMR-reader alleen nog maar de Docker-hosting ondersteunt. Want je merkt nu al hoe lastig het kan zijn om simpelweg Python of een virtualenv werkend te krijgen wanneer het niet soepel gaat. En dat is eigenlijk zonde van de tijd. En bij de images via Docker is het doorgaans 1 groot geheel wat wel of niet werkt in de basis.