dsmrreader/dsmr-reader #1715: Kostenberekening anders dan bij energieleverancier🚨

ridderr, 2022-10-01:
### Description

De DSMR software rekent voor de energiekosten: verbruik x leverkosten - teruglevering x terugleververgoeding. 
Bij Vattenfall wordt als volgt gerekend: verbruik - teruglevering. Indien positief x leverkosten, en indien negatief x terugleververgoeding.

Op de DSMR manier wordt er bij mij voor deze maand uitgerekend dat ik 2 euro retour krijg en bij vattenfall zie in in het overzicht dat ik 20 euro terugkrijg.

Het lijkt me ook logisch dat de berekening gebaseerd is op het verschil tussen gebruik en levering en dan vermenigvuldigd met kosten of vergoeding. Beide bedragen zijn natuurlijk ook afhankelijk van dal/normaal tarief. 

### DSMR-reader version

5.6

dennissiemensma, 2022-10-01:
Bedankt voor je melding. DSMR-reader heeft niet als doel om je energiemaatschappij te spiegelen qua kostenberekening. Daar is namelijk geen beginnen aan en kun je een compleet apart project aan wijden. Het doel is altijd geweest om je puur een indicatie te geven hoeveel een dag ongeveer kost, bijv een herfstdag vs winterdag of een weekdag vs weekenddag.

ridderr, 2022-10-01:
Hoi Dennis, dat begrijp ik. Maar in de huidige status van de wereld is het wel fijn om de kosten te kunnen zien. Ik kan natuurlijk inloggen bij Vattenfall maar het is natuurlijk mooi als DSRM een beetje in de range ligt van de leveranciers.

dennissiemensma, 2022-10-01:
Realiseer je dat er veel meer leveranciers zijn, met allemaal hun eigen methodes. En ook buitenlandse, voor bijvoorbeeld zuiderburen of Zweedse gebruikers. 
Om nog maar te zwijgen over energiebelasting, BTW, kortingen, jaarwisselingen, netbeheerderskosten en in deze wereld leveranciers die op elk moment hun beleid aanpassen. Bijvoorbeeld voor salderen. En dan moet alles ook nog met terugwerkende kracht gelden voor de regels van dat moment in de historie. Het gaat nooit in de buurt komen van wat er geldt.

Vandaar dat je daar een compleet apart voltijd project aan kan wijden. Dat is echt niet te doen. De enige waarheid is de energieleverancier. 

Je zou kunnen vragen of ze aan hun kant wellicht wat meer innovatie doen. Zoals dagelijkse alerts, of totaalafrekening-updates, etc etc. In deze tijden mag je ook wel wat verwachten aan hun zijde, ten gunste van voorkomen dat klanten verrast worden.

closed 2022-10-01 by dennissiemensma