dsmrreader/dsmr-reader #1722: 💡Kwartier piek niet berekenen op basis van kwartieren maar "sliding"

ramonsmits, 2022-10-04:
### Feature

Ik heb naar https://github.com/dsmrreader/dsmr-reader/issues/1084 gekeken en naar de grafiek. Daar zie ik dat er per kwartier van het uur wordt gekeken (00-15, 15-30, 30-45, 45-00/60). Echter, dit lijkt mij incorrect. Stel dat de piek ligt van 07-22 ?

Wellicht is de huidige implementatie goed genoeg maar het zou kunnen dat de daadwerkelijke piek dus hoger ligt.

Voor elke nieuw ontvangen waarde moet men de delta nemen van de laatste 15 minuten. Eenvoudig gezegd als je elke 5 seconden een fragment krijgt moet je de delta nemen tussen de huidige en die van 15 minuten (of -180 (60/5*15) records) geleden.

dennissiemensma, 2022-10-04:
Bedankt voor je melding. Op basis van wat is het incorrect? De implementatie is gebaseerd op een gebruiker die vrij veel details weet.

Het gaat dacht ik niet om de hoogste piek, maar om het gemiddelde. Het is specifiek voor Belgisch gebruikers. De intervallen zijn ook expliciet genoemd in #1084.

ramonsmits, 2022-10-05:
Ik zie daar in de comments idd dat in een reactie dat gecommuniceerd is maar dit kan ik niet vinden op officiële informatie op de vreg website.

Tenzij die informatie daadwerkelijk van hoger hand zo is bevestigd zou ik daar niet zo maar vanuit gaan dat een 1e lijns helpdesk medewerker exact weet hoe het zit.

Wellicht dan een goede vraag aan vreg om deze informatie exact te publiceren?

Met piek bedoelde ik dus de hoogste van die metingen. De meting is altijd het gemiddelde als je de (eind kWh-start kWh)*4

Zoals gezegd scheelt het in de meeste gevallen niets. B.v. als ik m'n EV ga laden zijn de sessies altijd langer dan een half uur dus zit ik meteen aan m'n max.

M.a.w. zodra een apparaat langer dan 22.5 (15 + 15/2) minuut verwarmd wat al snel alle apparaten zijn is er geen voordeel te behalen.

ramonsmits, 2022-10-05:
Tenzij alle meters het super dom meten op basis van het feit dat deze elke 15 minuten het verbruik door sturen. De berekening vind dan waarschijnlijk niet eens plaats in de meter maar op de servers van vreg? 

Maar hoe weet je welk fragment naar vreg verstuurd wordt en of deze dan wel gebruikt wordt?

dennissiemensma, 2022-10-05:
Bedankt voor je aanvulling. Ik heb voor je gezocht en het lijkt erop dat de VREG op dezelfde manier rekent met vaste momenten. Alleen hebben ze het verstopt in een filmpje en niet gewoon bij de FAQ of iets.

https://www.youtube.com/watch?v=XCfXMPpaT4Y#t=20s

[image]

Het was ergens hier verborgen in een willekeurig linkje: https://www.vreg.be/nl/nieuwe-nettarieven

dennissiemensma, 2022-10-05:
Overigens dacht ik dat het al per afgelopen 1 juli in ging, maar nu is het weer komende 1 januari ofzo. Anders konden we wellicht iemand vragen om de pieken te vergelijken met wat de VREG registreert en inzichtelijk maakt in het eigen portaal.

dennissiemensma, 2022-10-05:
@VBP8501 kun jij toevallig al in https://mijn.fluvius.be/ zien of je iets van je eigen verbruik qua pieken kan terugzien en vergelijken? Of pas vanaf 1 januari (en dan zelfs pas na paar maanden meten)?

VBP8501, 2022-10-06:
> @VBP8501 kun jij toevallig al in https://mijn.fluvius.be/ zien of je iets van je eigen verbruik qua pieken kan terugzien en vergelijken? Of pas vanaf 1 januari (en dan zelfs pas na paar maanden meten)?

Ja, dit zie je als je inlogt op Fluvius:

[image]

En ter info, dit is een andere grafiek dat je kunt opvragen over het verbruik:
[image]

[image]

ramonsmits, 2022-10-06:
Deze kan dan gesloten worden.

Overigens, ik heb zelf een beetje gekeken naar de fragmenten die mijn meter verstuurd maar dat is niet echt super nauwkeurig. Het interessante is dat de interne clock die de timestamps maakt niet exact elke 15 minuten `*/15 * * * *`) een fragment stuurt als ook dat het niet volgens de interne klok elke 5 seconden iets stuurt.

closed 2022-10-06 by ramonsmits