dsmrreader/dsmr-reader #1723: 🙋

Eppin, 2022-10-07:
### Description

Allereerst, wat een gave tool is dit! Ik ben al een tijdje aan het rondkijken om mijn huidige opstelling aan te passen. Ik maak nu nog gebruik van een RPi met software wat als .img wordt geleverd en om een lang verhaal kort te maken: ik wil graag de huidige data gaan migreren.

Hiervoor heb ik al de API + Postman documentatie ontdekt en ik heb nu een paar regels gepost via Postman en dat lijkt goed te gaan. Het enige wat mij opviel is dat in het request geen veld aanwezig is voor gas verbruik. Dat zou ik ook graag mee willen migreren en mijn vraag is dan ook: klopt het dat er geen veld voor gas-verbruik in de API is?

### DSMR-reader version

v5.8

### DSMR-reader platform

Docker (e.g. Xirixiz)

dennissiemensma, 2022-10-07:
Bedankt voor je vraag. Het veld dat je zoekt heet iets als extra device delivered (idem voor de timestamp van de gasmeting) en is optioneel. Die naam is zo omdat er niet alleen een gasmeter op de slimme meter aangesloten kan worden maar ook andere types, al ondersteunt DSMR-reader alleen gas. De andere database-tabellen zijn aggregaties van de tabel met metingen, dus die worden vanzelf gevuld en hoef je niet zelf te vullen.

Eppin, 2022-10-07:
Hartstikke bedankt voor de snelle reactie. Ik heb een paar API calls gemaakt met de `extra_device_` velden gevuld en dat werkt perfect! Aankomende week even een script maken om het eenvoudig over te kunnen zetten :-)

closed 2022-10-07 by Eppin

dennissiemensma, 2022-10-08:
Nog een paar tips

### Check welke tijdzone je meegeeft aan de metingen. 
Per meting ben je daar vrij in, want je moet zelf aangeven welke tijdzone elke datumtijd is die je stuurt. Mocht je huidige data geen tijdzone info hebben, dan is het geen ramp. Kies dan gewoon iets of zoek zelf de offsets per zomer/winter, maar zie je vermoedelijk wat kleine verschuivingen als je het vergelijkt met de oude data.

### Zet het dsmr_backend proces eventueel uit
Want DSMR-reader is niet gemaakt om al draaiende data in het verleden te importeren. Het kan zijn dat DSMR-reader bijvoorbeeld al een minuut in het verleden groepeert, terwijl nog maar de helft van de metingen erin had.

### Of reset alles na de import
Als alternatief kun je ook gewoon eerst alles importeren, daarna alle aggregaties verwijderen en DSMR-reader opnieuw alle metingen laten verwerken. 
Want dan pakt die gegarandeerd alles op volgorde en er komt immers geen oude data meer bij die die mist. Alleen nieuwe data en dat stoort niet.

- Zet dan dan retentie uit: `/admin/dsmr_datalogger/retentionsettings/`
- Zet backend process uit
- Leeg deze tabellen
```
dsmr_consumption_electricityconsumption
dsmr_consumption_gasconsumption
dsmr_stats_hourstatistics
dsmr_stats_daystatistics
dsmr_consumption_quarterhourpeakelectricityconsumption
```
- Bulk update `processed = false` voor alle metingen in `dsmr_datalogger_dsmrreading`
- Zet backend proces aan

dennissiemensma, 2022-10-11:
Bedankt voor je vraag. Het veld dat je zoekt heet iets als extra device delivered, id voor de timestamp, en is optioneel. Die naam is zo omdat er niet alleen een gasmeter op de slimme meter aangesloten kan worden, al ondersteunt DSMR-reader alleen gas.