dsmrreader/dsmr-reader #1724: 🙋Consuming unit in dashboard header

ToshY, 2022-10-09:
### Description

Hello 👋 

I've started using DSMR reader today (with [xirixiz/dsmr-reader-docker](xirixiz/dsmr-reader-docker) and Rapsberry Pi 3 model B) in order to get a better picture of what appliances are consuming a lot of power in our household. Having connected and setup everything, I found myself having **2** questions about the header in the dashboard that shows `Consuming`.

[image]

1. In the screenshot above, `Daltarief` is visible on the left, even though everything else is in English (browser language set to English). When looking at the JSON response, it also has the Dutch translation. Is this something that's untranslated or is there another reason for it?

2. While at `Daltarief` it states `Watt` as unit of power, there is no corresponding unit of time at the header of `Consuming`.
    * From a user point of view, I'm unsure what the number is I'm looking at. Could be `278 Watts per cows` for all I know. [^1]
    * From a developer point of view, looking at the JSON response having `cost_per_hour`, taking a guess that's `278 Watts per hour` seems plausible. I can't find it when searching in the documentation, so I just don't know for sure 🤷‍♂️.

I know these are nitpicky questions, but it would help me get rid of some confusion.

On a final note, the application looks really nice and functionality-wise it seems relatively easy to use. As someone who just started with this, I was looking for something hassle-free, convenient and organized. DSMR reader seems to deliver just that. Well done 👏 

[^1]: This is a reference to the teachers that (rightfully) state this over and over again when you forget to use (correct) units in your answers.

### DSMR-reader version

5.8.0

### DSMR-reader platform

Docker (e.g. Xirixiz)

dennissiemensma, 2022-10-10:
Hi, you can update the tariff names, or translate them, in the admin Frontend settings. Regarding the Watt, there is no unit of time. It's a moment in time. However, if you'd be consuming that amount of Watt for a full hour, you can see how much that would cost you, hence the cost per hour. E.g. consuming 278 Watt and keeping that up for a full hour, would result in an approximate 0.278 kWh usage.

ToshY, 2022-10-10:
Hey @dennissiemensma.

Thanks for the quick response. 

> you can update the tariff names, or translate them, in the admin Frontend settings.

Yes I found it. Thank you.

> It's a moment in time.

Does that relate to the "seconds ago" in the header? So at `Y` seconds ago, the actual power consumption was `X` Watt, and if you'd keep that up for an hour, that would be `Z` kWh.

By default it keeps refreshing every 5 seconds and I find the "seconds ago" a bit of confusing as well. If it does not exist yet, is it a possibility to add a configurable option to change this to a digital timestamp, for example `HH:mm:ss` format?

dennissiemensma, 2022-10-11:
You should be able to lower the refresh interval in the Frontend settings as well. I suggest setting it to 1 second in your case, as this will probably make it a bit easier for you to read. The timestamp is based on whatever the smart meter emits and is just a global indication for convenience. To see more details about the wattage recently, just check out its live graph.

ToshY, 2022-10-11:
Thank you for the explanations @dennissiemensma ! I have enough to go on for now. Have a great day.

closed 2022-10-11 by ToshY