dsmrreader/dsmr-reader #1727: Nieuwe slimmemeter, 3-fase DSMR leest niet uitzichzelf uit

D1StrX, 2022-10-13:
### Description

Vraagje betreft herconfigureren (denk bus snelheden) voor het uitlezen van de slimme meter.
Heb een 3-fase slimmemeter gekregen, waarvan de chineese leverancier op de website zegt dat de usb -> p1 kabel zou moeten werken. 
DSMR daarintegen leest mn nog niet uit. Hoe is dit het beste te testen/fixen?

### DSMR-reader version

v5.8

### DSMR-reader platform

Native (e.g. manual installation)

dennissiemensma, 2022-10-13:
Je kunt het beste "cu" installeren en buiten DSMR-reader om je meter proberen te lezen. Stop DSMR-reader wel even om gekke fouten te voorkomen. Zie deze specifieke stap: https://dsmr-reader.readthedocs.io/nl/v5/tutorial/installation/step-by-step.html#optional-your-first-reading

D1StrX, 2022-10-13:
positieve is dat die goede reading laat zien. Maar dsmr laat nu al -140000 watt zien, dus moet ergens wat mis gegaan zijn. Prima als dat de teruglevering zou zijn.. 😅 
Wat kunnen we nog meer doen? (naast een backup terugzetten of clean install)

dennissiemensma, 2022-10-13:
Als je meter verwisseld is, zul je een dag moeten wachten en daarna eenmalig met de hand herstellen. Zie https://dsmr-reader.readthedocs.io/nl/v5/how-to/troubleshooting/smart-meter-replacement.html

D1StrX, 2022-10-15:
Het draait weer alsof er niets gebeurd is. De rest van de dag de p1 eruit (ge)laten, en daarna inderdaad handmatig fixen.  P1 er terug in, en alles werkte weer.

closed 2022-10-15 by D1StrX