dsmrreader/dsmr-reader #1728: 🚨

vincedh, 2022-10-13:
### Description

Hoi Dennis,
Het lijkt erop dat mqtt value 'electricity_tariff = dsmr/meter-stats/electricity_tariff' niet meer goed wordt doorgegeven. Alle waardes uit dsmr reader worden keurig weergegeven in home assistant, behalve deze.
Is daar onlangs iets in veranderd?
Ik hoor 't wel als ik iets kan proberen.
Bedankt alvast,
Vince

### DSMR-reader version

5.8

### Debug info dump

```shell
DSMR-READER
    App / Python / Database                                                       v5.8 / v3.8.10 / postgresql
    BE sleep / DL sleep / Retention / Override                                    1.0s / 0.5s / 4032h / False
    Latest telegram version read / Parser settings                                                 "50" / "4"

DATA
    Telegrams total (est.)                                                                           14581892
    Consumption records electricity / gas (est.)                                               292342 / 97705

UNRESOLVED ISSUES
    Database groeit gestaag: 5236 MB, overweeg dataopschoning (indien nog niet ingeschakeld)                                      nu

POSTGRESQL SIZE OF LARGEST TABLES (> 500 MB)
    public.dsmr_datalogger_dsmrreading                                                                4919 MB
```

dennissiemensma, 2022-10-13:
Je kunt het beste DEBUG level logging tijdelijk aanzetten en in de logs van het backend proces kijken wat er gebeurt qua MQTT:
- https://github.com/dsmrreader/dsmr-reader/issues/1714#issuecomment-1261813560

Voor zover ik weet is er niets aan veranderd en het veld heet ook nog steeds `electricity_tariff` in de database.

dennissiemensma, 2022-10-20:
@vincedh ben je nog wat wijzer geworden?

vincedh, 2022-10-21:
Hoi Dennis,
Nee, helaas niet, ik heb ff in de database zelf gekeken en ook daar staat de waarde keurig in. Ook links bovenin het homescherm van DSMR reader zelf wordt het huidige tarief goed weergegeven. Alleen niet via MQTT dus... Ik zou niet weten wat ik verder nog kan doen en zo'n grote ramp is het ook weer niet. Dus mocht jij verder ook geen idee hebben laten we het hier bij. Toch bedankt voor je reactie.

dennissiemensma, 2022-10-21:
Je kunt nog de logging bekijken https://github.com/dsmrreader/dsmr-reader/issues/1728#issuecomment-1277979777

vincedh, 2022-10-24:
Hoi Dennis,
Ik ben een beetje bang dat het 'tarief bericht' alleen om 7u en 23u verstuurd wordt, daarna wordt het log vol gepropt met andere berichten, dus lastig om de juiste eruit te halen, heb jij een tip misschien?

dennissiemensma, 2022-10-24:
In principe kun je in de logfiles zoeken naar `electricity_tariff`. Weet je trouwens zeker dat je ook de MQTT van de meterstatistieken hebt ingeschakeld?

- Zie `/admin/dsmr_mqtt/splittopicmeterstatisticsmqttsettings/`

Het is namelijk geen onderdeel van de telegram update uit DSMR-reader

vincedh, 2022-10-28:
Ik heb het gevonden! 'electricity_tariff = dsmr/meter-stats/electricity_tariff' stond 2x in de lijst, waardoor deze niet verzonden werd. Aangepast en alles wordt direct keurig doorgegeven. Dank voor je tijd en ondersteuning.

closed 2022-10-28 by vincedh

dennissiemensma, 2022-10-29:
Top, fijn om te horen!