dsmrreader/dsmr-reader #1731: The Netherlands = Energe plafond💡

YLKersloot, 2022-10-22:
### Feature

In Nederland geldt straks een prijsplafond. Is hier al een feature request voor?

_Vanaf 1 januari 2023 geldt een prijsplafond op energie voor alle huishoudens en andere kleinverbruikers. Voor gas wordt het maximale tarief € 1,45 per kuub tot een verbruik van 1.200 kuub. Voor elektriciteit wordt het maximale tarief verlaagd tot € 0,40 per KWh en het maximale verbruik verhoogd tot 2.900 kWh._

Bron: https://www.rijksoverheid.nl/actueel/nieuws/2022/10/04/vanaf-1-januari-lagere-energierekening-door-verruimd-prijsplafond

dennissiemensma, 2022-10-22:
#1708

YLKersloot, 2022-10-22:
Duidelijk (uiteraard jammer). Is er misschien wel een mogelijkheid om een waarschuwingsmodule te maken? Dus bij meer verbruik van x Kuub / kWh per maand/jaar dan een waarschuwing tonen? Blijf je weg bij alle gekke prijsconstructies, maar help je wel bij bewustwording (en voor mij praktisch dat ik vanaf die dag dan een hoger tarief moet instellen :) :) )

dennissiemensma, 2022-10-22:
Zie ook https://github.com/xirixiz/dsmr-reader-docker/issues/296#issuecomment-1255463332

closed 2022-10-24 by dennissiemensma