dsmrreader/dsmr-reader #1739: MQTT errors met dsmr reader in hass🙋

maceddy, 2022-11-03:
### Description

Ik heb de DSMR reader geinstalleerd binnen HASS. Dat werkt uiteindelijk prima. Echter wil ik de data ook op het energydashboard hebben. Hier gaan dingen mis. Ik heb de MQTT aangezet binnen de DSMR reader. Hij verstuurt de data, maar in de DSMR logs vind ik de volgende entries:

2022-11-03 09:13:38,941 WARNING  messages     queue_message                    24 | MQTT: Rejecting message for topic due to maximum queue size (5000): dsmr/reading/electricity_currently_returned
2022-11-03 09:13:38,952 WARNING  messages     queue_message                    24 | MQTT: Rejecting message for topic due to maximum queue size (5000): dsmr/reading/phase_currently_delivered_l1
2022-11-03 09:13:38,963 WARNING  messages     queue_message                    24 | MQTT: Rejecting message for topic due to maximum queue size (5000): dsmr/reading/phase_currently_delivered_l2
2022-11-03 09:13:38,975 WARNING  messages     queue_message                    24 | MQTT: Rejecting message for topic due to maximum queue size (5000): dsmr/reading/phase_currently_delivered_l3
2022-11-03 09:13:38,986 WARNING  messages     queue_message                    24 | MQTT: Rejecting message for topic due to maximum queue size (5000): dsmr/reading/extra_device_timestamp

Hoe kan ik dit filteren of beperken? Ik heb de instructies, handleidingen etc uitgebreid doorgelopen, uren gegoogled maar ik kom er maar niet uit... 

De bedoeling is naast de DSMR reader, deze addin te gebruiken. Ik zie 78 entities maar zonder data.
https://www.home-assistant.io/integrations/dsmr_reader (als ik de gegevens zonder deze addin in het Energy dashboard kan krijgen is dat ook prima).

### DSMR-reader version

1.8.0 (5.8)

### DSMR-reader platform

Other

dennissiemensma, 2022-11-03:
Weet je zeker dat alle processen in DSMR-reader draaien en dat de broker berichten ontvangt? Want er wachten teveel berichten op verzending lijkt het.

rodeho, 2022-11-10:
Je kunt even MQTT explorer installeren en kijken of je berichten binnen krijgt

[image]

dennissiemensma, 2022-11-10:
Je kunt het beste DEBUG level logging tijdelijk aanzetten en in de logs van het backend proces kijken wat er gebeurt qua MQTT:
- https://github.com/dsmrreader/dsmr-reader/issues/1714#issuecomment-1261813560

Nu zie je alleen de warnings. Ik ben benieuwd of die uberhaupt constant bezig is met een achterstand wegwerken of iets.

closed 2022-11-21 by dennissiemensma