dsmrreader/dsmr-reader #1742: 🙋salderen over 3 fases

rodeho, 2022-11-10:
### Description

Ik heb hier ooit eerder naar gekeken, omdat ik van mening was/ben dat vroeger het verbruik naar 0 ging als je zonne-energie opwekte.

Wat je nu ziet is dat je nog steeds verbruik hebt, hoe kan ik nu valideren dat mijn meter saldeert over de fases? Stel dat ik 1kwh zonnenergie opwek, en 3x 250kwh afneem op 3 fases. Dan zou in mijn beleving het verbruik 0 moeten zijn (rood) en er 250kwh (groen) teruggeleverd moeten worden aan het net.

Wie weet hoe je dit goed kunt valideren? Hieronder twee screenshots één uit DSMR en één uit homeassistent waar de ververs snelheid 30 seconden is voor de SolarEdge modbus-tcp.

[image]

[image]

Als ik hier de som maak:

458 zonneenergie
255 verbruik in huis (klopt dat is in de nacht ook ivm servers etc)
405 teruglevering daaruit trek ik de conclusie dat hij niet saldeert over de fases/intern in de meter?

### DSMR-reader version

latest

### DSMR-reader platform

Native (e.g. manual installation)

rodeho, 2022-11-10:
Heb ook Stedin even gecontacteerd, voor nu waar salderen nog 100% is kost het geen geld. Als salderen gaat afbouwen dan is het intern salderen over de meter fases ineens een bron van inkomsten.

dennissiemensma, 2022-11-10:
Je kunt het beste een telegram uit je meter pakken op het tijdstip. Het zou overeen moeten komen met DSMR-reader, maar dichter bij de bron kan dan bijna niet.
In DSMR-reader kun je het laatste telegram telkens zien in de URL: `/admin/dsmr_datalogger/meterstatistics/`

----

Ik heb zelf geen zonnepanelen, ben geen elektricien en ken de interne hardware van meters niet. Echter krijg ik wel mee dat je niet de enige bent. Vaak ligt dit aan 1 fase teruglevering vs 3 fase verbruik. Ik weet niet of dat ook jouw situatie is.

De meter saldeert volgens mij niet tussen fases en wellicht kan dat technisch ook niet, tenzij je teruglevert over 3 fases (ik ben dus geen elektricien). Als dat het geval is dan is het vooral zaak om zoveel mogelijk stroom zelf te verbruiken om in de toekomst je teruglevering aan het net te drukken. Alleen weet ik dus niet hoeveel je dat gaat besparen en of het uberhaupt de moeite waard is om ermee bezig te zijn.

In België gaan ze overigens al zelfs zo ver dat de netbeheerder extra kosten rekent voor mensen die hoog piekverbruik hebben. Gebruikers met zonnepanelen hebben daarmee de taak om die piek zoveel mogelijk te drukken, waardoor het net ontlast wordt en ze dus ook minder hoeven te betalen aan de netbeheerder.