dsmrreader/dsmr-reader #1745: 🚨MQTT neemt de huidige contract prijzen niet over

rodeho, 2022-11-15:
### Description

Het valt mij al enige tijd op dat de (nieuwe) [DSMR integratie ](https://www.home-assistant.io/integrations/dsmr_reader/)met homeassistant bij mij niet goed omgaat met de prijzen van energie.

Ik heb even wat data uitgelezen, deel je mijn mening dat in MQTT de niet goed doorkomen? Als je kijkt naar de output van MQTT explorer dan is er inderdaad op de MQTT bus een energie bedrag van +/- 0,21 cent verstuurd. Alleen dit is niet wat ik ingevoerd heb bij de contracten. 

[image]

[image]

Mijn mappings (default)

[mapping]
```
# DATA = TOPIC PATH
electricity1 = dsmr/day-totals/electricity1
electricity2 = dsmr/day-totals/electricity2
electricity1_returned = dsmr/day-totals/electricity1_returned
electricity2_returned = dsmr/day-totals/electricity2_returned
electricity_merged = dsmr/day-totals/electricity_merged
electricity_returned_merged = dsmr/day-totals/electricity_returned_merged
electricity1_cost = dsmr/day-totals/electricity1_cost
electricity2_cost = dsmr/day-totals/electricity2_cost
electricity_cost_merged = dsmr/day-totals/electricity_cost_merged

# Gas (if any)
gas = dsmr/day-totals/gas
gas_cost = dsmr/day-totals/gas_cost
total_cost = dsmr/day-totals/total_cost

# Your energy supplier prices (if set)
energy_supplier_price_electricity_delivered_1 = dsmr/day-totals/energy_supplier_price_electricity_delivered_1
energy_supplier_price_electricity_delivered_2 = dsmr/day-totals/energy_supplier_price_electricity_delivered_2
energy_supplier_price_electricity_returned_1 = dsmr/day-totals/energy_supplier_price_electricity_returned_1
energy_supplier_price_electricity_returned_2 = dsmr/day-totals/energy_supplier_price_electricity_returned_2
energy_supplier_price_gas = dsmr/day-totals/energy_supplier_price_gas
```

Het enige wat ze aangeven dat je moet doen:

Enable the following data sources in the DSMR Reader administration pages with the default mappings:
Day consumption: Split topic
Gas consumption: Split topic
Meter Statistics: Split topic
Telegram: Split topic

### DSMR-reader version

latest

### DSMR-reader platform

Native (e.g. manual installation)

### Debug info dump

```shell
DSMR-READER
    App / Python / Database                                                       v5.8 / v3.10.6 / postgresql
    BE sleep / DL sleep / Retention / Override                                     1.0s / 0.5s / 168h / False
    Latest telegram version read / Parser settings                                                 "42" / "4"

DATA
    Telegrams total (est.)                                                                             116676
    Consumption records electricity / gas (est.)                                               119783 / 36164
```

dennissiemensma, 2022-11-15:
Kan het zijn dat je nog een oude mapping hebt? Ik zie bij mezelf dat alles nog op `day-totals` staat ipv `day-consumption`. Ik zie ook beide varianten terugkomen in je explorer, omdat ze altijd met `retain` aangemeld worden bij de broker.

Ik zie in de source van die integratie verwijzingen naar de laatste defaults in DSMR-reader: https://github.com/home-assistant/core/blob/dev/homeassistant/components/dsmr_reader/definitions.py#L223-L330

Je kunt proberen de default mapping in DSMR-reader over te nemen. Wellicht fixt dat iets. Het is ooit hernoemd, maar DSMR-reader hernoemt niet je config automatisch bij upgraden, omdat het dan iets stuk kan maken. Ironisch genoeg maakt het nu andersom ook iets stuk.

Het is trouwens al lange tijd geleden aangepast, meer dan 4 jaar terug: https://github.com/dsmrreader/dsmr-reader/commit/b6170f38#diff-8a948af85ff59a50709e718f7e58e04080d440c3246960d77c5f5b384f98c442

Maar wellicht loop je er nu pas tegenaan. Al weet ik niet 100% zeker of dit het oplost, maar ik vermoed van wel, gezien je prijzen in `day-totals` overeenkomen met je screenshot.

rodeho, 2022-11-15:
https://community.home-assistant.io/t/how-to-clear-embedded-mqtt-broker-database/98394/2

Er bijkt zoiets te bestaan als retained berichten welke altijd bewaard blijven. Deze heb ik nu via MQTT explorer in HA ontdaan van het retained label (zie link hierboven).

Ik start hem weer even op en dan doe ik een controle.

rodeho, 2022-11-15:
Dennis, ik heb nu alle default mappings van DSMR geladen, uit de tekst erboven geplakt in het veld eronder en op save gedrukt. Dat lijkt inderdaad beter te gaan tnx!

[image]

dennissiemensma, 2022-11-15:
Top! Gooi vooral nog even `day-totals` uit de retain van je broker, mocht die tussendoor weer teruggekomen zijn. Dan heb je geen last meer van legacy mocht je ooit opnieuw issues hebben met MQTT.

closed 2022-11-15 by dennissiemensma