dsmrreader/dsmr-reader #1748: Prijsplafond / variabele prijzen boven prijsplafond.

Koal4, 2022-11-23:
### Feature

Hi,

Misschien overbodig maar krijgen we een mogelijkheid om variabel contract aan te maken wat boven de prijsplafond uitkomt ivm nieuwe prijzen per 1 jan 2023?

Ik kreeg deze mail namelijk van Essent.
[image]

dennissiemensma, 2022-11-23:
Bedankt voor je verzoek. Helaas valt dit buiten de scope van het project. Zie ook alles qua prijzen in o.a.:   #1731 #1730 #1729 #1723 #1607 #1694 #1692

Koal4, 2022-11-23:
Okay jammer maar dat maakt de contract/prijsberekening wel een beetje nutteloos vrees ik. Begrijp ook wel dat dit project hier niks aan kan doen en dat we dit te danken hebben aan andere hoor ;) 

Maar veel van die topics gaan over variabele prijs per uur of teruggave. 

Wil wel even benadrukken dat dit de nieuwe opzet is die per 1 Jan ingaat voor een x aantal maanden. Ik ben geen 'codeklopper' maar het moet toch wel mogelijk zijn om ergens een startdatum veld te maken en wanneer je dus over de 2900kWh of 1200m3 gaat dat hij de naar de 'variabel tarief vanaf basisverbruik' tabel gaat kijken en zijn berekening gaat doen. Hiervoor moet dan wel een extra optie/veld in contract veld worden gemaakt. Maar nogmaals ik ben geen codeklopper maar dit lijkt mij simpeler te implementeren dan variable prijzen per uur teruggave/verbruik wat in de andere topics worden vermeld etc etc

closed 2022-11-23 by dennissiemensma

Koal4, 2022-11-23:
`De kosten in DSMR-reader zijn puur een indicatie om je een gevoel te geven wat een dag kost tov een andere dag (bijv met/zonder verwarming). Het is geen spiegeling of controle op daadwerkelijke prijsberekeningen.`

Ik gebruik de tool niet om daadwerkelijke controle op prijsberekening hoor.
Maar zoals je zelf al aangeeft zal deze 'functionaliteit' (om je het gevoel tegen geven) straks verdwijnen zodra men over de basisplafond gaat. En die prijzen wat gehanteerd worden zijn bijna het dubbele als niet meer van het prijsplafond. Persoonlijk lijkt mij juist belangrijk om dit inzichtelijk te krijgen.

Maar goed ik kan je niet dwingen uiteraard maar ik vind het wel vreemde beslissing. En zoals ik al aangaf hele tarief contract feature wordt dan nutteloos naar mijn mening. Het gros van de mensen zal hier namelijk overheen gaan.

Desalniettemin bedankt wat je zo ver gebouwd hebt!

dennissiemensma, 2022-11-23:
Wellicht kan ik het van een andere kant belichten. DSMR-reader is gemaakt om inzicht te krijgen in het live verbruik van je huishouden. Alles qua contracten of kosten is nooit een doel geweest van DSMR-reader, het is echt een bijzaak, iets om je een gevoel te geven. 
Elke gebruiker kan zelf bepalen of die feature nog nodig of nuttig is door de prijzen op nul te zetten. Wie weet is het over een jaar wel weer nuttig. De kern is immers inzicht in verbruik.

De wereld is in relatief korte tijd veranderd, waardoor contracten sowieso korter zijn (voor iedereen met variabel tarief). Daarnaast komen er ook meer leveranciers die bijvoorbeeld uurtarieven aanbieden. Dat heeft allemaal te maken met prijzen, maar niet met het verbruik.
Bijvoorbeeld salderen is ook een onderwerp wat ik bewust buiten DSMR-reader houd, omdat het buiten de slimme meter om gaat. Het is een administratieve handeling (die dus onlangs ook al willekeurig door een leverancier is aangepast en daarna weer teruggedraaid) en meer niet. En salderen komt eigenlijk qua administratie erg in de buurt van wat het prijsplafond is. Leveren voor X totdat een limiet bereikt is, daarna leveren voor Y. 
Plus dat het ook nog eens tijdelijk en eenmalig is. Vergeet niet dat alles wat ik toevoeg ook altijd moet blijven werken. Dat iets relatief makkelijk te maken is, zegt niets over de inspanning die daarna nodig is om het altijd correct te houden.

Zelfs als ik alle doelen van DSMR-reader zou laten varen en heroverwegen op dit moment, dan nog zie ik absoluut geen enkele rol voor DSMR-reader om daar iets mee te doen. Ik vind echt dat de energieleveranciers hier tegenwoordig genoeg in kunnen betekenen, helemaal met de huidige vraag van consumenten.

----

Nog pragmatische workaround:

Je kunt in DSMR-reader vrij makkelijk in het Archief/Dashboard zien wat je in het kalenderjaar verbruikt hebt. Als je ergens in 2023 ziet dat je over het platfond gaat, dupliceer/splits je je contract vanaf de datum dat je er overheen bent en stel je je nieuwe hoge prijs in. 
In het ergste geval moet je dat in 2023 dan 2x doen, als je ook gas afneemt. Dat lijkt me persoonlijk nog wel werkbaar. En het kan ook met terugwerkende kracht, als je er bijvoorbeeld pas na een maand ofzo achterkomt dat je er overheen bent.

extera-nl, 2022-12-22:
Zou het een optie kunnen zijn om met een script en de API van DSMR-Reader de prijzen dynamisch aan te passen?
Of moet er dan (bij wijze van) elk uur een nieuw contract worden aangemaakt?

Ik zie dat op dit moment alleen een GET mogelijk is voor de prijzen. 
Mocht de manier waarop DSMR-Reader nu de prijs gebruik in de berekening compatibel zijn met een prijs die we zelf dynamisch kunnen updaten, dan leg je de verantwoording voor het invoeren van de juiste dynamische prijzen buiten dit project, en op deze manier is deze functie niet erg intensief om te onderhouden lijkt me

dennissiemensma, 2022-12-22:
De prijzen in DSMR-reader zijn helaas alleen op dagbasis en worden ook slechts 1x per dag (na middernacht) gebruikt. Afhankelijk van je use-case, ben je daar dan vrij beperkt in.