dsmrreader/dsmr-reader #1750: 🙋geen telegram verwerking na slimme meterwissel

ph0real, 2022-11-23:
### Description

Hoi Dennis,

ik ben gister naar een sagemcom 211 overgestapt. Sindsdien lukt t me niet om DSMR de telegrammen goed te laten verwerken. Handmatig uitlezen geeft data hieronder. Heb jij misschien een oplosrichting  hiervoor?

pi@powerPI:/dev $ cu -l /dev/ttyUSB0 -s 115200 --parity=none -E q
Connected.

/Ene5\T211 ESMR 5.0

1-3:0.2.8(50)
0-0:1.0.0(220814123033S)
0-0:96.1.1(4530303632303030303039343937373232)
1-0:1.8.1(000010.039*kWh)
1-0:1.8.2(000000.000*kWh)
1-0:2.8.1(000004.232*kWh)
1-0:2.8.2(000000.000*kWh)
0-0:96.14.0(0001)
1-0:1.7.0(00.266*kW)
1-0:2.7.0(00.000*kW)
0-0:96.7.21(00009)
0-0:96.7.9(00006)
1-0:99.97.0(1)(0-0:96.7.19)(220813041433S)(0000001259*s)
1-0:32.32.0(00002)
1-0:52.32.0(00002)
1-0:72.32.0(00002)
1-0:32.36.0(00000)
1-0:52.36.0(00000)
1-0:72.36.0(00000)
0-0:96.13.0()
1-0:32.7.0(223.0*V)
1-0:52.7.0(225.0*V)
1-0:72.7.0(225.0*V)
1-0:31.7.0(000*A)
1-0:51.7.0(000*A)
1-0:71.7.0(000*A)
1-0:21.7.0(00.132*kW)
1-0:41.7.0(00.134*kW)
1-0:61.7.0(00.000*kW)
1-0:22.7.0(00.000*kW)
1-0:42.7.0(00.000*kW)
1-0:62.7.0(00.000*kW)
0-1:24.1.0(003)
0-1:96.1.0(0000000000000000000000000000000000)
0-1:24.2.1(632525252525S)(00000.000)
!7C68
/Ene5\T211 ESMR 5.0

1-3:0.2.8(50)
0-0:1.0.0(220814123034S)
0-0:96.1.1(4530303632303030303039343937373232)
1-0:1.8.1(000010.039*kWh)
1-0:1.8.2(000000.000*kWh)
1-0:2.8.1(000004.232*kWh)
1-0:2.8.2(000000.000*kWh)
0-0:96.14.0(0001)
1-0:1.7.0(00.267*kW)
1-0:2.7.0(00.000*kW)
0-0:96.7.21(00009)
0-0:96.7.9(00006)
1-0:99.97.0(1)(0-0:96.7.19)(220813041433S)(0000001259*s)
1-0:32.32.0(00002)
1-0:52.32.0(00002)
1-0:72.32.0(00002)
1-0:32.36.0(00000)
1-0:52.36.0(00000)
1-0:72.36.0(00000)
0-0:96.13.0()
1-0:32.7.0(223.0*V)
1-0:52.7.0(225.0*V)
1-0:72.7.0(225.0*V)
1-0:31.7.0(000*A)
1-0:51.7.0(000*A)
1-0:71.7.0(000*A)
1-0:21.7.0(00.132*kW)
1-0:41.7.0(00.134*kW)
1-0:61.7.0(00.000*kW)
1-0:22.7.0(00.000*kW)
1-0:42.7.0(00.000*kW)
1-0:62.7.0(00.000*kW)
0-1:24.1.0(003)
0-1:96.1.0(0000000000000000000000000000000000)
0-1:24.2.1(632525252525S)(00000.000)
!5D43
/Ene5\T211 ESMR 5.0

1-3:0.2.8(50)
0-0:1.0.0(220814123035S)
0-0:96.1.1(4530303632303030303039343937373232)
1-0:1.8.1(000010.040*kWh)
1-0:1.8.2(000000.000*kWh)
1-0:2.8.1(000004.232*kWh)
1-0:2.8.2(000000.000*kWh)
0-0:96.14.0(0001)
1-0:1.7.0(00.268*kW)
1-0:2.7.0(00.000*kW)
0-0:96.7.21(00009)
0-0:96.7.9(00006)
1-0:99.97.0(1)(0-0:96.7.19)(220813041433S)(0000001259*s)
1-0:32.32.0(00002)
1-0:52.32.0(00002)
1-0:72.32.0(00002)
1-0:32.36.0(00000)
1-0:52.36.0(00000)
1-0:72.36.0(0000cu: Got hangup signal

Disconnected.

 

### DSMR-reader version

5.8

### Debug info dump

```shell
DSMR-READER
    App / Python / Database                                                        v5.8 / v3.9.2 / postgresql
    BE sleep / DL sleep / Retention / Override                                     5.0s / 5.0s / 168h / False
    Latest telegram version read / Parser settings                                                 "50" / "4"

DATA
    Telegrams total (est.)                                                                             134707
    Consumption records electricity / gas (est.)                                                62788 / 54327

UNRESOLVED ISSUES
    Proces loopt achter op schema: Calculate quarter hour electricity peaks                      58 minuten geleden
    Geen recente metingen ontvangen                                                      1 dag, 8 uur geleden
    Dagstatistieken lopen achter                                                        1 dag, 19 uur geleden
```

dennissiemensma, 2022-11-23:
Bedankt voor je vraag. Zie je ook geen fouten in de logfiles? Specifiek die van de datalogger.

----

Een _vermoeden_ is dat het ligt aan een niet (goed) gekoppelde gasmeter. Heb je er uberhaupt een gasmeter aan?

Want als die er aan zou moeten zitten, lijkt die nog niet lekker gekoppeld:
```
0-1:96.1.0(0000000000000000000000000000000000)
0-1:24.2.1(632525252525S)(00000.000)
```
Geen serienummer, ongeldige datum, geen meterstand.

ph0real, 2022-11-23:
Traceback (most recent call last):
  File "/home/dsmr/dsmr-reader/dsmr_backend/mixins.py", line 92, in run_once
    self.run(data=self.data, **options)
  File "/home/dsmr/dsmr-reader/dsmr_datalogger/management/commands/dsmr_datalogger.py", line 29, in run
    telegram = next(self.telegram_generator)
  File "/home/dsmr/dsmr-reader/dsmr_datalogger/scripts/dsmr_datalogger_api_client.py", line 39, in read_telegram
    raise RuntimeError('Failed to connect: {}', error)
RuntimeError: ('Failed to connect: {}', SerialException(12, "could not open port /dev/ttyUSB0: [Errno 12] Cannot allocate memory: '/dev/ttyUSB0'"))

De slimme gasmeter is gekoppeld aan de 211 waarop ik met P1 ben aangesloten. Maar dit lijkt nu op een ander probleem… het benaderen van de usb poort…

ph0real, 2022-11-23:
De meest actuele uit de log: 

Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs>
2022-11-22 11:29:04,020 ERROR    mixins       run_once                        103 | dsmr_datalogger.manag>
  File "/home/dsmr/dsmr-reader/dsmr_backend/mixins.py", line 92, in run_once
    self.run(data=self.data, **options)
  File "/home/dsmr/dsmr-reader/dsmr_datalogger/management/commands/dsmr_datalogger.py", line 29, in run
    telegram = next(self.telegram_generator)
  File "/home/dsmr/dsmr-reader/dsmr_datalogger/scripts/dsmr_datalogger_api_client.py", line 47, in read_t>
    raise RuntimeError(
RuntimeError: It took too long to detect a telegram. Check connection params. Bytes currently in buffer: 0

Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs>
2022-11-22 11:29:33,257 ERROR    mixins       run_once                        103 | dsmr_datalogger.manag>
  File "/home/dsmr/dsmr-reader/dsmr_backend/mixins.py", line 92, in run_once
    self.run(data=self.data, **options)
  File "/home/dsmr/dsmr-reader/dsmr_datalogger/management/commands/dsmr_datalogger.py", line 29, in run
    telegram = next(self.telegram_generator)
  File "/home/dsmr/dsmr-reader/dsmr_datalogger/scripts/dsmr_datalogger_api_client.py", line 47, in read_t>
    raise RuntimeError(
RuntimeError: It took too long to detect a telegram. Check connection params. Bytes currently in buffer: 0

Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs>
2022-11-22 11:30:02,478 ERROR    mixins       run_once                        103 | dsmr_datalogger.manag>
  File "/home/dsmr/dsmr-reader/dsmr_backend/mixins.py", line 92, in run_once
    self.run(data=self.data, **options)
  File "/home/dsmr/dsmr-reader/dsmr_datalogger/management/commands/dsmr_datalogger.py", line 29, in run
    telegram = next(self.telegram_generator)
  File "/home/dsmr/dsmr-reader/dsmr_datalogger/scripts/dsmr_datalogger_api_client.py", line 47, in read_t>
    raise RuntimeError(
RuntimeError: It took too long to detect a telegram. Check connection params. Bytes currently in buffer: 0

Hierna nog wat meldingen van vandaag maar die waren vooral getriggerd door mijn reboots.
De gasmeter is goed gekoppeld aan de 211. Ik las wel op internet dat het bij sommige apparaten even duurt voordat de koppeling echt actief wordt. Misschien is dat iets?

dennissiemensma, 2022-11-23:
Die meldingen lijken nog van eind gisterochtend. Wellicht dat het idd even duurt.

Je kunt nog **tijdelijk** proberen om in `/admin/dsmr_backend/backendsettings/` gas keihard **uit** te zetten. Ik weet niet uit m'n hoofd of dat helpt, zodat elektra wel gemeten wordt, maar je kunt het altijd proberen. Nu meet die namelijk helemaal niets en daar heb je helemaal niets aan.

ph0real, 2022-11-24:
Doel je op de setting gasweergave aan/uit?
Die geeft geen verschil helaas. Overigens lijkt dit geval een beetje op bug 919. Die persoon had ook een zelfde entry elke keer in de gas telegram. Kennelijk gaat t daarop verkeerd.

dennissiemensma, 2022-11-25:
Je kunt het beste debug logging aanzetten en dan in de datalogger logfiles kijken wat er gebeurt. Dan zou je alles moeten zien.

ph0real, 2022-11-25:
Debug zag er niet gek uit muv de gaswaarden. Ik ben achter iets anders gekomen: de optie 'overschrijf tijdstippen van telegrammen' is een workaround om in ieder geval de electra weer te laten werken in het dashboard. Dat werkt nu; misschien kan ik enexis eens bellen om te vragen hoe het zit met de koppeling tussen gas en electrische meter...

2022-11-25 21:04:20,729 DEBUG    mixins       run_loop                         79 | dsmr_datalogger.management.commands.dsmr_datalogger: Sleeping 5.0s
2022-11-25 21:04:25,740 INFO     dsmr_datalogger_api_client read_telegram                    31 | [2022-11-25 21:04:25.740543] Opening connection "/dev/ttyUSB0" using options: {'baudrate': 115200, 'bytesize': 8, 'parity': 'N', 'stopbits': 1, 'xonxoff': 1, 'rtscts': 0}
2022-11-25 21:04:26,082 DEBUG    dsmr_datalogger_api_client read_telegram                    58 | [2022-11-25 21:04:26.082373] Read 1 Byte(s)
2022-11-25 21:04:26,416 DEBUG    dsmr_datalogger_api_client read_telegram                    58 | [2022-11-25 21:04:26.416789] Read 0 Byte(s)
2022-11-25 21:04:26,751 DEBUG    dsmr_datalogger_api_client read_telegram                    58 | [2022-11-25 21:04:26.751009] Read 883 Byte(s)
2022-11-25 21:04:26,760 DEBUG    datalogger   telegram_to_reading              81 | Received telegram:
/Ene5\T211 ESMR 5.0

1-3:0.2.8(50)
0-0:1.0.0(220816135705S)
0-0:96.1.1(4530303632303030303039343937373232)
1-0:1.8.1(000019.804*kWh)
1-0:1.8.2(000000.000*kWh)
1-0:2.8.1(000018.630*kWh)
1-0:2.8.2(000000.000*kWh)
0-0:96.14.0(0001)
1-0:1.7.0(00.461*kW)
1-0:2.7.0(00.000*kW)
0-0:96.7.21(00009)
0-0:96.7.9(00006)
1-0:99.97.0(1)(0-0:96.7.19)(220813041433S)(0000001259*s)
1-0:32.32.0(00002)
1-0:52.32.0(00002)
1-0:72.32.0(00002)
1-0:32.36.0(00000)
1-0:52.36.0(00000)
1-0:72.36.0(00000)
0-0:96.13.0()
1-0:32.7.0(225.0*V)
1-0:52.7.0(228.0*V)
1-0:72.7.0(229.0*V)
1-0:31.7.0(001*A)
1-0:51.7.0(001*A)
1-0:71.7.0(000*A)
1-0:21.7.0(00.183*kW)
1-0:41.7.0(00.277*kW)
1-0:61.7.0(00.000*kW)
1-0:22.7.0(00.000*kW)
1-0:42.7.0(00.000*kW)
1-0:62.7.0(00.000*kW)
0-1:24.1.0(003)
0-1:96.1.0(0000000000000000000000000000000000)
0-1:24.2.1(632525252525S)(00000.000)
!2E61

closed 2022-11-27 by ph0real