dsmrreader/dsmr-reader #1751: Laat daadwerkelijk verbruik zien, ondanks dat er teruglevering plaatsvind💡

D1StrX, 2022-11-27:
### Feature

Dsmr laat nu 0 (consuming) verbruik zien wanneer de zonnepanelen meer opwekken dan verbruikt wordt op het _dashboard_ en in de _live grafieken_. Wat ik denk te zien is het volgende; DSMR trekt van _consuming_ het _returning_ af en laat dat totaal zien.
De grafiek hieronder uit de _live grafieken_ laat eigenlijk een verwrongen beeld zien, want het daadwerkelijke verbruik is 330wH.
[image]
Ik kan de gedachtegang snappen achter wat nog daadwerkelijk geleverd word. 

Kan op het dashboard of in de live-grafiek (zie voorbeeld hieronder) nog steeds het totaal verbruik weergeven worden?
[image]
[image]

dennissiemensma, 2022-11-27:
Bedankt voor je melding. Het is mij niet helemaal duidelijk wat je bedoelt. Bedoel je dat de grafiek verkeerde waarden laat zien of de header bovenin? In je onderste screenshot vallen de waardes helaas weg.

dennissiemensma, 2022-11-27:
Je kunt op `/admin/dsmr_datalogger/meterstatistics/` het laatste telegram zien. Wellicht helpt je dat ook bij achterhalen wat wel en niet klopt.

dennissiemensma, 2022-11-28:
Voor de volgende versie heb ik overigens een aparte pagina gemaakt die het iets makkelijker maakt om bij je (laatste) telegram te komen. Dat maakt debuggen vermoedelijk wat eenvoudiger.

D1StrX, 2022-11-28:
> Bedankt voor je melding. Het is mij niet helemaal duidelijk wat je bedoelt. Bedoel je dat de grafiek verkeerde waarden laat zien of de header bovenin? In je onderste screenshot vallen de waardes helaas weg.
> 
> 

Zowel de grafiek als het dashboard laat niet verkeerde waardes zien, maar laat niet meer het actuele verbruik zien. Omdat het terugleveren van het verbruik is afgetrokken. 

Voorbeeld: 
Zonder zonnepanelen is bijvoorbeeld actueel verbruik 0,5kwH , maar met zonnepanelen is het actuele verbruik 0.2kwH of dus 0 als de teruglevering 0.3kwH of meer is dan het verbruik. 
Je mist dan in een oogopslag te kunnen zien wat je huishouden op dat moment gebruikt. (Het verwrongen beeld)

Het zou mooi zijn als dat los weergeven kan worden. Aldus op het dashboard als extra column, en/of losse diagram in de Live grafiek. (Zoals bijgetekend in de screenshots). Dan blijven nog steeds de kosten inzichtelijk/juist. Maar is ook te zien hoeveel je verbruikt, ondanks teruglevering.

Hopelijk geeft dit meer duidelijkheid 😄

dennissiemensma, 2022-11-28:
Bedankt voor je aanvulling. Deze gegevens zijn helaas niet beschikbaar in je slimme meter of DSMR-reader. 
Alle opgewerkte elektra die binnenshuis verbruikt wordt, bereikt nooit de slimme meter.

Voor een fictief voorbeeld van teruglevering op fase 1, waardoor het verbruik op die zelfde fase `0` wordt:
```
...
1-0:1.7.0(01.838*kW)   # Verbruik van net
1-0:2.7.0(00.867*kW)   # Teruglevering aan net
...
1-0:21.7.0(00.000*kW)  # Verbruik van net fase 1
1-0:41.7.0(00.885*kW)  # Verbruik van net fase 2
1-0:61.7.0(00.953*kW)  # Verbruik van net fase 3
1-0:22.7.0(00.867*kW)  # Teruglevering aan net fase 1
1-0:42.7.0(00.000*kW)  # Teruglevering aan net fase 2
1-0:62.7.0(00.000*kW)  # Teruglevering aan net fase 3
...
```

Vertaalt dat als het goed is naar:
----

[image]

----

[image]

----

Het kan best zijn dat in bovenstaand voorbeeld de zonnepanelen 2000 Watt opleveren, maar dat opgenomen wordt in het huishouden. Alleen de restende `867` bereikt dan de meter en dus DSMR-reader.

Je kunt dit checken door je eigen telegram te bekijken. Mocht daar toch iets niet kloppen dan hoor ik het uiteraard graag. Bovenstaande voorbeeld is gegenereerde nepdata.

closed 2022-11-30 by dennissiemensma