dsmrreader/dsmr-reader #1752: 🙋Verbruik per fase zien

ChristiaanW123, 2022-11-29:
### Description

Hoi,

Ik kan het huidige vermogen, sterkte en spanning zien per fase in de live grafieken, maar lijk nergens het (historisch) verbruik per fase te kunnen zien. Hoe doe ik dat?

### DSMR-reader version

5

dennissiemensma, 2022-11-29:
Bedankt voor je vraag. Dit is helaas beperkt mogelijk. Deze gegevens worden als "vluchtig" behandeld omdat de dataset anders te groot wordt en het vooral een momentopname is, in tegenstelling tot het verbruik (meterstanden).

Echter kun je met onderstaande instellingen wel relatief recente gegevens terugzien in de grafieken, maar er is geen zoekfunctie. Dit kan helpen, ten koste van datagrootte/performance:

- `/admin/dsmr_consumption/consumptionsettings/` -> **Electricity grouping type:** -> per meting
- `/admin/dsmr_frontend/frontendsettings/` -> **Live graphs hours range:** -> zo hoog mogelijk (ik dacht 168 uur = 1 week max)

closed 2022-11-29 by dennissiemensma

ChristiaanW123, 2022-11-29:
Dank voor de informatie, is er een manier om dit met MQTT nog weg te schrijven naar eventueel Grafana? Het gaat mij er met name om dat ik het verbruik van een fase per maand wil zien, zodat we weten wat dat deel van het huis per maand gebruikt.

dennissiemensma, 2022-11-29:
Ik heb je initiele vraag niet helemaal goed gelezen, ik dacht dat het over de fase-gegevens ging, die DSMR-reader voor beperkte tijd kan laten zien. 

Er is **geen** (kWh) verbruik/meterstanden beschikbaar **per fase** bij slimme meters, alleen de eerder genoemde gegevens.

dennissiemensma, 2022-11-29:
Zie overigens deze PDF voor een overzicht wat de (meeste) meters ontsluiten: https://github.com/dsmrreader/dsmr-reader/blob/v5/docs/_static/downloads/netbeheer-nederland-dsmr-specs-v5.0.2.pdf

Binnen de PDF pagina's 20 tot en met 23 hebben tabellen met de gegevens.