dsmrreader/dsmr-reader #1753: Python 3.12+ support

dennissiemensma, 2022-11-30:
### Feature

Future Python version(s), but has some incompatible changes. Some are 3rd party, some are internal.

```
../opt/venv/lib/python3.11/site-packages/pytest_cov/plugin.py:256
  /opt/venv/lib/python3.11/site-packages/pytest_cov/plugin.py:256: PytestDeprecationWarning: The hookimpl CovPlugin.pytest_configure_node uses old-style configuration options (marks or attributes).
  Please use the pytest.hookimpl(optionalhook=True) decorator instead
   to configure the hooks.
   See https://docs.pytest.org/en/latest/deprecations.html#configuring-hook-specs-impls-using-markers
    def pytest_configure_node(self, node):
```

```
../opt/venv/lib/python3.11/site-packages/django/http/request.py:1
  /opt/venv/lib/python3.11/site-packages/django/http/request.py:1: DeprecationWarning: 'cgi' is deprecated and slated for removal in Python 3.13
    import cgi
```

```
../opt/venv/lib/python3.11/site-packages/django/utils/encoding.py:266
  /opt/venv/lib/python3.11/site-packages/django/utils/encoding.py:266: DeprecationWarning: Use setlocale(), getencoding() and getlocale() instead
    encoding = locale.getdefaultlocale()[1] or 'ascii'
```

```
dsmr_backend/tests/services/test_backend.py: 4 warnings
dsmr_frontend/tests/webinterface/test_about.py: 22 warnings
dsmr_frontend/tests/webinterface/test_generic.py: 44 warnings
  /app/dsmr_backend/services/backend.py:124: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.
    if StrictVersion(comparable_release_tag) <= StrictVersion(local_version):

-- Docs: https://docs.pytest.org/en/stable/how-to/capture-warnings.html
```

dennissiemensma, 2023-07-21:
Nu nog in pre-release, pas in Q4 ofzo stable.

GieltjE, 2024-07-01:
De eerste OS'en hebben deze ondertussen als stable gemarkeerd.

dennissiemensma, 2024-07-01:
@GieltjE dank voor de update!

Het plan voor DSMR-reader 6.x is een flinke bump. Sterker nog, alleen Python 3.12 wordt dan nog ondersteund icm containers:
- #1594
- #1196
- #1763
- #1753

Dit om het upgraden flink te versimpelen.

dennissiemensma, 2024-07-01:
Inmiddels zit de v6-x werk-branch ook al op 3.12:
- https://github.com/dsmrreader/dsmr-reader/tree/work-in-progress/v6

Dus ik sluit deze.

closed 2024-07-01 by dennissiemensma