dsmrreader/dsmr-reader #1754: Krijg DMSR Reader niet aan de gang (user problem...)

Mischa1975, 2022-11-30:
### Description

Hallo,

ik draai HA in een Virtual Machine op een Intel NUC. Ik heb de DMSR Reader geinstalleerd, daarnaast ook PostgresDB en Mosquitto MQTT.

- PostgresDB: ik heb daar DSMRreader toegevoegd als een "database" (bij 'opties')
- Ik denk dat het (oa) misgaat bij stap 9 van de "DMSR configuration". Specifiek: 

Go to Datalogger -> Dataloggerconfiguratie and specify the correct serial USB port or configure a remote network socket input method (using ser2net).

Ik heb hierbij het IP adres van P1 meter geconfigureerd en de standaard 23 als port aangehouden. Weet niet of dit juist is?

Als ik dan terugga naar het hoofdscherm van DSMR, dan krijg ik deze foutmelding:  Geen gegevens gevonden. Zorg ervoor dat het dsmr_datalogger-proces draait of dat je gegevens aanlevert via de API. Daarnaast hoort het dsmr_backend-proces eveneens te draaien.

Ik heb duidelijk iets niet goed geconfigureerd - maar kom er niet meer uit wat niet? Wat kan ik nog meer doen/proberen?

### DSMR-reader version

5.8

### Debug info dump

```shell
DSMR-READER
    App / Python / Database                                                       v5.8 / v3.10.7 / postgresql
    BE sleep / DL sleep / Retention / Override                                     1.0s / 5.0s / 672h / False
    Latest telegram version read / Parser settings                                               "None" / "4"

DATA
    Telegrams total (est.)                                                                                 -1
    Consumption records electricity / gas (est.)                                                      -1 / -1

UNRESOLVED ISSUES
    Geen dagstatistieken gevonden                                                                          nu
    Nog nooit een meting ontvangen                                                                         nu
    Proces loopt achter op schema: Calculate quarter hour electricity peaks                      34 minuten geleden
```

dennissiemensma, 2022-11-30:
Bedankt voor je melding. Je kunt het beste beginnen bij de logfiles: https://dsmr-reader.readthedocs.io/nl/v5/how-to/troubleshooting/logfiles.html

Specifiek die van de datalogger. Mogelijk staat daar al een fout in.

Mischa1975, 2022-11-30:
dankje voor je snelle reactie. Ik ben inmiddels al zo'n 15-20 min aant kijken en googlen hoe/waar ik die commands die staan om het DEBUG te enable kan invoeren. Ik heb ook de bestanden proberen te vinden via FILE EDITOR, ook daar zie ik nergens een verwijzing naar DSMR

dennissiemensma, 2022-11-30:
Via welke handleiding heb je DSMR-reader geinstalleerd? Er zijn meerdere mogelijkheden en niet elke variant heeft de logfiles op dezelfde plek:

- https://github.com/sanderdw/hassio-addons
- https://github.com/xirixiz/dsmr-reader-docker
- https://dsmr-reader.readthedocs.io/nl/v5/tutorial/installation/step-by-step.html

_DSMR-reader vereist overigens wel wat van je, omdat je je installatie ook altijd zelf beheert. Weet dat vooraf ;-)_

Mischa1975, 2022-11-30:
via de sanderdw (add-on install). Ja ik realiseer me dat dit wellicht (te) diep gaat in vergelijk met mijn ervaring met HA :-)

dennissiemensma, 2022-11-30:
Je kunt denk ik het beste even een issue aanmaken daar, want ik weet niet hoe het qua logs ingeregeld is via dat project.

Mischa1975, 2022-11-30:
ok - dankje!

closed 2022-12-01 by dennissiemensma