dsmrreader/dsmr-reader #1757: 🙋database mist data

rodeho, 2022-12-01:
### Description

Ha

Ik was zojuist wat maanden aan het vergelijken en toen zag ik dat de volgende maanden last hebben van ontbrekende data. 

Na wat onderzoek blijkt dat homeassistant wel de data heeft ontvangen vanuit DSMR reader. De telegrammen zijn ingelezen > DSMR reader > MQTT > homeassistant. 

Is het mogelijk om met wat commando’s te controleren of de database te herstellen is?

Handmatig de dagtotalen overtikken van homeassistant naar DSMR reader kan dat ook?

### DSMR-reader version

5.9

### DSMR-reader platform

Native (e.g. manual installation)

### Debug info dump

```shell
DSMR-READER
    App / Python / Database                                                       v5.9 / v3.10.6 / postgresql
    BE sleep / DL sleep / Retention / Override                                     1.0s / 0.5s / 672h / False
    Latest telegram version read / Parser settings                                                 "42" / "4"

DATA
    Telegrams total (est.)                                                                             119550
    Consumption records electricity / gas (est.)                                               111408 / 36164
```

rodeho, 2022-12-01:
```
ik@dsmr:/home/dsmr/dsmr-reader/backups/archive/2022/01$ ls -la
total 288
drwxr-xr-x  2 dsmr dsmr  4096 Jan 31  2022 .
drwxr-xr-x 13 dsmr dsmr  4096 Nov  7 15:45 ..
-rw-r--r--  1 dsmr dsmr 55635 Jan  3  2022 dsmrreader-postgresql-partial-backup-2022-01-03.sql.gz
-rw-r--r--  1 dsmr dsmr 55932 Jan 10  2022 dsmrreader-postgresql-partial-backup-2022-01-10.sql.gz
-rw-r--r--  1 dsmr dsmr 56219 Jan 17  2022 dsmrreader-postgresql-partial-backup-2022-01-17.sql.gz
-rw-r--r--  1 dsmr dsmr 56485 Jan 24  2022 dsmrreader-postgresql-partial-backup-2022-01-24.sql.gz
-rw-r--r--  1 dsmr dsmr 56757 Jan 31  2022 dsmrreader-postgresql-partial-backup-2022-01-31.sql.gz
ik@dsmr:/home/dsmr/dsmr-reader/backups/archive/2022/01$ cd ..
ik@dsmr:/home/dsmr/dsmr-reader/backups/archive/2022$ cd 02
ik@dsmr:/home/dsmr/dsmr-reader/backups/archive/2022/02$ ls -la
total 240
drwxr-xr-x  2 dsmr dsmr  4096 Feb 28  2022 .
drwxr-xr-x 13 dsmr dsmr  4096 Nov  7 15:45 ..
-rw-r--r--  1 dsmr dsmr 57039 Feb  7  2022 dsmrreader-postgresql-partial-backup-2022-02-07.sql.gz
-rw-r--r--  1 dsmr dsmr 57336 Feb 14  2022 dsmrreader-postgresql-partial-backup-2022-02-14.sql.gz
-rw-r--r--  1 dsmr dsmr 57626 Feb 21  2022 dsmrreader-postgresql-partial-backup-2022-02-21.sql.gz
-rw-r--r--  1 dsmr dsmr 57927 Feb 28  2022 dsmrreader-postgresql-partial-backup-2022-02-28.sql.gz
rodeho@dsmr:/home/dsmr/dsmr-reader/backups/archive/2022/02$ cd ..
rodeho@dsmr:/home/dsmr/dsmr-reader/backups/archive/2022$ cd 02
rodeho@dsmr:/home/dsmr/dsmr-reader/backups/archive/2022/02$ cd 03
-bash: cd: 03: No such file or directory
ik@dsmr:/home/dsmr/dsmr-reader/backups/archive/2022/02$ cd ..
ik@dsmr:/home/dsmr/dsmr-reader/backups/archive/2022$ cd 03
ik@dsmr:/home/dsmr/dsmr-reader/backups/archive/2022/03$ ls -la
total 240
drwxr-xr-x  2 dsmr dsmr  4096 Mar 28  2022 .
drwxr-xr-x 13 dsmr dsmr  4096 Nov  7 15:45 ..
-rw-r--r--  1 dsmr dsmr 58284 Mar  7  2022 dsmrreader-postgresql-partial-backup-2022-03-07.sql.gz
-rw-r--r--  1 dsmr dsmr 58634 Mar 14  2022 dsmrreader-postgresql-partial-backup-2022-03-14.sql.gz
-rw-r--r--  1 dsmr dsmr 56760 Mar 21  2022 dsmrreader-postgresql-partial-backup-2022-03-21.sql.gz
-rw-r--r--  1 dsmr dsmr 57066 Mar 28  2022 dsmrreader-postgresql-partial-backup-2022-03-28.sql.gz
```

rodeho, 2022-12-01:
Okey ik ben nu hier en wacht even op @dennissiemensma 

https://dsmr-reader.readthedocs.io/nl/v4/how-to/database/postgresql-restore-backup.html#importing-a-partial-backup

[image]

rodeho, 2022-12-01:
Wat ik heb gedaan:

Alle files geopend in een SQL viewer en er één bewerkt en daar alle lijnen in gezet die nog bestonden. In plaats van 6 weken mis ik netto maar 1 week. Maar het gaat mis op de key ID's

```
ik@dsmr:/home/dsmr/dsmr-reader/backups$ sudo cat /home/dsmr/dsmr-reader/backups/dsmrreader-postgresql-partial-backup-ik.sql | sudo -u postgres psql dsmrreader
SET
SET
SET
SET
SET
 set_config
------------

(1 row)

SET
SET
SET
SET
ERROR:  duplicate key value violates unique constraint "dsmr_stats_daystatistics_pkey"
DETAIL:  Key (id)=(1292) already exists.
CONTEXT:  COPY dsmr_stats_daystatistics, line 1
 setval
--------
   1335
(1 row)
```

Gedeelte sql file inhoud:

```
1292    2022-01-29      0.00    9.566   0.000   0.411   0.000   0.00    0.00    2.723   0.00    8.5     11.2    6.3     0.00    14418.550       2047.446        10976.604       4894.502        4555.234
1293    2022-01-30      0.00    19.283  0.000   2.495   0.000   0.00    0.00    3.940   0.00    5.3     8.4     2.1     0.00    14428.116       2047.857        10976.604       4894.502        4557.957
1294    2022-01-31      0.00    8.314   6.214   0.000   0.000   0.00    0.00    3.961   0.00    5.1     6.4     2.8     0.00    14447.430       2050.352        10976.604       4894.502        4561.897
1295    2022-02-01      0.00    13.954  7.534   0.000   0.000   0.00    0.00    4.462   0.00    6.3     9.1     2.1     0.00    14455.745       2050.352        10982.818       4894.502        4565.858
1296    2022-02-02      0.00    42.454  5.883   0.000   3.998   0.00    0.00    2.566   0.00    8.0     10.6    6.2     0.00    14469.730       2050.352        10990.352       4894.502        4570.320
1297    2022-02-03      0.00    5.720   7.923   0.000   0.005   0.00    0.00    3.420   0.00    7.3     8.2     6.8     0.00    14512.215       2050.352        10996.235       4898.500        4572.887
1298    2022-02-04      0.00    2.249   9.391   0.000   0.182   0.00    0.00    2.816   0.00    6.4     9.0     3.3     0.00    14517.935       2050.352        11004.158       4898.505        4576.307
1299    2022-02-05      0.00    32.353  0.000   1.835   0.000   0.00    0.00    2.980   0.00    5.5     8.6     2.1     0.00    14520.185       2050.352        11013.549       4898.687        4579.123
1300    2022-02-06      0.00    34.874  0.000   0.144   0.000   0.00    0.00    3.844   0.00    7.1     8.6     4.0     0.00    14552.538       2052.187        11013.549       4898.687        4582.103
1301    2022-02-07      0.00    2.131   5.047   0.000   3.942   0.00    0.00    2.710   0.00    5.5     9.2     2.5     0.00    14587.413       2052.331        11013.549       4898.687        4585.947
1302    2022-02-08      0.00    2.139   28.194  0.000   1.059   0.00    0.00    4.555   0.00    9.2     11.7    6.5     0.00    14589.545       2052.331        11018.596       4902.629        4588.657
1303    2022-02-09      0.00    2.189   6.948   0.000   0.537   0.00    0.00    2.803   0.00    8.7     10.2    7.9     0.00    14591.685       2052.331        11046.790       4903.688        4593.212
1304    2022-02-10      0.00    12.807  5.741   0.000   1.938   0.00    0.00    2.915   0.00    6.7     8.7     3.4     0.00    14593.874       2052.331        11053.738       4904.225        4596.015
1305    2022-02-11      0.00    26.692  7.885   0.000   3.775   0.00    0.00    2.338   0.00    2.9     7.0     -5.5    0.00    14606.709       2052.331        11059.479       4906.163        4598.930
```

```
sudo su - postgres
psql dsmrreader

select id, day, electricity1, electricity2 from dsmr_stats_daystatistics order by day desc;
```

Dan is zichtbaar dat de ID's al bestaan en dat maakt het voor mij te complex. Kan ik met de hand niet-bestaande ID's toewijzen? Of kan dit ook met handige SQL commando's "inserted" worden?

```
 1299 | 2022-03-26 |        8.463 |        0.000
 1298 | 2022-03-25 |        2.896 |        3.372
 1297 | 2022-03-24 |        2.740 |        5.131
 1296 | 2022-03-23 |       29.748 |       10.406
 1295 | 2022-03-22 |       13.966 |        3.769
 1294 | 2022-03-21 |        2.485 |        4.275
 1293 | 2022-03-20 |        7.373 |        0.000
 1292 | 2022-03-19 |        0.160 |        0.000
 1291 | 2022-01-28 |       32.586 |        8.258
 1290 | 2022-01-27 |       30.708 |        8.080
 1289 | 2022-01-26 |        3.144 |        6.327
 1288 | 2022-01-25 |        3.265 |       11.371
 1287 | 2022-01-24 |        3.178 |       33.543
 1286 | 2022-01-23 |       11.170 |        0.000
 1285 | 2022-01-22 |       24.844 |        0.000
 1284 | 2022-01-21 |        3.318 |        6.530
 1283 | 2022-01-20 |        2.352 |       36.023
 1282 | 2022-01-19 |       19.481 |        9.795
```

rodeho, 2022-12-01:
Webgui:

Wat gemiddelde waarden van de betreffende maand gepakt en geprobeerd om in de database handmatig in te voeren via de web GUI

[image]

DSMR reader kiest dan ook een key die al in gebruik is

```
Server Error

Sorry, something unexpected happened.
Exception:

IntegrityError: duplicate key value violates unique constraint "dsmr_stats_daystatistics_pkey"
DETAIL:  Key (id)=(1337) already exists.
```

```
 1341 | 2022-05-07 |       11.689 |        0.000
 1340 | 2022-05-06 |        2.510 |        3.435
 1339 | 2022-05-05 |        2.564 |        3.011
 1338 | 2022-05-04 |        3.012 |        8.818
 1337 | 2022-05-03 |        3.514 |        3.597
 1336 | 2022-05-02 |        3.454 |        2.818
 1335 | 2022-05-01 |       17.548 |        0.000
 1334 | 2022-04-30 |        9.373 |        0.000
 1333 | 2022-04-29 |       18.756 |        3.748
 1332 | 2022-04-28 |       18.021 |        5.322
```

dennissiemensma, 2022-12-01:
In principe kun je een partial backup relatief makken terugzetten als je de ID-kolom weghaalt uit de dump, omdat die botst. Dat vereist wel wat handwerk.

Eventuele al bestaande datums geven dan vanzelf wel een fout en kun je dan telkens uit je dump halen die je inleest. Beetje een shotgun methode door gewoon hard telkens data proberen te importeren.

dennissiemensma, 2022-12-01:
Alternatief is inderdaad via de webinterface invoeren, maar zodra je meer dan 5 of 10 moet doen, zou ik daar al niet aan beginnen

dennissiemensma, 2022-12-01:
> Kan ik met de hand niet-bestaande ID's toewijzen? 

Je hoeft zelf nooit de ID's te kiezen, dat kan de DB het beste voor je doen. Als je dus uit de dump de kolom ID + de waardes van ID eruit haalt, pakt de DB vanzelf oplopende getallen (als het goed is).

dennissiemensma, 2022-12-01:
Dit is overigens exact de reden dat die partial backups gemaakt worden, als last-resort backup. Fijn dat die om een of andere reden nog wel gemaakt en bewaard zijn.

rodeho, 2022-12-02:
Ha @dennissiemensma 

Ik heb een file gemaakt met een verwijderde ID kolom, gaat goed hij valt echter over hetzelfde punt als met een handmatige webgui toevoeging. Hij wil het ID nummer doornummeren vanaf een reeds bestaand nummer.

Heb jij nog een oplossing in petto om hier toe te passen?

```
ik@dsmr:/home/dsmr/dsmr-reader/backups$ sudo cat backup1.sql | sudo -u postgres psql dsmrreader
SET
SET
SET
SET
SET
 set_config
------------

(1 row)

SET
SET
SET
SET
ERROR:  duplicate key value violates unique constraint "dsmr_stats_daystatistics_pkey"
DETAIL:  Key (id)=(1336) already exists.
CONTEXT:  COPY dsmr_stats_daystatistics, line 1
 setval
--------
   1335
(1 row)

ik@dsmr:/home/dsmr/dsmr-reader/backups$ sudo su - postgres
postgres@dsmr:~$ psql dsmrreader
psql (14.5 (Ubuntu 14.5-0ubuntu0.22.04.1), server 12.11 (Ubuntu 12.11-0ubuntu0.20.04.1))
Type "help" for help.

dsmrreader=# select id, day, electricity1, electricity2 from dsmr_stats_daystatistics order by day desc;
  id  |    day     | electricity1 | electricity2
------+------------+--------------+--------------
 1549 | 2022-12-01 |       15.527 |        5.253
 1548 | 2022-11-30 |       25.622 |        6.651
 1547 | 2022-11-29 |        3.066 |        6.501
 1546 | 2022-11-28 |        3.638 |        5.930
 1545 | 2022-11-27 |       14.805 |        0.000
 1544 | 2022-11-26 |       44.350 |        0.000
 1543 | 2022-11-25 |        3.496 |        6.013
 1542 | 2022-11-24 |        3.467 |        6.050
 1541 | 2022-11-23 |       37.622 |        8.550
 1540 | 2022-11-22 |        3.193 |        6.382
 1539 | 2022-11-21 |        3.397 |        5.631
 1538 | 2022-11-20 |       11.360 |        0.000
 1537 | 2022-11-19 |       36.382 |        0.000
 1536 | 2022-11-18 |        3.082 |        6.852
 1535 | 2022-11-17 |        3.641 |       10.988
 1534 | 2022-11-16 |       31.713 |        5.985
 1533 | 2022-11-15 |        3.298 |        5.433
 1532 | 2022-11-14 |       26.265 |        6.350
 1531 | 2022-11-13 |        6.087 |        0.000
 1530 | 2022-11-12 |        6.372 |        0.000
 1529 | 2022-11-11 |       20.045 |        3.740
 1528 | 2022-11-10 |        2.784 |        5.372
 1527 | 2022-11-09 |        2.649 |       10.183
 1526 | 2022-11-08 |       30.663 |       10.144
 1525 | 2022-11-07 |        2.807 |        7.289
 1524 | 2022-11-06 |       11.914 |        0.000
 1523 | 2022-11-05 |       41.598 |        0.000
 1522 | 2022-11-04 |        3.777 |        4.994
 1521 | 2022-11-03 |        3.698 |        7.104
 1520 | 2022-11-02 |       33.527 |        5.065
 1519 | 2022-11-01 |        3.659 |        5.710
 1518 | 2022-10-31 |        3.362 |        5.357
[... 208 lines cut ...]
 1309 | 2022-04-05 |       25.216 |       18.347
 1308 | 2022-04-04 |        3.911 |       21.421
 1307 | 2022-04-03 |       47.640 |        0.000
 1306 | 2022-04-02 |       38.504 |        0.000
 1305 | 2022-04-01 |       13.512 |        7.345
 1304 | 2022-03-31 |        2.487 |       28.443
 1303 | 2022-03-30 |        2.824 |        5.676
 1302 | 2022-03-29 |        2.709 |       39.883
 1301 | 2022-03-28 |        2.381 |        3.089
 1300 | 2022-03-27 |       21.638 |        0.000
 1299 | 2022-03-26 |        8.463 |        0.000
 1298 | 2022-03-25 |        2.896 |        3.372
 1297 | 2022-03-24 |        2.740 |        5.131
 1296 | 2022-03-23 |       29.748 |       10.406
 1295 | 2022-03-22 |       13.966 |        3.769
 1294 | 2022-03-21 |        2.485 |        4.275
 1293 | 2022-03-20 |        7.373 |        0.000
 1292 | 2022-03-19 |        0.160 |        0.000
 1291 | 2022-01-28 |       32.586 |        8.258
 1290 | 2022-01-27 |       30.708 |        8.080
 1289 | 2022-01-26 |        3.144 |        6.327
 1288 | 2022-01-25 |        3.265 |       11.371
 1287 | 2022-01-24 |        3.178 |       33.543
 1286 | 2022-01-23 |       11.170 |        0.000
```

rodeho, 2022-12-02:
Ik denk dat het gelukt is, ben eens gaan grasduinen in een full backup en kwam wat tegen.

`SELECT pg_catalog.setval('public.dsmr_stats_daystatistics_id_seq', 1542, true);`

Daarmee heb ik de ID teller vanaf 1542 laten tellen (wat vandaag de laatste waarde was) en dan komen de waardes er perfect in.

Kwam er op door bv [dit](https://stackoverflow.com/questions/4448340/postgresql-duplicate-key-violates-unique-constraint) google onderzoek

[image]

dennissiemensma, 2022-12-02:
Oke top! Je kon uiteindelijk deze proberen, die fixt het ook voor je: https://dsmr-reader.readthedocs.io/en/v5/how-to/troubleshooting/error-key-already-exists.html

rodeho, 2022-12-02:
Tnx voor je hulp! Ben erg blij dat ik nu een compleet jaar aan data heb voor vergelijkingen.

closed 2022-12-04 by dennissiemensma